ARMmbed/mbed-os #10817: DataFlash & LFS: fails to format

pilotak, 2019-06-12:
### Description
Following code worked fine just before #10478 
```cpp
#include <mbed.h>
#include "LittleFileSystem.h"
#include "DataFlashBlockDevice.h"

LittleFileSystem fs("fs");
DataFlashBlockDevice bd(PA_7, PA_6, PA_5, PA_4);

int main() {
    if (bd.init() == BD_ERROR_OK) {
        printf("BD initialized\n");

        if (fs.format(&bd) == BD_ERROR_OK) {
            printf("Storage formated\n");

        } else {
            printf("Storage failed to format\n");
        }

    } else {
        printf("Could not init BD\n");
    }

    return 0;
}
```
but after that i got an error
```
BD initialized
lfs info:243: format(0x20000d2c, 64, 64, 512, 512)

++ MbedOS Error Info ++
Error Status: 0x80FF0144 Code: 324 Module: 255
Error Message: Assertion failed: lfs->cfg->block_size % lfs->cfg->prog_size == 0
Location: 0x8004C39
File: .\mbed-os\features\storage\filesystem\littlefs\littlefs\lfs.c+2075
Error Value: 0x0
Current Thread: main  Id: 0x20001344 Entry: 0x8005957 StackSize: 0x1000 StackMem: 0x20001AD0 SP: 0x200028DC
For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=NUCLEO_F303RE
-- MbedOS Error Info --
```
the workaround is to specify `block size` directly to format. 
```cpp
fs.format(&bd, MBED_LFS_READ_SIZE, MBED_LFS_PROG_SIZE, 4224, MBED_LFS_LOOKAHEAD)
```
But is this a correct problem solver? or does `MBED_LFS_BLOCK_SIZE` need to be changed to `4224` but than if i'm right each file currently takes up an entire block as per documentation. https://github.com/ARMmbed/mbed-os/blob/378e997e2509d04c875981d1a9405308e40ff5f4/features/storage/filesystem/littlefs/LittleFileSystem.h#L83

MCU: STM32F303RE
Flash: AT45DB161E-SSHD

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

chrissnow, 2019-06-19:
I suspect that because the default MBED_LFS_BLOCK_SIZE is 512 and you are using native dataflash pages of 4224 the check that pages are multiple of erases is failing,

I'm not sure why it worked before though.

Unless you have a reason for 4224  over 4096 pages try this in your mbed_app.json
"dataflash.binary-size": "1"

I'm not sure why LFS is using defaults from mbed_lib.json rather than taking them from the underlying BD though.

chrissnow, 2019-06-19:
Also this might work

fs.format(&bd, MBED_LFS_READ_SIZE, 528, 4224, MBED_LFS_LOOKAHEAD)

chrissnow, 2019-06-19:
This might also work for native dataflash and give you smaller files, erases are 512 or 528 on the 16mb. 

fs.format(&bd, MBED_LFS_READ_SIZE, 528, 528, MBED_LFS_LOOKAHEAD)

I can't test it at the moment though.

pilotak, 2019-06-19:
thank you, i will try that next week. You raised a lot of questions about LFS it might be good to look at it a bit deeper.

vidavidorra, 2019-07-03:
I've had the exact same problem. As @chrissnow pointed out, adding `"dataflash.binary-size": 1` to my _mbed_app.json_ resolved the issue. According to the debug traces from LFS, the issue seems to be a mismatch in `lfs->cfg->block_size % lfs->cfg->prog_size` so the result is not `0`. In my case I had `block_size = 528` and `prog_size = 64` so the remaining is 16 (`528 mod 64 = 16`), thus the assertion fails. I'm not sure whether that an issue with the default values from the DataFlash API for the `PROG_SIZE` parameter. I at least couldn't find proper documentation for the `binary-size` option.

pilotak, 2019-07-03:
@chrissnow Here are some results:

- with no overriding in `mbed_app.json` and `fs.format(&bd, MBED_LFS_READ_SIZE, 528, 528, MBED_LFS_LOOKAHEAD)` there is an **format storage error (mbed error)**

- with no overriding in `mbed_app.json` and `fs.format(&bd, MBED_LFS_READ_SIZE, 528, 4224, MBED_LFS_LOOKAHEAD)` there is an **format storage error (mbed error)**

- with `"littlefs.block_size": 528` in `mbed_app.json` and `fs.format(&bd, MBED_LFS_READ_SIZE, MBED_LFS_PROG_SIZE, 4224, MBED_LFS_LOOKAHEAD)` **storage is formatted but `mount()` fails**

- with `"littlefs.block_size": 4224` in `mbed_app.json` and `fs.mount(&bd)` there is an **format storage error (mbed error)**

- with `"littlefs.block_size": 4224` in `mbed_app.json` and `fs.mount(&bd, MBED_LFS_READ_SIZE, 528, MBED_LFS_BLOCK_SIZE, MBED_LFS_LOOKAHEAD)`  there is an **format storage error (mbed error)**

- with `"littlefs.block_size": 4224` and `"littlefs.prog_size": 528` in `mbed_app.json` and `fs.mount(&bd)`  **there no mbed error it fails to format**

- with `"dataflash.binary-size": 1` in `mbed_app.json` and `fs.mount(&bd)` surprisingly **no problems** even though flash is not binary-sized

So in order to get dataflash up and running for beginers there should be binary size set to 1 as default if i'm right

https://github.com/ARMmbed/mbed-os/blob/608e4c245fe44bf281b1887508c71b729c669a29/components/storage/blockdevice/COMPONENT_DATAFLASH/mbed_lib.json#L11

vidavidorra, 2019-07-04:
I also want to point out that this was a breaking change without (IMHO) any proper documentation on how to get the same behaviour as a previous version of the OS or what changes would be needed at all.

0xc0170, 2019-07-04:
@vidavidorra Thanks for the feedback. How did `10478` break your code (@chrissnow added release notes section in PR , they should be in the 5.13 release notes) ?

I would like to get details for this one, to understand what should have raised red flag in that PR.

0xc0170, 2019-08-13:
@pilotak I'll close this issue as it seems it was answered. The remaining question is about the breakage (if it was not properly handled) - can be discussed further with issue closed.

closed 2019-08-13 by 0xc0170

vidavidorra, 2019-08-13:
@0xc0170 To be honest, I'm not sure that #10478 specifically broke my code. All I know is that somewhere between v5.11.6 and v5.13.0 it became required to set `"dataflash.binary-size": "1"` in order to get the same functionality I had in v5.11.6. I didn't see that documented in the release notes between that range so unfortunately can't tell you where it went wrong. Anyhow, `"dataflash.binary-size": "1"` fixed my issue regardless from where it is originated.
What I do want to note is that that option isn't documented in the v5.13.0 [docs](https://os.mbed.com/docs/mbed-os/v5.13/reference/storage.html#Default-BlockDevice-configuration) so IMHO that should be added and than its fine.

pilotak, 2019-08-13:
well basically with option `"dataflash.binary-size": "0"` it doens't work

0xc0170, 2019-08-23:
> @0xc0170 To be honest, I'm not sure that #10478 specifically broke my code. All I know is that somewhere between v5.11.6 and v5.13.0 it became required to set "dataflash.binary-size": "1" in order [...]
What I do want to note is that that option isn't documented in the v5.13.0 docs so IMHO that should be added and than its fine.

@ARMmbed/mbed-os-storage Can you please review the docs if there is anything we can fix for this one?