ARMmbed/mbed-os #10819: [OoB 5.13] Cannot compile with ARMC6 and Python 3.7.3 on Windows 10

coisme, 2019-06-13:
### Description

I tried compiling these example programs with ARMC6 but failed with the same error:
- [mbed-os-example-blinky](https://github.com/ARMmbed/mbed-os-example-blinky)
- [mbed-os-example-tls-socket](https://github.com/ARMmbed/mbed-os-example-tls-socket)

```
PS C:\Users\osakoi01\mbed\mbed-os-example-blinky> mbed compile -t ARM -m LPC55S69_NS
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "C:\Users\osakoi01\mbed\mbed-os-example-blinky" (program)
Building project mbed-os-example-blinky (LPC55S69_NS, ARMC6)
Scan: mbed-os-example-blinky
Compile [  0.1%]: main.cpp
[ERROR] '_queue.SimpleQueue' object has no attribute 'queue'
[mbed] ERROR: "c:\users\osakoi01\appdata\local\programs\python\python37\python.exe" returned error.
       Code: 1
       Path: "C:\Users\osakoi01\mbed\mbed-os-example-blinky"
       Command: "c:\users\osakoi01\appdata\local\programs\python\python37\python.exe -u C:\Users\osakoi01\mbed\mbed-os-example-blinky\mbed-os\tools\make.py -t ARM -m LPC55S69_NS --source . --build .\BUILD\LPC55S69_NS\ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---
[mbed] WARNING: Using Python 3 with Mbed OS 5.8 and earlier can cause errors with compiling, testing and exporting
---
```

**Environment**
Windows 10

```
PS C:\Users\osakoi01\mbed\mbed-os-example-blinky> mbed --version
1.10.0
PS C:\Users\osakoi01\mbed\mbed-os-example-blinky> python --version
Python 3.7.3
PS C:\Users\osakoi01\mbed\mbed-os-example-blinky> mbed ls
[mbed] WARNING: If you're using Python 3 with Mbed OS 5.8 and earlier versions, Python errors will occur when compiling, testing and exporting
---
[mbed] Working path "C:\Users\osakoi01\mbed\mbed-os-example-blinky" (program)
mbed-os-example-blinky (#0840020675f3)
`- mbed-os (#cd8e315ba587, tag: mbed-os-5.13.0-rc2)
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-06-13:
I retested this on master and also on 5.13.0-rc2 tag:

 mbed compile -m LPC55S69_NS -t ARM -> no error 

Where is `_queue.SimpleQueue` coming from ? It's not in the blinky example code file, neither in Mbed OS codebase (I searched in both).

coisme, 2019-06-14:
I tried another Windows machine installed Python 3.7.3 and mbed-cli 1.10.0. This issue was not reproduced on this machine, too. Is there any possibility to cause this issue...?

0xc0170, 2019-06-14:
Interesting enough I got this similar error once I added few more stats flags for regular test command, I am investigating.

0xc0170, 2019-08-14:
Fixed via #10832 issue

closed 2019-08-14 by 0xc0170