ARMmbed/mbed-os #10831: setTxPower is not implemented

harc-cypress, 2019-06-14:
### Description
For ble::gap::setTxPower  API, it is returning BLE_ERROR_NOT_IMPLEMENTED error.
From code looks like for extended adv case, setTxPower is supported via AdvertisingParameters class. 
But if extended adv is not supported, how application can set tx power?

### Issue request type

    [ ] Question
    [ x] Enhancement
    [ ] Bug

Blackfire-Foo, 2019-07-01:
I have a similar question here. How can I increase (change) BLE TxPower if I use Legacy rather than Extended mode? I'm using nRF52480 + Cordio LL. Anyone can help on this?
Thanks in advance.

felipeLeast, 2019-08-14:
Did anyone had a follow up on this issue?

0xc0170, 2019-08-23:
cc @ARMmbed/mbed-os-pan

paul-szczepanek-arm, 2019-08-23:
You can still use the new API with extended advertising disabled, use setAdvertisingParameters with LEGACY_ADVERTISING_HANDLE.

0xc0170, 2019-08-23:
I closed this as resolved via proposed solution above. If that is not the case, let us know

closed 2019-08-23 by 0xc0170