ARMmbed/mbed-os #10832: PR tests are failing:  FVP_MPS2_xxx - IAR assembly returns error

0xc0170, 2019-06-14:
### Description

Couple of latest CI tests overnight failed. Also nightly: http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/239/build-IAR/FAIL/

Looks like latest PR within a day introduced this or change in CI config?

Locally, if I invoke `mbed test -m FVP_MPS2_M0 -t iar -n  tests-mbedmicro-mbed-cpp --compile` , no build errors. However, once I add few macros our CI uses (stats), I get similar cmain.S error - the only addition is I get an error

```
mbed test -m FVP_MPS2_M0 -t iar -n  tests-mbedmicro-mbed-cpp --compile

....

Build successes:
  * FVP_MPS2_M0::IAR::MBED-BUILD
  * FVP_MPS2_M0::IAR::TESTS-MBEDMICRO-MBED-CPP
```

```
mbed test -m FVP_MPS2_M0 -t iar -n  tests-mbedmicro-mbed-cpp --compile -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_ALL_STATS_ENABLED -DSKIP_TIME_DRIFT_TESTS=1

...

Compile [ 83.9%]: Dir.cpp
Compile [ 84.0%]: FlashSimBlockDevice.cpp
Compile [ 84.2%]: mbed_mpu_v8m.c
Compile [ 84.3%]: cmain.S
'_queue.SimpleQueue' object has no attribute 'queue'
Failed to build library

```

cc @mbed-os-tools @jamesbeyond (not certain why only FVP targets are triggering this error)

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-06-14:
I've got similar setup than reported here (same queue error): https://github.com/ARMmbed/mbed-os/issues/10819

Mine:

python 3.7.3
Mbed OS latest master: ` 5a8f795976 - (HEAD -> master, upstream/master) Merge pull request #10804 from VVESTM/vve_fix_10589 (29 minutes ago)`
windows 10

however me using IAR , and the other report is for ARM (I can't reproduce with ARM).

I can't find the code line for queue error:

```
Searching 29014 files for "_queue.SimpleQueue"

0 matches
```

0xc0170, 2019-06-14:
verbose uncovers the tools error

```
Compile [ 88.4%]: except.S
Compile: IAR Systems\Embedded Workbench 8.2\arm\bin\iasmarm --cpu Cortex-M0 -D__CORTEX_M0 -DCMSDK_CM0 -D__MBED_CMSIS_RTOS_CM -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -DSKIP_TIME_DRIFT_TESTS=1 -DMBED_HEAP_STATS_ENABLED=1 -D__CMSIS_RTOS -DSKIP_TIME_DRIFT_TESTS -DARM_MATH_CM0 -DMBED_TEST_MODE -DMBED_ALL_STATS_ENABLED -f .\BUILD\tests\FVP_MPS2_M0\IAR\asm_macros_64a4b1af7064246e726d6d1ee1bcd753.xcl -o BUILD\tests\FVP_MPS2_M0\IAR\platform\TARGET_CORTEX_M\TOOLCHAIN_IAR\except.o .\platform\TARGET_CORTEX_M\TOOLCHAIN_IAR\except.S
[DEBUG] Return: 3
Traceback (most recent call last):
  File "mbed-os\tools\toolchains\mbed_toolchain.py", line 522, in compile_queue
    res['command']
  File "mbed-os\tools\toolchains\mbed_toolchain.py", line 649, in compile_output
    raise ToolException(stderr)
tools.utils.ToolException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "F:\Code\mbed-os\tools\test.py", line 256, in main
    resource_filter=resource_filter
  File "F:\Code\mbed-os\tools\build_api.py", line 769, in build_library
    res, res.get_file_paths(FileType.INC_DIR))
  File "F:\Code\mbed-os\tools\toolchains\mbed_toolchain.py", line 458, in compile_sources
    return self.compile_queue(queue, objects)
  File "F:\Code\mbed-os\tools\toolchains\mbed_toolchain.py", line 526, in compile_queue
    if p._taskqueue.queue:
AttributeError: '_queue.SimpleQueue' object has no attribute 'queue'
'_queue.SimpleQueue' object has no attribute 'queue'
Failed to build library
```

jamesbeyond, 2019-06-14:
I am looking into that now

0xc0170, 2019-06-14:
@OPpuolitaival this started the last night (even some earlier PRs test was not failing - 17 hours ago job for 10812 passed then the job for 10776 few hours later failed and few others as well.

The latest nightly same, but previous days all good. This means something broke yesterday. I can't find any PR merged yesterday to master (prior failing and after nightly passed from 2 nights ago). This leads us to a question - what has changed in CI ?

Was it `-DSKIP_TIME_DRIFT_TESTS=1` addition? If I run test command from above without this macro, it passes

```
mbed test -m FVP_MPS2_M0 -t iar -n  tests-mbedmicro-mbed-cpp --compile -DMBED_HEAP_STATS_ENABLED=1 -DMBED_STACK_STATS_ENABLED=1 -DMBED_TRAP_ERRORS_ENABLED=1 -DMBED_ALL_STATS_ENABLED -v

...

  * FVP_MPS2_M0::IAR::MBED-BUILD
  * FVP_MPS2_M0::IAR::TESTS-MBEDMICRO-MBED-CPP

```

One additional -D to the compile triggers the exception in our tools. The syntax error for simplequeue should be fixed and lets find out what is causing the exception.

cc @mbed-os-tools to be aware

0xc0170, 2019-06-15:
fixed via #10833

closed 2019-06-15 by 0xc0170