ARMmbed/mbed-os #10848: IAR exporter - IAR 8 support

embeddedteam103, 2019-06-17:
### Description
We're trying to use the new mbed-os 5.12.4 with IAR 8.30.1.
  
It seems like there's a version incompatibility with the IAR version of the generated workspace,
causing the compiler and linker flags to be discarded.

The errors when opening the workspace are as follows:

`Configuration <project_name> contains non-recoganized options for the tool `ICCARM`
Settings `ICCARM` unsupported version - 35`
   
`Configuration <project_name> contains non-recoganized options for the tool `ILINK`
Settings `ILINK` unsupported version - 22`

- mbed 5.12.4
- IAR 8.30.1
- mbed-cli 1.9.2
- Target is: DISCO_F407VG

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-06-18:
Hi
I am not an export expert... but maybe you should update your IAR version to 8.32.1,
as recommended in https://os.mbed.com/docs/mbed-os/v5.12/tools/index.html
?

embeddedteam103, 2019-07-08:
Seems like that was the problem,
Thanks @jeromecoutant

closed 2019-07-08 by embeddedteam103