ARMmbed/mbed-os #10852: [Cellular] AT_CellularContext::disconnect() may return fail if activate a new context on Lierda NB86-G module

ccchang12, 2019-06-18:
### Description

When activate a new context, both of the flag _is_context_activated and _new_context_set may be one, and the AT+CGDCONT=cid AT command will be called twice. On some modules such as Quectel BG96, it is OK. But on Lierda NB86-G, only the first time is success, the second time will return an error due to the cid is not exist anymore. And this will cause the AT_CellularContext::disconnect() returns an error, even the process is success.

    // deactivate a context only if we have activated
    if (_is_context_activated) {
        if (_nonip_req && _cp_in_use) {
            deactivate_non_ip_context();
        } else {
            deactivate_ip_context();
        }
    }
    _is_context_active = false;
    _connect_status = NSAPI_STATUS_DISCONNECTED;

    // call device's callback, it will broadcast this to here (cellular_callback)
    _device->cellular_callback(NSAPI_EVENT_CONNECTION_STATUS_CHANGE, NSAPI_STATUS_DISCONNECTED, this);

    if (_new_context_set) {
        delete_current_context();
    }
    _cid = -1;

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

AriParkkila, 2019-07-23:
@ccchang12 Please see PR #11090

AnttiKauppila, 2019-09-12:
Closed in JIRA due to inactivity