ARMmbed/mbed-os #10853: [STM32L4]serial_device irq

huarunlin, 2019-06-18:
### Description
TARGET_STM32L4/serial_device.c function serial_irq_handler_asynch & uart_irq,  get Interrupt Source should use the macro __HAL_UART_GET_IT_SOURCE, instead of __HAL_UART_GET_IT.

> static void uart_irq(UARTName uart_name)

            if (__HAL_UART_GET_FLAG(huart, UART_FLAG_TXE) != RESET) {
                if (__HAL_UART_GET_IT(huart, UART_IT_TXE) != RESET) {  //should use __HAL_UART_GET_IT_SOURCE_
                    irq_handler(serial_irq_ids[id], TxIrq);
                }
            }
            if (__HAL_UART_GET_FLAG(huart, UART_FLAG_RXNE) != RESET) {
                if (__HAL_UART_GET_IT(huart, UART_IT_RXNE) != RESET) {  //should use __HAL_UART_GET_IT_SOURCE
                    irq_handler(serial_irq_ids[id], RxIrq);
                    /* Flag has been cleared when reading the content */
                }
            }
            if (__HAL_UART_GET_FLAG(huart, UART_FLAG_ORE) != RESET) {
                if (__HAL_UART_GET_IT(huart, UART_IT_ORE) != RESET) { //should use __HAL_UART_GET_IT_SOURCE
                    volatile uint32_t tmpval __attribute__((unused)) = huart->Instance->RDR; // Clear ORE flag
                }

> int serial_irq_handler_asynch(serial_t *obj)

    // Handle error events
    if (__HAL_UART_GET_FLAG(huart, UART_FLAG_PE) != RESET) {
        if (__HAL_UART_GET_IT(huart, UART_IT_PE) != RESET) { //should use __HAL_UART_GET_IT_SOURCE
            return_event |= (SERIAL_EVENT_RX_PARITY_ERROR & obj_s->events);
        }
    }

    if (__HAL_UART_GET_FLAG(huart, UART_FLAG_FE) != RESET) {
        if (__HAL_UART_GET_IT(huart, UART_IT_FE) != RESET) { //should use __HAL_UART_GET_IT_SOURCE
            return_event |= (SERIAL_EVENT_RX_FRAMING_ERROR & obj_s->events);
        }
    }

    if (__HAL_UART_GET_FLAG(huart, UART_FLAG_ORE) != RESET) {
        if (__HAL_UART_GET_IT(huart, UART_IT_ORE) != RESET) { //should use __HAL_UART_GET_IT_SOURC

            return_event |= (SERIAL_EVENT_RX_OVERRUN_ERROR & obj_s->events);
        }
    }

### Issue request type

    [ ] Question
    [ ] Enhancement
    [* ] Bug

LMESTM, 2019-06-18:
Hi @huarunlin 

Is there a use case where you can show this is a bug and it can be corrected with your patch ? 

The reason for using such MACRO can be found in history :
```
$ git show be8024de28e14697d50fc577baf82275c68cac49 -q
commit be8024de28e14697d50fc577baf82275c68cac49 (origin/stm32_serial_it_source, lmestm/stm32_serial_it_source, stm32_serial_it_source)
Author: Laurent MEUNIER <laurent.meunier@st.com>
Date:   Tue Jul 4 09:57:10 2017 +0200

    STM32: serial: use proper GET_IT macro

    Depending on families, different HAL macros are defined to check the
    state of serial interrupts. In several cases, we can find only 1 macro:
    __HAL_UART_GET_IT_SOURCE
    Checks whether the specified UART interrupt has occurred or not

    But in F0, F3, F7, L0, L4 there are 2 different macros
    __HAL_UART_GET_IT
    Checks whether the specified UART interrupt has occurred or not
    __HAL_UART_GET_IT_SOURCE
    Checks whether the specified UART interrupt source is enabled.

    In the later case, __HAL_UART_GET_IT_SOURCE was being used so far,
    but actually needs to be replaced by __HAL_UART_GET_IT. Using the right
    macro, we also check the proper flags accordingly.
```

But of course if you think there is a bug and you can show the bug in an application we will have a look.

jeromecoutant, 2019-06-18:
Hi
Please check also #10774

huarunlin, 2019-06-18:
Hi @LMESTM 

__HAL_UART_GET_FLAG is to get the current interrupt flag by ISR register
__HAL_UART_GET_IT is the same as __HAL_UART_GET_FLAG, which is the current interrupt flag obtained from the ISR register.
__HAL_UART_GET_IT_SOURCE is to get the interrupt enable

> In the uart_irq function

first call __HAL_UART_GET_FLAG to check for an interrupt, then __HAL_UART_GET_IT_SOURCE should be called to check if the interrupt is enabled. It is not called __HAL_UART_GET_IT to repeat the check to see if there is an interrupt.

> In the serial_irq_handler_asynch function

Is used in the treatment TX_COMPLETE __HAL_UART_GET_IT_SOURCE, but later in the code Handle error events, with the __HAL_UART_GET_IT

> use __HAL_UART_GET_IT

    // TX PART:
    if (__HAL_UART_GET_FLAG(huart, UART_FLAG_TC) != RESET) {
        if (__HAL_UART_GET_IT_SOURCE(huart, UART_IT_TC) != RESET) { 
            // Return event SERIAL_EVENT_TX_COMPLETE if requested
            if ((obj_s->events & SERIAL_EVENT_TX_COMPLETE) != 0) {
                return_event |= (SERIAL_EVENT_TX_COMPLETE & obj_s->events);
            }
        }
    }

> use __HAL_UART_GET_IT

    // Handle error events
    if (__HAL_UART_GET_FLAG(huart, UART_FLAG_PE) != RESET) {
        if (__HAL_UART_GET_IT(huart, UART_IT_PE) != RESET) {  
            return_event |= (SERIAL_EVENT_RX_PARITY_ERROR & obj_s->events);
        }
    }

    if (__HAL_UART_GET_FLAG(huart, UART_FLAG_FE) != RESET) {
        if (__HAL_UART_GET_IT(huart, UART_IT_FE) != RESET) {   
            return_event |= (SERIAL_EVENT_RX_FRAMING_ERROR & obj_s->events);
        }
    }

    if (__HAL_UART_GET_FLAG(huart, UART_FLAG_ORE) != RESET) {
        if (__HAL_UART_GET_IT(huart, UART_IT_ORE) != RESET) {
            return_event |= (SERIAL_EVENT_RX_OVERRUN_ERROR & obj_s->events);
        }
    }

huarunlin, 2019-06-18:
@jeromecoutant OK,Thanks

LMESTM, 2019-06-18:
@huarunlin ok clearer now. 

I actually based the changes on HAL comments but I was wrong.
So I understand now:
  __HAL_UART_GET_IT is not needed as we can simply check the flag

On F1, F2, F4, 
    /** @brief  Check whether the specified UART interrupt has occurred or not.
    __HAL_UART_GET_IT_SOURCE
for the comment is wrong, it actually checks whether interrupt is enabled or not

So I now agree that we should check the flag and whether interrupt is enabled or not.

jeromecoutant, 2019-10-11:
Hi
Issue can be closed as #10774 is merged
Thx

closed 2019-10-24 by 0xc0170