ARMmbed/mbed-os #10854: [ATHandler] Mbed Os 5.12.4  ATHandler::resp_stop

huarunlin, 2019-06-18:
### Description

my program works fine in Mbed Os 5.12.2 version, updated to version 5.12.4 is abnormal, and finally found ATHandler :: resp_stop returns a failure, I alone put ATHandler :: resp_stop fall back Mbed Os 5.12.2 version, the program It works fine.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [*] Bug

AriParkkila, 2019-07-22:
@huarunlin I don't find any difference in `git diff mbed-os-5.12.2..mbed-os-5.12.4 -- features/cellular/framework/AT/ATHandler.cpp`.

Can you share the changes you made in `ATHandler :: resp_stop` and/or AT debug logs?

huarunlin, 2019-07-22:
@AriParkkila , I found the problem, I am using the QUECTEL M26 module. QUECTEL_M26_CellularStack.cpp: socket_sendto_impl in on "AT + QISEND =" return directive judgment in question, there is no right to call ATHandler :: resp_start, modified to the following code to run properly.

    _at.cmd_start("AT+QISEND=");
    _at.write_int(socket->id);
    _at.write_int(sent_len);
    _at.cmd_stop();

    _at.resp_start(">");
    _at.write_bytes((uint8_t *)data, sent_len);
    //_at.resp_start();
    _at.resp_start("SEND ", true);
    _at.resp_stop();

AriParkkila, 2019-07-22:
@huarunlin Since it works in mbed-os-5.12.2 and AT driver has not changed to mbed-os-5.12.4, it's difficult to see what could have caused this. Can you share your AT spec and firmware version.

closed 2019-09-20 by ciarmcom