ARMmbed/mbed-os #10860: Compliance tests at Objenious (Bouygues Telecom) returned many errors

nicocarv, 2019-06-18:
### Description

I just sent my device to pass the LoRaWAN compliance tests of Objenious (Bouygues Telecom's LoRaWAN national network in France) and the result was a NO GO.
Problems detected with the stack:

**LinkADR / nbRepetitions = 2 or 3:**
If we send a LinkADR command to the device with parameter nbRepetitions = 2 or 3, the device will not take it into account and will continue to send the uplinks normally (no repetition).

**ADR back off:**
After 64 uplinks the device has to put the DR back off bit to 1 and wait for a confirmation from the network server.
After 32 uplinks with no confirmation, the device should increment the DR by one.
After 32 uplinks with no confirmation, the device should increment the DR by one.
..
When in SF12 it should just keep at SF12 while sending new frames.

The device is not doing the DR back off, it will just continue sending uplinks with no change on SF (in this case SF7)

**Modify RX delay for DW**
expected behavior: The network sends a new RX delay configuration to the device in order to change the time between RX1 and RX2. The device answers Ok to the network with a piggyback MAC command and waits until the network ACKs it before changing the RX delay. After the network's ACK, the device uses the new RX delay.

**Here we have 2 observed results:**

**1) The device send a normal uplink (data and port standard, different than 0) and adds the piggyback MAC command. But for a strange reason:**
- It send 2 times the piggyback MAC command the first time (FOptsLen = 4 bytes (Length of piggyback MAC commands)) 
- 4 times the piggyback MAC command the second time (FOptsLen = 8 bytes). 
- The third time it sends the data on PORT=0 with FOptsLen = 0 (normal) and FRMPayload is 16B total. So we can suppose that the piggyback MAC command was double of previous (16B) and this maximum size for piggyback MAC command on a normal frame is 15B (FCtrl = 0xXF, so FOptsLen = 15), it send it inside the FRMPayload with port= 0 (normal behave).

-> It looks like a bug with this configuration.

Sending config dw:

> 17-06-2019 18:13:07.491 : Downlink parameters : rxDelay = 6000 milli, frequency = 869.525 MHz, datarate = SF12BW125
> 17-06-2019 18:13:07.503 : LoRa Frame to be sent = 60DDCCBBAA80040000BAAC0B0456C889E42E
> 17-06-2019 18:13:07.514 : MHDR = 0x60
> 17-06-2019 18:13:07.525 : |=> MessageType  : 0x03 => [ Unconfirmed Data Down ]
> 17-06-2019 18:13:07.537 : |=> MajorVersion : 0x00 => [ LoRaWAN R1 ]
> 17-06-2019 18:13:07.548 : DevAddr = 0xAABBCCDD
> 17-06-2019 18:13:07.559 : FCtrl = 0x80
> 17-06-2019 18:13:07.570 : |=> ADR = true
> 17-06-2019 18:13:07.581 : |=> ADRACKReq = false
> 17-06-2019 18:13:07.595 : |=> ACK = false
> 17-06-2019 18:13:07.608 : |=> FPending = false
> 17-06-2019 18:13:07.619 : |=> FOptsLen = 0 bytes (Length of piggyback MAC commands)
> 17-06-2019 18:13:07.631 : FCnt = 0x0004 => 4
> 17-06-2019 18:13:07.645 : FPort = 0x00 => 0
> 17-06-2019 18:13:07.656 : FrmPayload = 0xBAAC0B0456
> 17-06-2019 18:13:07.669 : Decrypted FrmPayload = 0x0505D2AD84
> 17-06-2019 18:13:07.680 : Found MAC command in frmPayload
> 17-06-2019 18:13:07.690 : |=> [ RXParamSetupReq ] = 0x0505D2AD84
> 17-06-2019 18:13:07.702 : |=> RX1DataRateOffset = 0x0 => [ SF12BW125 ]
> 17-06-2019 18:13:07.713 : |=> RX2DataRate = 0x5 => [ SF7BW125 ]
> 17-06-2019 18:13:07.724 : |=> RX2Frequency = 869525000 Hz
> 17-06-2019 18:13:07.736 : MIC = 0xC889E42E
> 17-06-2019 18:13:07.747 : ====> SENDING JSON MESSAGE TO GATEWAY : {"txpk":{"modu":"LORA","data":"YN3Mu6qABAAAuqwLBFbIieQu","ant":1,"freq":869.525,"prea":8,"tmst":257882892,"rfch":0,"codr":"4/5","size":18,"datr":"SF12BW125","ipol":true,"ncrc":true,"powe":29}}

Receiving first Uplink after config:

> 17-06-2019 18:14:27.590 : <==== RECEIVED JSON MESSAGE FROM GATEWAY : {"modu":"LORA","stat":1,"rssi":-60,"data":"QN3Mu6qEGAAFBwUHmWoyB25zBljIBe+JAQ==","ant":1,"freq":866.7,"locstatus":"NoGps","tmst":332191620,"codr":"4/5","size":25,"datr":"SF10BW125","lsnr":12,"time":"1970-01-06T07:07:21.102343Z","chan":23,"chipid":"APFDBDBz9wo=","did":7}
> 17-06-2019 18:14:28.042 : Received LoRa Frame = 0x40DDCCBBAA84180005070507996A32076E730658C805EF8901
> 17-06-2019 18:14:28.055 : MHDR = 0x40
> 17-06-2019 18:14:28.063 : |=> MessageType  : 0x02 => [ Unconfirmed Data Up ]
> 17-06-2019 18:14:28.070 : |=> MajorVersion : 0x00 => [ LoRaWAN R1 ]
> 17-06-2019 18:14:28.077 : DevAddr = 0xAABBCCDD
> 17-06-2019 18:14:28.083 : FCtrl = 0x84
> 17-06-2019 18:14:28.089 : |=> ADR = true
> 17-06-2019 18:14:28.096 : |=> ADRACKReq = false
> 17-06-2019 18:14:28.102 : |=> ACK = false

here we see there is 4 bytes of piggyback MAC command:

[... 103 lines cut ...]
> 17-06-2019 17:57:32.636 : FCtrl = 0x80
> 17-06-2019 17:57:32.645 : |=> ADR = true
> 17-06-2019 17:57:32.654 : |=> ADRACKReq = false
> 17-06-2019 17:57:32.664 : |=> ACK = false
> 17-06-2019 17:57:32.673 : |=> FPending = false
> 17-06-2019 17:57:32.683 : |=> FOptsLen = 0 bytes (Length of piggyback MAC commands)
> 17-06-2019 17:57:32.693 : FCnt = 0x0001 => 1
> 17-06-2019 17:57:32.704 : FPort = 0x00 => 0
> 17-06-2019 17:57:32.716 : FrmPayload = 0xCA27EF3013
> 17-06-2019 17:57:32.726 : Decrypted FrmPayload = 0x0505D2AD84
> 17-06-2019 17:57:32.737 : Found MAC command in frmPayload
> 17-06-2019 17:57:32.748 : |=> [ RXParamSetupReq ] = 0x0505D2AD84
> 17-06-2019 17:57:32.757 : |=> RX1DataRateOffset = 0x0 => [ SF12BW125 ]
> 17-06-2019 17:57:32.767 : |=> RX2DataRate = 0x5 => [ SF7BW125 ]
> 17-06-2019 17:57:32.777 : |=> RX2Frequency = 869525000 Hz
> 17-06-2019 17:57:32.790 : MIC = 0xA8277E67
> 17-06-2019 17:57:32.801 : ====> SENDING JSON MESSAGE TO GATEWAY : {"txpk":{"modu":"LORA","data":"YN3Mu6qAAQAAyifvMBOoJ35n","ant":1,"freq":869.525,"prea":8,"tmst":3007817987,"rfch":0,"codr":"4/5","size":18,"datr":"SF12BW125","ipol":true,"ncrc":true,"powe":29}}
> 17-06-2019 17:57:32.812 : STEP : Waiting 300 sec for a LoRa Data message from the endpoint
> 17-06-2019 17:57:39.468 : <==== RECEIVED JSON MESSAGE FROM GATEWAY : {"modu":"LORA","stat":1,"rssi":-60,"data":"gN3Mu6qAAwAAn92w8wfY","ant":1,"freq":867.3,"locstatus":"NoGps","tmst":3009327852,"codr":"4/5","size":15,"datr":"SF7BW125","lsnr":9.5,"time":"1970-01-06T06:50:32.894687Z","chan":6,"chipid":"APFDBDBz9wo=","did":7}

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x ] Bug

nicocarv, 2019-07-01:
Hello again,

We are using Mbed-OS 5.9.6

Any body has had this kind of problem? Because of this we cannot start selling our devices on the LoRaWAN network here in France.

Any help is very appreciated. 
Thanks.

0xc0170, 2019-07-01:
Not certain why it was not cloned to our private tracker. cc @adbridge 

cc @ARMmbed/mbed-os-wan

adbridge, 2019-07-01:
> 
> 
> Not certain why it was not cloned to our private tracker. cc @adbridge
> 

Not sure either, probably came in when we were getting some github failures . I've closed and re-opened to see if it will get picked up next time the script runs.

adbridge, 2019-07-01:
@nicocarv 5.9.6 is quite an old release now (basically a year ago). Could I suggest you try with the latest mbed-os-5.13.0 and see if you get the same issues ? Thanks.

nicocarv, 2019-07-09:
Hello,

Ok, I will try that approach and come back to you.

Thanks.

adbridge, 2019-11-26:
@nicocarv any update on this or can we close it ?

adbridge, 2019-12-02:
As we have had no update on this issue since July we will close it. Please re-open if this is still a problem.

closed 2019-12-02 by adbridge