ARMmbed/mbed-os #10863: [kvstore]configure used QSPIF Block Device

huarunlin, 2019-06-19:
### Description

On the NUCLEO_L496ZG target, I used the KVStore module of Mbed Os 5.12.4, An error occurred while compiling the kv_config.cpp _get_blockdevice_QSPIF function

> Compile error

           [Error] kv_config.cpp@475,9: 'QSPI_FLASH1_IO0' was not declared in this scope
           [Error] kv_config.cpp@476,9: 'QSPI_FLASH1_IO1' was not declared in this scope
           [Error] kv_config.cpp@477,9: 'QSPI_FLASH1_IO2' was not declared in this scope
           [Error] kv_config.cpp@478,9: 'QSPI_FLASH1_IO3' was not declared in this scope
           [Error] kv_config.cpp@479,9: 'QSPI_FLASH1_SCK' was not declared in this scope
           [Error] kv_config.cpp@480,9: 'QSPI_FLASH1_CSN' was not declared in this scope
           [ERROR] .\mbed-os\features\storage\kvstore\conf\kv_config.cpp: In function 'mbed::BlockDevice* _get_blockdevice_QSPIF(mbed::bd_addr_t, mbed::bd_size_t)':
           .\mbed-os\features\storage\kvstore\conf\kv_config.cpp:475:9: error: 'QSPI_FLASH1_IO0' was not declared in this scope
         QSPI_FLASH1_IO0,
         ^~~~~~~~~~~~~~~
           .\mbed-os\features\storage\kvstore\conf\kv_config.cpp:476:9: error: 'QSPI_FLASH1_IO1' was not declared in this scope
         QSPI_FLASH1_IO1,
         ^~~~~~~~~~~~~~~
           .\mbed-os\features\storage\kvstore\conf\kv_config.cpp:477:9: error: 'QSPI_FLASH1_IO2' was not declared in this scope
         QSPI_FLASH1_IO2,
         ^~~~~~~~~~~~~~~
           .\mbed-os\features\storage\kvstore\conf\kv_config.cpp:478:9: error: 'QSPI_FLASH1_IO3' was not declared in this scope
         QSPI_FLASH1_IO3,
         ^~~~~~~~~~~~~~~
           .\mbed-os\features\storage\kvstore\conf\kv_config.cpp:479:9: error: 'QSPI_FLASH1_SCK' was not declared in this scope
         QSPI_FLASH1_SCK,
         ^~~~~~~~~~~~~~~
           .\mbed-os\features\storage\kvstore\conf\kv_config.cpp:480:9: error: 'QSPI_FLASH1_CSN' was not declared in this scope
         QSPI_FLASH1_CSN,

> kv_config.cpp

    BlockDevice *_get_blockdevice_QSPIF(bd_addr_t start_address, bd_size_t size)
    {
    #if COMPONENT_QSPIF

    bd_addr_t aligned_end_address;
    bd_addr_t aligned_start_address;

    static QSPIFBlockDevice bd(
        QSPI_FLASH1_IO0,
        QSPI_FLASH1_IO1,
        QSPI_FLASH1_IO2,
        QSPI_FLASH1_IO3,
        QSPI_FLASH1_SCK,
        QSPI_FLASH1_CSN,
        QSPIF_POLARITY_MODE_0,
        MBED_CONF_QSPIF_QSPI_FREQ
    );

    if (bd.init() != MBED_SUCCESS) {
        tr_error("KV Config: QSPIFBlockDevice init fail");

> mbed_app.json

            "components_add" : ["QSPIF"],
            "device_has_add" : ["QSPI"],
            "qspif.QSPI_IO0" : "PE_12",
            "qspif.QSPI_IO1" : "PE_13",
            "qspif.QSPI_IO2" : "PE_14",
            "qspif.QSPI_IO3" : "PE_15",
            "qspif.QSPI_SCK" : "PE_10",
            "qspif.QSPI_CSN" : "PE_11",
            "storage.storage_type" : "FILESYSTEM_NO_RBP",
            "storage_filesystem_no_rbp.blockdevice" : "QSPIF"

Target NUCLEO_L496ZG does not define QSPI_FLASH1_XXX on PinNames.h，and should use my qspif pin configuration on mbed_app.json, so should _get_blockdevice_QSPIF function of QSPI_FLASH1_XXX changed MBED_CONF_QSPIF_QSPI_XXX
### Issue request type

    [ ] Question
    [*] Enhancement
    [ ] Bug

0xc0170, 2019-06-25:
What is defined in your mbed_config file? Looking at the mbed_app.json file above. I dont think `QSPI_FLASH1_IO0` is defined - it would have different name, wouldn't it?

`MBED_CONF_QSPI....` is there? Use `macro_name` to have the specific name

huarunlin, 2019-06-27:
@0xc0170 

> mbed_app.json

            "qspif.QSPI_IO0" : "PE_12",
            "qspif.QSPI_IO1" : "PE_13",
            "qspif.QSPI_IO2" : "PE_14",
            "qspif.QSPI_IO3" : "PE_15",
            "qspif.QSPI_SCK" : "PE_10",
            "qspif.QSPI_CSN" : "PE_11",

> mbed_config.h

            #define MBED_CONF_QSPIF_QSPI_CSN                        PE_11
            #define MBED_CONF_QSPIF_QSPI_FREQ                       40000000
            #define MBED_CONF_QSPIF_QSPI_IO0                        PE_12     
            #define MBED_CONF_QSPIF_QSPI_IO1                        PE_13  
            #define MBED_CONF_QSPIF_QSPI_IO2                        PE_14 
            #define MBED_CONF_QSPIF_QSPI_IO3                        PE_15  
            #define MBED_CONF_QSPIF_QSPI_MIN_PROG_SIZE              1  
            #define MBED_CONF_QSPIF_QSPI_MIN_READ_SIZE              1   
            #define MBED_CONF_QSPIF_QSPI_POLARITY_MODE              0  
            #define MBED_CONF_QSPIF_QSPI_SCK                        PE_11

jeromecoutant, 2019-06-27:
Hi

> Target NUCLEO_L496ZG does not define QSPI_FLASH1_XXX on PinNames.h

Yes, because there is no QSPI on NUCLEO boards...

0xc0170, 2019-06-27:
@huarunlin I think this question is answered, shall be closed

closed 2019-07-03 by 0xc0170