ARMmbed/mbed-os #10876: DeviceKey was not declared for RZ_A1H target

gggekov, 2019-06-20:
### Description
Hello! 
I'm building the cloud-client-example for the [rz_a1h](https://os.mbed.com/platforms/Renesas-GR-PEACH/) target, I'm on the Release candidate for mbed-os-5.13.0-rc3(commit id 92a58dff9960788a3582b0f6dced1d5280b7f2f8). 
The error I'm getting when building on Ubuntu is:
```
Compile [  8.4%]: pal_plat_rot.cpp
Compile: /home/george/Downloads/gcc-arm-none-eabi-8-2018-q4-major/bin/arm-none-eabi-g++ -std=gnu++98 -fno-rtti -Wvla -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -g1 -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-a9 -mthumb-interwork -marm -march=armv7-a -mfpu=vfpv3 -mfloat-abi=hard -mno-unaligned-access -DTARGET_RZA1H -DTARGET_MBRZA1H -D__FPU_PRESENT -DDEVICE_I2C=1 -DDEVICE_EMAC=1 -D__MBED__=1 -DDEVICE_USBDEVICE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTINOUT=1 -DDEVICE_PORTIN=1 -DDEVICE_RTC=1 -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DDEVICE_ETHERNET=1 -DCOMPONENT_SD=1 -D__EVAL -DTOOLCHAIN_GCC -DFEATURE_BOOTLOADER=1 -DTARGET_CORTEX_A -DDEVICE_CAN=1 -DARM_MATH_CA9 -DDEVICE_I2C_ASYNCH=1 -DCOMPONENT_PSA_SRV_EMUL=1 -D__MBED_CMSIS_RTOS_CA9 -DCOMPONENT_PSA_SRV_IMPL=1 -DTARGET_LIKE_CORTEX_A9 -DDEVICE_LPTICKER=1 -DDEVICE_PWMOUT=1 -DTARGET_RZ_A1_EMAC -DTARGET_CORTEX -DDEVICE_I2CSLAVE=1 -DDEVICE_PORTOUT=1 -DMBED_BUILD_TIMESTAMP=1560956233.19 -DTARGET_NAME=RZ_A1H -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_RENESAS -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DCOMPONENT_NSPE=1 -DDEVICE_SERIAL_FC=1 -DFEATURE_STORAGE=1 -DTARGET_A9 -D__CORTEX_A9 -DTARGET_RZ_A1H -DTARGET_RZ_A1XX -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_USTICKER=1 -DDEVICE_SPI=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DDEVICE_FLASH=1 @../BUILD/RZ_A1H/GCC_ARM/.includes_1f243d49f4034147d31a375b733f3129.txt -include ../BUILD/RZ_A1H/GCC_ARM/mbed_config.h -MD -MF ../BUILD/RZ_A1H/GCC_ARM/mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.d -o ../BUILD/RZ_A1H/GCC_ARM/mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.o ../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp
[Error] pal_plat_rot.cpp@67,5: 'DeviceKey' was not declared in this scope
[Error] pal_plat_rot.cpp@67,16: 'devkey' was not declared in this scope
[Error] pal_plat_rot.cpp@67,25: 'DeviceKey' is not a class or namespace
[Error] pal_plat_rot.cpp@81,22: 'DEVICEKEY_ALREADY_EXIST' was not declared in this scope
[DEBUG] Return: 1
[DEBUG] Output: ../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp: In function 'palStatus_t pal_plat_osSetRoT(uint8_t*, size_t)':
[DEBUG] Output: ../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:67:5: error: 'DeviceKey' was not declared in this scope
[DEBUG] Output:      DeviceKey &devkey = DeviceKey::get_instance();
[DEBUG] Output:      ^~~~~~~~~
[DEBUG] Output: ../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:67:16: error: 'devkey' was not declared in this scope
[DEBUG] Output:      DeviceKey &devkey = DeviceKey::get_instance();
[DEBUG] Output:                 ^~~~~~
[DEBUG] Output: ../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:67:16: note: suggested alternative: 'key'
[DEBUG] Output:      DeviceKey &devkey = DeviceKey::get_instance();
[DEBUG] Output:                 ^~~~~~
[DEBUG] Output:                 key
[DEBUG] Output: ../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:67:25: error: 'DeviceKey' is not a class or namespace
[DEBUG] Output:      DeviceKey &devkey = DeviceKey::get_instance();
[DEBUG] Output:                          ^~~~~~~~~
[DEBUG] Output: ../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:81:22: error: 'DEVICEKEY_ALREADY_EXIST' was not declared in this scope
[DEBUG] Output:          if (error == DEVICEKEY_ALREADY_EXIST) {
[DEBUG] Output:                       ^~~~~~~~~~~~~~~~~~~~~~~
[DEBUG] Output: ../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:81:22: note: suggested alternative: 'DEVICEKEY_ENABLED'
[DEBUG] Output:          if (error == DEVICEKEY_ALREADY_EXIST) {
[DEBUG] Output:                       ^~~~~~~~~~~~~~~~~~~~~~~
[DEBUG] Output:                       DEVICEKEY_ENABLED
Traceback (most recent call last):
  File "/home/george/repo/mbed-cloud-client-example/mbed-os/tools/make.py", line 78, in wrapped_build_project
    *args, **kwargs
  File "/home/george/repo/mbed-cloud-client-example/mbed-os/tools/build_api.py", line 598, in build_project
    objects = toolchain.compile_sources(resources, sorted(resources.get_file_paths(FileType.INC_DIR)))
  File "/home/george/repo/mbed-cloud-client-example/mbed-os/tools/toolchains/mbed_toolchain.py", line 458, in compile_sources
    return self.compile_queue(queue, objects)
  File "/home/george/repo/mbed-cloud-client-example/mbed-os/tools/toolchains/mbed_toolchain.py", line 531, in compile_queue
    raise ToolException(err)
ToolException: ../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp: In function 'palStatus_t pal_plat_osSetRoT(uint8_t*, size_t)':
../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:67:5: error: 'DeviceKey' was not declared in this scope
     DeviceKey &devkey = DeviceKey::get_instance();
     ^~~~~~~~~
../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:67:16: error: 'devkey' was not declared in this scope
     DeviceKey &devkey = DeviceKey::get_instance();
                ^~~~~~
../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:67:16: note: suggested alternative: 'key'
     DeviceKey &devkey = DeviceKey::get_instance();
                ^~~~~~
                key
../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:67:25: error: 'DeviceKey' is not a class or namespace
     DeviceKey &devkey = DeviceKey::get_instance();
                         ^~~~~~~~~
../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:81:22: error: 'DEVICEKEY_ALREADY_EXIST' was not declared in this scope
         if (error == DEVICEKEY_ALREADY_EXIST) {
                      ^~~~~~~~~~~~~~~~~~~~~~~
../mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:81:22: note: suggested alternative: 'DEVICEKEY_ENABLED'
         if (error == DEVICEKEY_ALREADY_EXIST) {
                      ^~~~~~~~~~~~~~~~~~~~~~~
                      DEVICEKEY_ENABLED

[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/home/george/repo/mbed-cloud-client-example/mbed-os"
       Command: "/usr/bin/python -u /home/george/repo/mbed-cloud-client-example/mbed-os/tools/make.py -t GCC_ARM -m rz_a1h --source .. --build ../BUILD/RZ_A1H/GCC_ARM -v"
```
Note also that I'm getting the same error regardless whether MBEDTLS has been disabled or enabled with `MBEDTLS_ENTROPY_NV_SEED`

Do you have suggestions how to fix this problem? 

Many thanks! 
George

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ x] Bug

toyowata, 2019-06-21:
@gggekov you can add `FLASHIAP` component in your mbed_app.json to avoid the compile error.

```json
    "target.components_add" : ["FLASHIAP"],
```

gggekov, 2019-06-21:
Thank you @toyowata, your suggestion fixes the problem. However, I get stuck into a different problem - 
```
[Error] pal_plat_rtos.cpp@743,18: 'mbedtls_hardware_poll' was not declared in this scope
Traceback (most recent call last):
  File "/home/george/repo/mbed-cloud-client-example/mbed-os/tools/make.py", line 78, in wrapped_build_project
    *args, **kwargs
  File "/home/george/repo/mbed-cloud-client-example/mbed-os/tools/build_api.py", line 595, in build_project
    objects = toolchain.compile_sources(resources, sorted(resources.get_file_paths(FileType.INC_DIR)))
  File "/home/george/repo/mbed-cloud-client-example/mbed-os/tools/toolchains/mbed_toolchain.py", line 458, in compile_sources
    return self.compile_queue(queue, objects)
  File "/home/george/repo/mbed-cloud-client-example/mbed-os/tools/toolchains/mbed_toolchain.py", line 531, in compile_queue
    raise ToolException(err)
ToolException: In file included from ./mbed-os/features/mbedtls/inc/mbedtls/platform.h:37,
                 from ./mbed-os/features/netsocket/TLSSocketWrapper.h:28,
                 from ./mbed-os/features/netsocket/nsapi.h:43,
                 from ./mbed-os/mbed.h:26,
                 from ./mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/RTOS/pal_plat_rtos.cpp:22:
```

toyowata, 2019-06-22:
@gggekov The RZ/A1H doesn't have TRNG either on-chip or on-board.  So, you have to add below in your mbed_app.json.

```json
            "target.macros_add"                         : ["MBEDTLS_TEST_NULL_ENTROPY", "MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES"],
```

You can also refer this example.

https://os.mbed.com/teams/Renesas/code/pelion-example-common/file/5ab220a2ac08/mbed_app.json/

0xc0170, 2019-06-24:
@gggekov Does this fixes your issue?

gggekov, 2019-06-24:
Hello @0xc0170 @toyowata ! 
I'm still getting an error:
```
Compile [  8.5%]: pal_plat_rot.cpp
[Error] pal_plat_rot.cpp@67,5: 'DeviceKey' was not declared in this scope
[Error] pal_plat_rot.cpp@67,16: 'devkey' was not declared in this scope
[Error] pal_plat_rot.cpp@67,25: 'DeviceKey' is not a class or namespace
[Error] pal_plat_rot.cpp@81,22: 'DEVICEKEY_ALREADY_EXIST' was not declared in this scope
[ERROR] ./mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp: In function 'palStatus_t pal_plat_osSetRoT(uint8_t*, size_t)':
./mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:67:5: error: 'DeviceKey' was not declared in this scope
     DeviceKey &devkey = DeviceKey::get_instance();
     ^~~~~~~~~
./mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:67:16: error: 'devkey' was not declared in this scope
     DeviceKey &devkey = DeviceKey::get_instance();
                ^~~~~~
./mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:67:25: error: 'DeviceKey' is not a class or namespace
     DeviceKey &devkey = DeviceKey::get_instance();
                         ^~~~~~~~~
./mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/ROT/pal_plat_rot.cpp:81:22: error: 'DEVICEKEY_ALREADY_EXIST' was not declared in this scope
         if (error == DEVICEKEY_ALREADY_EXIST) {
                      ^~~~~~~~~~~~~~~~~~~~~~~

[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/home/george/repo/mbed-cloud-client-example"
       Command: "/usr/bin/python -u /home/george/repo/mbed-cloud-client-example/mbed-os/tools/make.py -t GCC_ARM -m rz_a1h --source . --build ./BUILD/RZ_A1H/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output

```
My `mbed_app.json` begins as
```
{
    "macros": [
        "MBED_BOOTLOADER_SIZE=(32*1024)",
        "ARM_UC_USE_PAL_BLOCKDEVICE=1"
    ],
    "target_overrides": {
        "*": {
            "target.features_add"                       : ["BOOTLOADER", "STORAGE"],
            "target.macros_add"                         : ["FLASHIAP","MBEDTLS_TEST_NULL_ENTROPY", "MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES"],
            "platform.stdio-baud-rate"                  : 115200,
            "platform.stdio-convert-newlines"           : true,
            "platform.stdio-buffered-serial"            : true,
            "platform.stdio-flush-at-exit"              : true,
            "rtos.main-thread-stack-size"               : 5120,
            "rtos.timer-thread-stack-size"              : 256,
            "rtos.idle-thread-stack-size"               : 256,
            "client_app.platform-enable-button"         : 0,
            "client_app.platform-enable-led"            : 0,
            "mbed-client-pal.pal-dns-api-version"       : 2,
            "mbed-cloud-client.external-sst-support"    : 1,
            "update-client.storage-locations"           : 1,
            "mbed-trace.enable"                         : null,
            "events.shared-stacksize"                   : 2048,
            "nsapi.default-wifi-security"               : "WPA_WPA2",
            "nsapi.default-wifi-ssid"                   : "\"SSID\"",
            "nsapi.default-wifi-password"               : "\"Password\""
        },

```
Did I misinterpret your instructions? I haven't done any other changes to the project. 
Thanks,
George

toyowata, 2019-06-24:
@gggekov The `FLASHAIP` should be components, not macros.

```
    "target.components_add" : ["FLASHIAP"],
```

gggekov, 2019-06-25:
Hi @toyowata,
I've added `FLASHIAP` in `target.components_add` and left the `MBEDTLS_TEST_NULL_ENTROPY` and `MBEDTLS_NO_DEFAULT_ENTROPY_SOURCES` macros. However, I am still getting the following error:
```
Compile [  8.6%]: pal_plat_rtos.cpp
[Warning] config.h@3438,2: #warning "MBEDTLS_TEST_NULL_ENTROPY has been enabled. This " "configuration is not secure and is not suitable for production use" [-Wcpp]
[Error] pal_plat_rtos.cpp@743,116: 'mbedtls_hardware_poll' was not declared in this scope
[ERROR] In file included from ./mbed-os/features/mbedtls/inc/mbedtls/platform.h:37:0,
                 from ./mbed-os/features/netsocket/TLSSocketWrapper.h:28,
                 from ./mbed-os/features/netsocket/nsapi.h:43,
                 from ./mbed-os/mbed.h:26,
                 from ./mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/RTOS/pal_plat_rtos.cpp:22:
./mbed-os/features/mbedtls/inc/mbedtls/config.h:3438:2: warning: #warning "MBEDTLS_TEST_NULL_ENTROPY has been enabled. This " "configuration is not secure and is not suitable for production use" [-Wcpp]
 #warning "MBEDTLS_TEST_NULL_ENTROPY has been enabled. This " \
  ^~~~~~~
./mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/RTOS/pal_plat_rtos.cpp: In function 'palStatus_t pal_plat_osRandomBuffer(uint8_t*, size_t, size_t*)':
./mbed-cloud-client/mbed-client-pal/Source/Port/Reference-Impl/OS_Specific/mbedOS/RTOS/pal_plat_rtos.cpp:743:116: error: 'mbedtls_hardware_poll' was not declared in this scope
     platStatus = mbedtls_hardware_poll(NULL /*Not used by the function*/, randomBuf, bufSizeBytes, &actualOutputLen);
                                                                                                                    ^

[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
```
Is there anything else that can be tried?

toyowata, 2019-06-25:
@gggekov Latest version of the `mbed-cloud-client-example` does not like configs for previous version.  I have quickly made configs for the RZ_A1H target as below:

```json
        "RZ_A1H": {
            "client_app.mbedtls-user-config-file"       : "\"mbedTLSConfig_mbedOS_SW_TRNG_PSA.h\"",
            "client_app.pal-user-defined-configuration" : "\"pal_config_non_trng_MbedOS.h\"",
            "target.extra_labels_add"                   : ["PSA"],
            "target.macros_add"                         : ["MBEDTLS_PSA_CRYPTO_C", "MBEDTLS_ENTROPY_NV_SEED"],
            "target.macros_remove"                      : ["MBEDTLS_CONFIG_HW_SUPPORT"],
            "target.components_add"                     : ["SD", "FLASHIAP"],
            "target.network-default-interface-type"     : "ETHERNET",
            "update-client.application-details"         : "(MBED_CONF_APP_FLASH_START_ADDRESS + 64*1024)",
            "update-client.storage-address"             : "(64*1024*1024)",
            "update-client.storage-size"                : "(2*1024*1024)",
            "update-client.storage-locations"           : "1",
            "target.bootloader_img"                     : "bootloader/mbed-bootloader-RZ_A1H.bin",
            "target.header_offset"                      : "0x10000",
            "target.app_offset"                         : "0x10400"
        }
```

However, I got errors below ...

```
Compile [ 14.7%]: arm_uc_pal_delta_paal_original_reader.c
[Error] arm_uc_pal_delta_paal_original_reader.c@29,49: 'APPLICATION_ADDR' undeclared (first use in this function)
[ERROR] ./mbed-cloud-client/update-client-hub/modules/delta-paal/source/arm_uc_pal_delta_paal_original_reader.c: In function 'arm_uc_deltapaal_original_reader':
./mbed-cloud-client/update-client-hub/modules/delta-paal/source/arm_uc_pal_delta_paal_original_reader.c:29:49: error: 'APPLICATION_ADDR' undeclared (first use in this function)
 #define MBED_CONF_APP_APPLICATION_START_ADDRESS APPLICATION_ADDR
                                                 ^
./mbed-cloud-client/update-client-hub/modules/delta-paal/source/arm_uc_pal_delta_paal_original_reader.c:67:25: note: in expansion of macro 'MBED_CONF_APP_APPLICATION_START_ADDRESS'
     uint32_t appStart = MBED_CONF_APP_APPLICATION_START_ADDRESS;
                         ^
./mbed-cloud-client/update-client-hub/modules/delta-paal/source/arm_uc_pal_delta_paal_original_reader.c:29:49: note: each undeclared identifier is reported only once for each function it appears in
 #define MBED_CONF_APP_APPLICATION_START_ADDRESS APPLICATION_ADDR
                                                 ^
./mbed-cloud-client/update-client-hub/modules/delta-paal/source/arm_uc_pal_delta_paal_original_reader.c:67:25: note: in expansion of macro 'MBED_CONF_APP_APPLICATION_START_ADDRESS'
     uint32_t appStart = MBED_CONF_APP_APPLICATION_START_ADDRESS;
                         ^

[mbed] ERROR: "/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python" returned error.
       Code: 1
       Path: "/Users/toyowata/dev/mbed-cloud-client-example"
       Command: "/Library/Frameworks/Python.framework/Versions/2.7/Resources/Python.app/Contents/MacOS/Python -u /Users/toyowata/dev/mbed-cloud-client-example/mbed-os/tools/make.py -t gcc_arm -m rz_a1h --source . --build ./BUILD/RZ_A1H/GCC_ARM"
       Tip: You could retry the last command with "-v" flag for verbose output
---

```

I guess there is missing information in the `mbed-os/tools/arm_package_manager/index.json` for the target (there is no `sectors` info for `ROM regions bootloader`).

toyowata, 2019-06-27:
cc @ARMmbed/team-renesas

d-kato, 2019-06-28:
@gggekov, @toyowata, Hello. How about rewriting ``R7S72100`` of ``mbed-os/tools/arm_package_manager/index.json`` as follows.

``` json
    "R7S72100": {
        "algorithms": [], 
        "family": "RZ_A",
        "from_pack": {
            "pack": "RZ_DFP", 
            "url": "http://www.keil.com/pack/", 
            "vendor": "Keil", 
            "version": "1.2.1"
        }, 
        "memories": {
            "PROGRAM_FLASH": {
                "access": {
                    "execute": true, 
                    "non_secure": false, 
                    "non_secure_callable": false, 
                    "peripheral": false, 
                    "read": true, 
                    "secure": false, 
                    "write": false
                }, 
                "default": true, 
                "size": 8388608, 
                "start": 402653184, 
                "startup": true
            }
        },
        "name": "R7S72100", 
        "processor": {
            "Symmetric": {
                "core": "CortexA9", 
                "fpu": "DoublePrecision", 
                "mpu": "Present", 
                "units": 1
            }
        }, 
        "sectors": [
            [
                0, 
                4096
            ]
        ], 
        "sub_family": "RZ_A1H",
        "vendor": "Renesas:117"
    }, 
```

toyowata, 2019-07-05:
@d-kato Thank you very much for this patch.  Can you make PR for this?

@gggekov I succeeded to build/run the example on rz_a1h target with configuration below:

```json
{
    "macros": [
        "ARM_UC_USE_PAL_BLOCKDEVICE=1",
        "MBED_CLOUD_CLIENT_STL_API=0",
        "MBED_CLOUD_CLIENT_STD_NAMESPACE_POLLUTION=1"
    ],
    "target_overrides": {
        "*": {
            "target.features_add"               : ["BOOTLOADER", "STORAGE"],
            "platform.stdio-baud-rate"          : 115200,
            "platform.stdio-convert-newlines"   : true,
            "platform.stdio-buffered-serial"    : true,
            "platform.stdio-flush-at-exit"      : true,
            "rtos.main-thread-stack-size"       : 5120,
            "mbed-client.event-loop-size"       : 1024,
            "nanostack-hal.event_loop_thread_stack_size": 8192,
            "update-client.storage-address"     : "(1024*1024*64)",
            "update-client.storage-size"        : "(1024*1024*2)",
            "update-client.storage-locations"   : 1,
            "mbed-trace.enable": null
        },
        "RZ_A1H": {
            "target.components_add"                     : ["SD", "FLASHIAP"],
            "target.macros_add"                         : ["MBEDTLS_ENTROPY_NV_SEED"],
            "target.macros_remove"                      : ["MBEDTLS_CONFIG_HW_SUPPORT"],
            "target.network-default-interface-type"     : "ETHERNET",
            "target.features_add"                       : ["BOOTLOADER"],
            "target.bootloader_img"                     : "tools/mbed-bootloader-RZ_A1H.bin",
            "target.header_offset"                      : "0x10000",
            "target.app_offset"                         : "0x10400",
            "client_app.mbedtls-user-config-file"       : "\"mbedTLSConfig_mbedOS_SW_TRNG.h\"",
            "client_app.pal-user-defined-configuration" : "\"pal_config_legacy_non_trng_MbedOS.h\"",
            "client_app.sotp-section-1-address"         : "(0x18000000 + (8*1024*1024) - 2*(4*1024))",
            "client_app.sotp-section-1-size"            : "(4*1024)",
            "client_app.sotp-section-2-address"         : "(0x18000000 + (8*1024*1024) - 1*(4*1024))",
            "client_app.sotp-section-2-size"            : "(4*1024)",
            "update-client.bootloader-details"          : "0x1800CBCC",
            "update-client.application-details"         : "(0x18000000 + 64*1024)",
            "mbed-cloud-client.external-sst-support"    : null,
            "mbed-cloud-client.update-storage"          : "ARM_UCP_FLASHIAP_BLOCKDEVICE",
            "mbed-client-pal.pal-dns-api-version"       : 0
        }
    },
    "config": {
        "developer-mode": {
            "help"      : "Enable Developer mode to skip Factory enrollment",
            "options"   : [null, 1],
            "value"     : 1
        },
        "button-pinname": {
            "help"      : "PinName for button.",
            "value"     : "BUTTON1"
        },
        "led-pinname"   : {
            "help"      : "PinName for led, which is attached to led blink resource.",
            "value"     : "LED_RED"
        }
    }
}
```

```
$ mbed compile -m rz_a1h -t gcc_arm
```

0xc0170, 2019-08-13:
Fixed via 10876

closed 2019-08-13 by 0xc0170