ARMmbed/mbed-os #10880: VREFINT_CAL_ADDR wrong address for MTB_RAK811

eskan, 2019-06-20:
### Description

target : MTB_RAK811
toolchain: GCC_ARM

VREFINT_CAL_ADDR is resolved to a wrong address that result in a segfault. The correct address (working address) seems to be 0x1FF80078

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-06-24:
@eskan Can you send PR fixing this?

cc @ARMmbed/team-multitech

eskan, 2019-06-24:
unfortunately i've just override the address in my code, i'm not able to understand the inherits chain and what target is concerned and even if its the right chain :/ 
We need to find a STM32L1 expert or a RAK811 one. 
@ithinuel, can you help about this ?

ithinuel, 2019-07-19:
Hi,  
I did a quick check and it seems like st's SDK : `device/stm32l1xx_ll_adc.h:343` sets it to `((uint16_t*) (0x1FF800F8U))` while the [datasheet](https://www.st.com/resource/en/datasheet/stm32l151c6-a.pdf) for the `stm32l151xB-A` shows indeed `VREFINT_CAL` at `0x1FF8 0078-0x1FF8 0079`.

@ARMmbed/team-st-mcd What would be the plan to address that issue ?

jeromecoutant, 2019-07-19:
Hi

I agree, value in stm32l1xx_ll_adc.h is the value for STM32L152...
which seems not compatible with STM32L151

Could you set this issue as "devices: st" ?

Thx

jeromecoutant, 2019-07-19:
ST_INTERNAL_REF 69852

jeromecoutant, 2019-10-03:
@eskan Please check #11621

PR #11621 (2019-10-14): STM32L151: update calibration memory address
### Description

Fix #10880 

Memory address for some calibration values depends on targets.

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ] Target update
    [ ] Functionality change
    [ ] Docs update
    [ ] Test update
    [ ] Breaking change

### Reviewers

### Release Notes
closed 2019-10-14 by 0xc0170