ARMmbed/mbed-os #10886: rtc_reset test DELAY_TOLERANCE value modification possibility

romanjoe, 2019-06-24:
### Description

I'm Cypress Semiconductor engineer working on BSP for new Cypress target to add in mbed-os. 
New target would have a mandatory custom bootloader, that enables pretty heavy security functionality.
Bootloader execution time takes up to 2 seconds and it has no RTC functionality enabled.

### Issue request type
Existing DELAY_TOLERANCE set to 1 second for tests-hots_tests-rct_reset leads to continuous test fail with error message like "DELAY_TIME set to 5 seconds, but 3 seconds received from target" 

Is it possible to modify original test, by changing DELAY_TOLERANCE to 2 second or add possibility to pass this argument as a parameter at test execution?

    [ ] Question
    [X] Enhancement
    [ ] Bug

0xc0170, 2019-06-24:
@mprse Can you review?

romanjoe, 2019-06-27:
Hi, we managed to overcome this issue internally.

Thank you. This can be closed.

closed 2019-06-27 by romanjoe