ARMmbed/mbed-os #10887: TFM cannot be compiled with GCC 6

jeromecoutant, 2019-06-24:
### Description

TFM cannot be compiled with GCC version 6:
(GNU Tools for ARM Embedded Processors 6-2017-q2-update)

````
Compile [  4.2%]: tfm_nspm.c
C:\xxx\AppData\Local\Temp\cc353dof.s: Assembler messages:
C:\xxx\AppData\Local\Temp\cc353dof.s:566: Error: branch out of range
[DEBUG] Return: 1
````

Setting GCC version 7 solves the issue.
(GNU Tools for Arm Embedded Processors 7-2018-q2-update)

````
Compile [  6.5%]: tfm_nspm.c
[DEBUG] Return: 0
````

Issue requested by @devran01 in order to follow the point.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

jeromecoutant, 2019-06-24:
Reference: https://git.trustedfirmware.org/trusted-firmware-m.git/tree/docs/user_guides/tfm_sw_requirement.rst#n124

The currently supported compiler versions are:
- GNU Arm compiler v6.3.1
- GNU Arm compiler v7.3

0xc0170, 2019-06-24:
Strange, should work according to the requirements but fails

```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
```

6-2017-q2 - gcc 6.3.1. 

@jeromecoutant is this issue also in the upstream repository or just ours (our import issue) ?

jeromecoutant, 2019-06-24:
Hi
I have only mbed environment,
I don't have any "pure" TFM.

Patater, 2019-07-04:
Looking at [the latest version of `tfm_nspm.c` in TF-M upstream](https://git.trustedfirmware.org/trusted-firmware-m.git/tree/secure_fw/core/tfm_nspm.c?id=7619ee04427925959e5500c17b77d7c590e98785), there are only minor differences with what we have in Mbed OS. I don't see why this problem wouldn't also exist in upstream TF-M.

Please raise an issue at https://developer.trustedfirmware.org

jeromecoutant, 2019-07-04:
> Please raise an issue at https://developer.trustedfirmware.org

I can't know if compile issue comes from MBED port or if it is pure TFM
So, I will not raise any ticket...
And to be honest, GCC6 issue is a bit outdated...

devran01, 2019-07-11:
Hi @jeromecoutant, @Patater and I looked into this issue and it is a GCC6.3 compiler issue. It looks like the compiler is generating a branch instruction before expanding CMSE related instructions which is causing the assembler to fail with "branch out of range". We'd recommend switching to GCC7 or higher while compiling the secure side.

kjbracey-arm, 2019-12-19:
I believe this can be closed - we now use GCC 9.

closed 2019-12-19 by kjbracey-arm