ARMmbed/mbed-os #10890: NRF52840: Custom bootloader hangs when calling mbed_start_application on a re-flash of new firmware.

yogeshk19, 2019-06-24:
Description
Target : NRF52840
Toolchain : GCC_ARM
mbed-cli
Mbed OS 5.12.4

I have built a custom UART/BLE bootloader for my MBED BLE Application on NRF52840. I have been using the Flash API to flash the new firmware onto the flash. 

1) In the first case I used Nordic tools to erase the NRF52840. Install the custom bootloader at the start of the ROM and the bootloader listens for incoming firmware packets. Once the Firmware is flashed, I call the  mbed_start_application API call to jump to the application and it works fine. 

2) In the second case I have an existing firmware but I am overwriting the firmware in place and once the firmware is flashed successfully and the Flash API indicate no error while flashing the firmware when I call the mbed_start_application API call to jump to the new installed application the bootloader seems to hang at this point. I am not sure how to troubleshoot this further and every time I am writing the application to the same location on the flash.

I am hoping to get some urgent help on this issue as we are looking to go production end of this month and this issue is blocking us currently.

Issue request type
[ ] Question
[] Enhancement
[X] Bug

Thanks,
Yogesh

loverdeg-ep, 2019-06-24:
@trowbridgec @AGlass0fMilk

loverdeg-ep, 2019-06-24:
Can we assume you are not using softdevice?

Are you using cordio?

yogeshk19, 2019-06-24:
Yes I am not using Softdevice.

Here is my Bootloader mbed_app.json

`{
     "macros": [
        "MBED_CONF_APP_MAIN_STACK_SIZE=6000"
    ],
    "target_overrides": {
        "NRF52840_DK": {
            "target.features_add": ["BLE"],
            "target.extra_labels_add": ["CORDIO", "CORDIO_LL", "SOFTDEVICE_NONE", "NORDIC_CORDIO"],
            "target.features_remove": ["CRYPTOCELL310"],
            "target.macros_remove": ["MBEDTLS_CONFIG_HW_SUPPORT"],
            "target.extra_labels_remove": ["SOFTDEVICE_COMMON", "SOFTDEVICE_S140_FULL", "NORDIC_SOFTDEVICE"],
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200,
            "target.console-uart-flow-control": null,
            "target.restrict_size" : "0x‭40000‬",
            "target.OUTPUT_EXT": "hex",
            "drivers.uart-serial-txbuf-size" : 1024,
            "drivers.uart-serial-rxbuf-size" : 1024
        }
    }
}`

Here is my application JSON.
`{
    "target_overrides": {
        "NRF52840_DK": {
            "target.features_add": ["BLE"],
            "target.extra_labels_add": ["CORDIO", "CORDIO_LL", "SOFTDEVICE_NONE", "NORDIC_CORDIO"],
            "target.features_remove": ["CRYPTOCELL310"],
            "target.macros_remove": ["MBEDTLS_CONFIG_HW_SUPPORT"],
            "target.extra_labels_remove": ["SOFTDEVICE_COMMON", "SOFTDEVICE_S140_FULL", "NORDIC_SOFTDEVICE"],
            "platform.stdio-baud-rate": 115200,
            "platform.default-serial-baud-rate": 115200,
            "target.console-uart-flow-control": null,
            "target.bootloader_img" : "..\\mTestBootloader\\BUILD\\NRF52840_DK\\GCC_ARM\\mTestBootloader.hex",
            "target.header_offset" : "0x‭40000‬",
            "target.app_offset" : "0x40400"
        }
    }
}`

Thanks,
Yogesh

yogeshk19, 2019-06-25:
@loverdeg-ep  It is failing when erasing sector with error code -1. Not sure what that error specifically means. So my application is flashed at the address : 0x40400, and it errors out when trying to erase the 4kb sector, starting at that address. There was an issue #6140 that was fixed, but I am not sure if this is because of a similar issue on NRF52840 module as well or the fix addresses this on the Nordic Targets as well? 

Thanks,
Yogesh

yogeshk19, 2019-06-25:
@0xc0170, @davidsaada, @LiyouZhou 
I am wondering if the flash erase issue I am facing currently on NRF52840 which is returning an error code of -1 from no matter which memory address I pick. I was hoping because I was looking at the Nordic documentation, it talks about the erase should align with Page Boundaries. Is the fix made for issue #6140 not supposed to address this target as well? I was hoping one of you had an insight into this issue I am facing when erasing the flash where existing firmware has been installed and programing the same memory location with the new firmware. Any thoughts can help me get unblocked, since we are in a critical stage of our project.

Thanks,
Yogesh

yogeshk19, 2019-06-25:
Is there an escalation path with the mbed os team when you are blocked?

trowbridgec, 2019-06-25:
@yogeshk19 

Maybe looking at the mbed-bootloader as a template could provide some insight? The mbed-bootloader uses functions called [`eraseActiveFirmware()`](https://github.com/ARMmbed/mbed-bootloader/blob/master/source/active_application.cpp#L187) to clear out the internal flash and [`writeActiveFirmware()`](https://github.com/ARMmbed/mbed-bootloader/blob/master/source/active_application.cpp#L187) to write the new firmware to the internal flash.

loverdeg-ep, 2019-06-25:
@yogeshk19 Mbed OS is now maintained in Cambridge UK.

I am on eastern US time and tend to check my github notifications around 6:00AM Eastern US in order to respond to the team and still receive responses back before they finish up for the day.

AGlass0fMilk, 2019-06-25:
@yogeshk19 It's hard to debug something like this without any code or being able to step through the code. There are a multitude of things that can go wrong when reprogramming the chip.

When you say "hang", what do you mean? Can you use a JLink or other debugger to view the registers when the error occurs? Look for something like this:
 
[image]

Check the IPSR register to see if the processor is hard faulting. This could indicate a memory alignment or other program issue.

A few ideas:
- As a stop gap, can you simply try triggering a chip reset through the NVIC? Typically can be done with `NVIC_SystemReset();` This should clear up anything the bootloader was doing with RAM, etc. You may already be doing this after reprogramming.
- You should strictly keep in mind memory alignment when doing in-field updates. All executable code must be word-aligned.
- Try exporting your code to an IDE that allows you to step through code and debug exactly what's happening.

yogeshk19, 2019-06-26:
@trowbridgec Thanks for pointing me to the mbed-bootloader code base. I am pretty much doing the same as what is being done in the code base to erase the flash. However in my case whether I erase the flash based either on sector size or page size it fails with an error of -1 or rather FlashIAP erase call returns -1.  I am not sure if there is an issue because my start application address is at 0x40400. I am not sure how you determine if this falls on the Page boundary for the NRF52840 BLE Module?

@AGlass0fMilk : The reason I was calling it hangs at mbed-start-application call, is because I have code before and after the call and neither does the main application run nor is the code after that mbed-start-application call is executed. When it does work in the non re-flash scenario, everything before and after the call is executed and the main application is invoked too. Having said that I will try and get all the information you have mentioned I could collect. I will share my code base as soon as I get my client to okay my sharing the code base. But mostly the code base mimics the mbed-bootloader [eraseActiveFirmware()](https://github.com/ARMmbed/mbed-bootloader/blob/master/source/active_application.cpp#L187) that @trowbridgec called out earlier.

Thanks,
Yogesh

yogeshk19, 2019-06-28:
This is resolved. The address I was using where to flash my firmware doesn't fall on a page boundary. It needs to be a multiple of 4kb.

closed 2019-06-28 by yogeshk19