ARMmbed/mbed-os #10891: Cannot use Maxim Feather with mbed-os > 5.10

nick-atmosphereiot, 2019-06-24:
### Description
I don't think that the Maxim Feather 32630 board can be used with mbed-os versions > 5.10. 

It seems as though I'm stuck in a position where all of the functions that used to work for advertising have been deprecated in 5.11, and none of the newer ones have been implemented for this board.

The main problem function is the `setAdvertisingData` function, which is no longer able to be used. The newer function to set the payload with the advertising data builder is not supported.

Is there a way around this besides just sticking with 5.10?

### Issue request type
    [ ] Question
    [ ] Enhancement
    [ X] Bug

0xc0170, 2019-06-27:
@bentcooke @ARMmbed/team-maximintegrated Please review

0xc0170, 2019-06-27:
cc @ARMmbed/mbed-os-pan

0xc0170, 2019-08-13:
@nick-atmosphereiot have you made any progress ?

cc @maclobdell

idea--list, 2020-09-10:
Well... the hardware specs of the MAX32630FTHR look still really great. Also i thought it would be a perfect fit for prototyping.
But for some reason it seems as if Maxim could not keep up maintaining software support for their boards.
Mbed-os V5.9.7 is the last version that runs as expected on this board.

As mbed-os evolved, i had to face more and more issues with this board: first their USB stack got broken, then also their BLE stack stopped working. Then back in February i had to discover #12493 , which really turned me down. There was some hope as someone from Maxim promised they will fix both stacks in some weeks. But even several months later there is no status update at least.

idea--list, 2020-10-04:
For anyone intereted, it seems i found solution for #12493 at least so at least timing issues should be gone when using Mbed OS 5.15 or above versions.

nick-atmosphereiot, 2020-10-18:
Please close this issue. I have no further information and I have moved on.

Nick

0xc0170, 2020-10-19:
Thanks for the update. I'll close this.

closed 2020-10-19 by 0xc0170