ARMmbed/mbed-os #10892: CY8CKIT_062_4343W does not support bootloader?

JanneKiiskila, 2019-06-24:
### Description

Cypress board does not support bootloader according to `targets.json`, is there a special reason for that or should it be enabled? Without bootloader support is it not possible to enable Pelion Device Management Firmware update -feature.

Ref:
https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L8332-L8354, especially line https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L8354.
- We can see target `CY8CKIT_062_WIFI_BT_PSA` is supporting bootloader.
- However, for target [`CY8CKIT_062_4343W`](https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L8390-L8404) bootloader is not enabled.

### Issue request type
    [x] Question
    [ ] Enhancement
    [ ] Bug

Attn: @ARMmbed/team-cypress @maclobdell  @MarceloSalazar

JanneKiiskila, 2019-06-25:
The lack of bootloader will disable generation of APPLICATION_ADDR, which in turn will then block the firmware update from getting compiled.

maclobdell, 2019-06-26:
do you mean "CY8CPROTO" instead of "CY8CKIT"?

I found that with these lines added to the CY8CPROTO_062_4343W configuration in targets.json, then a bootloader and application can be built. 

``` 
        "bootloader_supported" : true,
        "mbed_rom_start" : "0x10002000",
        "mbed_rom_size" : "0x1F6000",
        "sectors": [[268435456, 512]]		
```

After more testing, I can make a PR to be evaluated by Cypress.

```
[BOOT] Mbed Bootloader
[BOOT] ARM: 0000000000000000000000000000000000000000
[BOOT] OEM: 0000000000000000000000000000000000000000
[BOOT] Layout: 0 1000B398
[BOOT] Active firmware integrity check:
[BOOT] [++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++]
[BOOT] SHA256: E7C77214A29ECDC3A08C16F8F5E51CBB3FEE758871591E60E4A2099ECCF1F955
[BOOT] Version: 1561569274
[ERR ] Couldn't find valid DeviceKey - err: -2130771705
[BOOT] Slot 0 is empty
[BOOT] Active firmware up-to-date
[BOOT] Application's start address: 0x1000F400
```

Sunilhegde84, 2019-07-10:
Hi can you pls share your repo , I have tried with your suggestion for CY8CPROTO_062_4343W. But unsuccessful so far . Thank you for your help

maclobdell, 2019-07-24:
I am re-opening because we discovered a bootloader incompatibility with Cypress targets in 5.13.1.  

In the past, a separate hex file with a Cortex M0+ image was selectively added by the build tools depending on if building with a bootloader or not.  However, as of 5.13.1, a new method (area allocated in the linker/scatter files) is used to include the cortex M0+ image.  

This presents a problem for merging a bootloader with an application.  As is, if they both try to pre-pend the Cortex M0+ image, then the M4 application start address will be off, and the SHA is not calculated correctly.  Not to mention the extra 8K of wasted space.  

The fix is to selectively exclude the M0+ image depending on whether a user is building an application that is to include a bootloader. 

A simple hack to get it working is to build the bootloader as normal, but in the application project modify the linker/scatter to change these defines and then remove the  cy_m0p_image section completely. 
     FLASH_CM4_SIZE   = LENGTH(flash);
     FLASH_CM4_START  = ORIGIN(flash);
  
But we need a real solution urgently.   Some logic needs to be implemented.  Such as this for example which selectively includes a header: 
https://github.com/ARMmbed/mbed-os/pull/11041/files#diff-65cd7ba907df332c14810aca2f778982

vmedcy, 2019-07-24:
Hi @maclobdell,

I am responsible for both old (https://github.com/ARMmbed/mbed-os/commit/a48ee113eaf8fd5919f1556e775645b8e8e2bcc1) and new (https://github.com/ARMmbed/mbed-os/commit/c9105eb068bd2082d6d581143c05d0ff9f414437) implementations of the CM0+ prebuilt image merging for Cypress PSoC6 targets.

Indeed, in Mbed OS from 5.11.4 to 5.13.0, the prebuilt CM0+ HEX image selection was handled by the `hex_filename` key in the targets.json. Custom targets can specify custom hex filename by overriding the attribute (this was implemented for PSA targets). For targets without hex_filename attribute, no image merging happened.

Since Mbed OS 5.13.1, the prebuilt HEX images were replaced by C files with array placed to the combined ELF executable by the linker script during regular CM4 application build flow (no python hooks involved). This avoids a need to execute post-build commands, and greatly improves compatibility with third-party build environments.

As you highlighted, it is important that no more that one array with prebuilt CM0+ image should be passed to the linker, otherwise the linker script will merge both variables contiguously into a single section - this is not an intended behavior (however, note that the CM4 CPU still executes at a fixed address). Since the images are placed into [TARGET_CM0P_SLEEP](https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.13.1/targets/TARGET_Cypress/TARGET_PSOC6/psoc6cm0p) directory, each PSoC6 target selects the image by extra label in [targets.json](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.13.1/targets/targets.json#L8373). In case when `CM0P_SLEEP` label is not defined for the target, no section is placed to the area dedicated to the CM0+ executable image in the linker scripts, hence only CM4 application code is programmed to internal flash.

My recommendation for PSOC6 targets that do not need the CM0+ prebuilt image merging capability is to exclude `CM0P_SLEEP` from the target labels in mbed_app.json (I might have screwed the syntax since I hate JSON):

```json
{
    "target_overrides": {
        "CY8CPROTO_062_4343W": {
            "target.extra_labels_remove": ["CM0P_SLEEP"]
      }
}
```

Let me know is this solution works for bootloader use case.

maclobdell, 2019-07-26:
@vmedcy  - Thanks for this information.  This is very helpful.  

Using a very simple application (blinky) and pre-compiled bootloader (mbed-bootloader), I built it with the following configurations in mbed_app.json.  

```
        "CY8CPROTO_062_4343W": {
            "target.bootloader_img"                     : "mbed-bootloader_cy8cproto_big.hex",			
            "target.header_offset"                      : "0xE000",
            "target.app_offset"                         : "0xE400",
            "target.bootloader_supported"               : true,
            "target.mbed_rom_start"                     : "0x10000000",
            "target.mbed_rom_size"                      : "0x00200000",
            "target.sectors"                            : [[268435456, 512]],
            "target.extra_labels_remove"                : ["CM0P_SLEEP"]
        }
```

The output hex file looks correct.  With this define added, each section is in the right place.

However, the SHA calculated by the bootloader doesn't match the SHA calculated by the build tools.  More investigation is needed.

```
[BOOT] event: 0
[BOOT] event: 0
[BOOT] Mbed Bootloader
[BOOT] ARM: 0000000000000000000000000000000000000000
[BOOT] OEM: 0000000000000000000000000000000000000000
[BOOT] Layout: 0 1000BF9C
[BOOT] Active firmware integrity check:
[BOOT] checkActiveApplication
[DBG ] readActiveFirmwareHeader
[BOOT] event: 6
[BOOT] header start: 0x1000E000
[BOOT] app start: 0x1000E400
[BOOT] app size: 71704
[BOOT] [++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++]
[BOOT] SHA256: 9927389D98EECDD8B966302AF9EFF1B381A512821E456F1CE10FB94C15FEA899
[BOOT] SHA256: C8B110AF76DA3E875B8EB62B4BB26BE5CC7093D7A8E4FC3FCE2D0ED66E4971EF
[BOOT] SHA Compare: FAIL

[BOOT] SHA256: 9927389D98EECDD8B966302AF9EFF1B381A512821E456F1CE10FB94C15FEA899
[BOOT] SHA256: C8B110AF76DA3E875B8EB62B4BB26BE5CC7093D7A8E4FC3FCE2D0ED66E4971EF
[DBG ] heapVersion: 1564142815
[DBG ] bootCounter: 0
[ERR ] Active firmware integrity check failed

```

maclobdell, 2019-07-26:
After further investigation,  I believe there is still a problem.  The hex file output is actually not correct.

If I use the above configuration with a bootloader file added, the header is expected to be at memory address 0x1000E000 and the application at 0x1000E400.  The build tools will set MBED_APP_START = 0x1000E400.

In the linker file, I see that the start of flash is set to MBED_APP_START

```
 flash             (rx)    : ORIGIN = MBED_APP_START
```
And the start of application (for Cortex M4) is calculated to be MBED_APP_START + 0x2000.

```
/* Size and start address of the Cortex-M0+ application image */
FLASH_CM0P_SIZE  = 0x2000;
FLASH_CM0P_START = ORIGIN(flash);
/* Size and start address of the Cortex-M4 application image */
FLASH_CM4_SIZE   = LENGTH(flash) - FLASH_CM0P_SIZE;
FLASH_CM4_START  = FLASH_CM0P_START + FLASH_CM0P_SIZE;
```

So in the binary, the M4 application code actually starts at 0x10400 instead of 0xE400.  This throws off the SHA calculation.

```
/* Cortex-M4 application image */
    .text FLASH_CM4_START :
    {
...
    } > flash
```

Is it possible to preserve the start of the M4 image, even if the M0+ image is not being added?

maclobdell, 2019-09-11:
Mbed OS 5.14 adds bootloader support, however upon further testing, it still seems to be broken.  There is a new section which the Mbed merging tools do not handle, which seems to interfere with the calculation of the hash.

maclobdell, 2019-09-25:
Update: Cypress pushed an update into Mbed OS 5.14 that relies on supervisory flash section that is non-contiguous with the rest of flash.  This new section stores meta data for the Quad SPI configuration.  I have found that the mbed tools crash when trying to generate an update binary due to the large gap between flash sections.  Also, if I remove the new flash sections in the linker, I get a hard fault when the network tries to connect, so the network stack is depending on this new section.   

We need to determine with Cypress, if we want to get this kit working with Pelion or not.  Or if there is a way to turn off the need for new flash section if building an application that uses the Pelion client.

0xc0170, 2019-09-26:
> I have found that the mbed tools crash when trying to generate an update binary due to the large gap between flash sections

How does it crash in the tools, python traceback somewhere  ? or is it coming from the toolchain itself?

maclobdell, 2019-09-26:
Summary of issues:
•	Issue 1 - After flashing the bootloader + application hex file, the Pelion bootloader fails to verify the hash of the application.  I believe this is because the tools are generating a hash based on the application area + supervisory flash area, but the bootloader is only able to hash and check the application area.  

•	Issue 2 - The mbed tools generate a bin file that is used as the payload in the FOTA.  With the large gap in the sections, the generated bin file is very large ~99 MB - too big to work with.

•	Issue 3 - With respect to the tools issue I previously found, I created issue https://github.com/ARMmbed/mbed-os/issues/11576 and made PR https://github.com/ARMmbed/mbed-os/pull/11577.  I discovered it was python 3 related.  It works without this change on Python 2. 

Temporary workarounds tried:
•	If I modify the linker to not add the supervisory section, when I run the code, the bootloader is able to verify the hash and boot the application,  But the application gets a hard fault when trying to connect to the network.  I assume this is because the network stack is using the supervisory flash to check where the 4343W firmware is.  

Possible solutions?
•	Remove need for use of supervisory flash area - under investigation
•	Update bootloader to support multiple flash regions – this will be a challenge

Waiting on Cypress feedback.

maclobdell, 2019-11-06:
As of Mbed OS 5.14.1, this issue is resolved now.  

Example configuration for mbed-bootloader:
```
        "CY8CPROTO_062_4343W": {
            "target.bootloader_supported"                  : true,
            "target.mbed_rom_start"                        : "0x10000000",
            "target.mbed_rom_size"                         : "0x00200000",
            "target.sectors"                               : [[268435456, 512]],
            "target.components_remove"                     : [ "SPM_MAILBOX", "SD", "RTX", "SOFTFP" ],
            "target.features_remove"                       : [ "BLE" ],			
            "mbed-bootloader.bootloader-size"              : "0x0000C000",
	    "target.restrict_size"                         : "0x0000C000",
            "update-client.application-details"            : "0x1000C000",
            "mbed-bootloader.application-start-address"    : "0x1000C400",
            "storage.storage_type"                         : "TDB_INTERNAL",
            "storage_tdb_internal.internal_base_address"   : "0x100F0000",
            "storage_tdb_internal.internal_size"           : "0x00020000",
            "update-client.storage-address"                : "0x10110000",
            "mbed-bootloader.max-application-size"         : "0x000F0000",
            "update-client.storage-size"                   : "0x000F0000", 
            "platform.stdio-baud-rate"                     : 115200,
            "target.extra_labels_remove"                   : [ "WICED", "CORDIO", "PSA", "MBED_SPM", "WHD", "4343W", "CYW43XXX" ],
            "target.device_has_remove"                     : [
                                                              "SERIAL_ASYNCH",
                                                              "SERIAL_FC",
                                                              "PORTOUT",
                                                              "PORTINOUT",
                                                              "RTC",
                                                              "I2C_ASYNCH",
                                                              "SPI_ASYNCH",
                                                              "STDIO_MESSAGES",
                                                              "LPTICKER",
                                                              "TRNG",
                                                              "CRC",
                                                              "PORTIN",
                                                              "PWMOUT",
                                                              "ANALOGIN",
                                                              "ANALOGOUT",
                                                              "I2C",
  	                                                      "I2CSLAVE",
						              "SPI",
							      "SPISLAVE"   
                                                             ] 
        },
```

Example configuration for mbed-os-example-pelion: 

```
        "CY8CPROTO_062_4343W": {
            "target.macros_add"                         : ["MBEDTLS_USER_CONFIG_FILE=\"mbedTLSConfig_mbedOS.h\"", "MBED_BOOTLOADER_SIZE=(48*1024)", "ARM_UC_USE_PAL_BLOCKDEVICE=0"],
            "target.network-default-interface-type"     : "WIFI",            
            "target.bootloader_img"                     : "bootloader/mbed-bootloader_cy8cproto.hex",		
            "target.header_offset"                      : "0x0000C000",
            "target.app_offset"                         : "0x0000C400",
            "update-client.bootloader-details"          : "0x10009C8C",
            "update-client.application-details"         : "0x1000C000",
            "update-client.storage-address"             : "0x10110000",      
            "update-client.storage-size"                : "0x000F0000",
            "update-client.storage-locations"           : 1,
            "mbed-cloud-client.update-storage"          : "ARM_UCP_FLASHIAP",
            "storage_tdb_internal.internal_base_address": "0x100F0000",
            "storage_tdb_internal.internal_size"        : "0x00020000",
            "storage.storage_type"                      : "TDB_INTERNAL",
            "target.bootloader_supported"               : true,
            "target.mbed_rom_start"                     : "0x10000000",
            "target.mbed_rom_size"                      : "0x00200000",
            "target.sectors"                            : [[268435456, 512]],
            "target.extra_labels_remove"                : ["CM0P_SLEEP"]     
```

This uses internal flash only.

Note that by default, the Cortex M0+ co-processor runs its own application that is pre-built and combined with the Cortex M4 application in the final binary.  This Cortex M0+ application just puts itself to sleep.  When building the bootloader hex file, the Cortex M0+ application gets included.  But when building the application, you need to exclude it because it is already present in the bootloader hex fie.  To do this, the "CM0P_SLEEP" label is removed.

closed 2019-11-06 by maclobdell