ARMmbed/mbed-os #10897: K64 FSL needs an upgrade

Alex-EEE, 2019-06-25:
### Description

In case anyone out there runs into this issue, and in case anyone wants to file a PR to address it.

The current FSL for Kinetis K64 is out of date.  That hasn't generally been a problem, but when trying to bring up a custom board, mbed-os was getting stuck in this step in the mbed version of the FSL:
```
status_t CLOCK_SetExternalRefClkConfig(mcg_oscsel_t oscsel) {
...
    MCG->C7 = (MCG->C7 & ~MCG_C7_OSCSEL_MASK) | MCG_C7_OSCSEL(oscsel);
    if (kMCG_OscselOsc == oscsel)
    {
        if (MCG->C2 & MCG_C2_EREFS_MASK)
        {
            while (!(MCG->S & MCG_S_OSCINIT0_MASK))
            {
            }
        }
    }
    if(needDelay)
...
```

After pulling the latest FSL from NXP, I noticed they removed this while loop.  Looks like mbed has a buggy version of fsl_clock.c

FSL version:
```
status_t CLOCK_SetExternalRefClkConfig(mcg_oscsel_t oscsel) {
...
    MCG->C7 = (MCG->C7 & ~MCG_C7_OSCSEL_MASK) | MCG_C7_OSCSEL(oscsel);
    if(needDelay)
...
```

Using this version of fsl_clock.h, fsl_clock.c, my board boots to main no problem.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X ] Bug

0xc0170, 2019-06-27:
@Alex-EEE Can you send PR to address this ? I'll mention nxp team to review this and update the entire SDK but to process this faster - to patch the file (will be overwritten by the update anyway).

@ARMmbed/team-nxp Please review

mmahadevan108, 2019-06-28:
PR submitted with update clock driver.

linlingao, 2019-07-03:
Thanks for the PR. Can you close this issue?

closed 2019-07-03 by Alex-EEE