ARMmbed/mbed-os #10898: M23: Compile error with Arm fault handler on GCC

ccli8, 2019-06-26:
### Description

#### Environment

**GCC**
arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]

**mbed-os-examble-blinky**:
5907912 (HEAD, tag: mbed-os-5.12.0-rc4, tag: mbed-os-5.12.0) Merge pull request #168 from ARMmbed/release-candidate

**mbed-os**:
8a7f5914f6 (HEAD -> master, origin/master, origin/HEAD) Merge pull request #10826 from ABOSTM/IAR_EXPORT_STM32H743ZI

**Target**:
NUMAKER_PFM_M2351

I am trying Arm fault handler on NUMAKER_PFM_M2351. To enable this function, I
remove `MBED_FAULT_HANDLER_DISABLED` from `macros` list and add `GCC_ARM` into `supported_toolchains` for NUMAKER_PFM_M2351 target in `mbed-os/targets/targets.json`. 

<pre>
    "NUMAKER_PFM_M2351": {
        "core": "Cortex-M23-NS",
        "default_toolchain": "ARMC6",
        "extra_labels": [
            "NUVOTON",
            "M2351",
            "M2351KIAAEES",
            "FLASH_CMSIS_ALGO"
        ],
        "OUTPUT_EXT": "hex",
        <b>
        "macros": [
            "MBED_TZ_DEFAULT_ACCESS=1",
            "LPTICKER_DELAY_TICKS=3"
        ],
        </b>
        "trustzone": true,
        "is_disk_virtual": true,
        <b>
        "supported_toolchains": ["ARMC6", "GCC_ARM"],
        </b>
</pre>

This change can compile successfully on ARMC6 but fail on GCC_ARM with error message below. It seems the error comes from limit of GCC's UAL syntax support.

```
Compile [ 53.2%]: except.S
[ERROR] except.S: Assembler messages:
except.S:50: Error: cannot honor width suffix -- `ldr R3,=FAULT_TYPE_HARD_FAULT'
except.S:64: Error: cannot honor width suffix -- `ldr R3,=FAULT_TYPE_MEMMANAGE_FAULT'
except.S:78: Error: cannot honor width suffix -- `ldr R3,=FAULT_TYPE_BUS_FAULT'
except.S:92: Error: cannot honor width suffix -- `ldr R3,=FAULT_TYPE_USAGE_FAULT'
except.S:108: Error: cannot honor width suffix -- `ldr R1,=0x4'
except.S:158: Error: cannot honor width suffix -- `ldr R6,=0x200'
except.S:165: Error: cannot honor width suffix -- `ldr R6,=0x10'
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-06-26:
@ccli8 Wouldn't this better reported to GCC ARM team?

I found one related issue to this: https://www.mail-archive.com/bug-binutils@gnu.org/msg30901.html from this year.

kjbracey-arm, 2019-06-26:
Well found, @0xc0170.  Yes, that looks like a real assembler bug.

Rather nasty - `LDR =` should be the safest CPU-independent way to load constants.

In this case though you could replace them all with `MOVS R3,#xxx`, which can take an 8-bit immediate, and will work on all CPUs.

All except the `0x200`. I guess `MOVS R6, #2; LSLS R6,#8` would do. And it's no bigger than whatever the `LDR=` would have produced.

ccli8, 2019-06-27:
@0xc0170 @kjbracey-arm Thanks for your information. Keep this ticket open and wait for next GCC release to fix the issue.

MarceloSalazar, 2020-05-15:
Closing as the m2351 is not enabled in master for Mbed OS 6.0.
We'll retest when reintroduced. It will have to work with gcc 9.

closed 2020-05-15 by MarceloSalazar