ARMmbed/mbed-os #10901: u-blox SARA-R412M with cellular PPP

vidavidorra, 2019-06-26:
### Description

I'm trying to use the u-blox PPP with the u-blox SARA-R412M-02B. The u-blox module is running on firmware version `M0.09.00`. The latest version is `M0.10.00` but in the release notes of that I can't see any relevant changes to going to PPP mode.
I'm running on mbed-os-5.11.4, but didn't see significant changes for this in the release notes of later releases. Please let me know if there are significant changes (and which) that can help with this issue.

The issue seems related to #7682, but I also cannot get this working when I manually change `AT+CGDATA="PPP",1` to `ATD*99***1#`. That change is basically using [this](https://github.com/ARMmbed/mbed-os/pull/7870/commits/d208706fd6f1df8be89863f380d1ca601b7f13d2#diff-648c841900b02b662e396ea7f65d382eR387) `else`-statement. See the log below (when I was using `AT+CGDATA` but output is the same for the `ATD` option.

### Log
```
1970-01-01 00:00:00 [3] Starting connection thread, timeout in 360000 ms (submodules/mbed-system/source/wtg_service/wtg_service.cpp:190)
[INFO][CELL]: CellularContext set blocking 0
[INFO][CELL]: CellularContext connect
[DBG ][CELL]: AT OoB readable 1, len 0
[224][DBG ][CELL]: AT OoB done
[INFO][CELL]: Start connecting (timeout 1 s)
AT
[DBG ][CELL]: AT error -3012
[WARN][CELL]: AT timeout
[DBG ][CELL]: AT error -3012
[DBG ][CELL]: consume_to_tag not found
[DBG ][CELL]: AT stop tag not found
[DBG ][CELL]: AT error -3012
AT
[DBG ][CELL]: AT error -3012
[WARN][CELL]: AT timeout
[DBG ][CELL]: AT error -3012
[DBG ][CELL]: consume_to_tag not found
[DBG ][CELL]: AT stop tag not found
[DBG ][CELL]: AT error -3012
[DBG ][CELL]: Init => Power
[INFO][CELL]: Modem power ON (timeout 1 s)
[DBG ][CELL]: Power => Device ready
[DBG ][CELL]: AT OoB readable 1, len 0
[0][DBG ][CELL]: AT OoB done
[DBG ][CELL]: AT flush
ATE0
[DBG ][CELL]: AT error -3012
[WARN][CELL]: AT timeout
[DBG ][CELL]: AT error -3012
[DBG ][CELL]: consume_to_tag not found
[DBG ][CELL]: AT stop tag not found
[DBG ][CELL]: AT error -3012
[WARN][CELL]: AT timeout
[DBG ][CELL]: AT error -3012
[DBG ][CELL]: consume_to_tag not found
[DBG ][CELL]: AT stop tag not found
[DBG ][CELL]: AT error -3012
[DBG ][CELL]: Device ready: retry 1/10
[INFO][CELL]: Continue after 2 seconds
[DBG ][CELL]: AT flush
ATE0
ATE0

OK
AT+CMEE=1

OK
[INFO][CELL]: Modem ready
AT+CGMM

SARA-R412M-02B

OK
[WARN][CELL]: AT timeout
[DBG ][CELL]: AT error -3012
[DBG ][CELL]: AT error -3012
[WARN][CELL]: AT timeout
[DBG ][CELL]: AT error -3012
[DBG ][CELL]: consume_to_tag not found
[DBG ][CELL]: AT stop tag not found
[DBG ][CELL]: AT error -3012
[DBG ][CELL]: callback: 4096, err: 0, data: -1
[DBG ][CELL]: CellularContext: event 4096, err 0, data -1
1970-01-01 00:00:09 [3] Connection status: 1, cellular status: 1 (submodules/mbed-system/source/connection/network_connection.cpp:173)
[DBG ][CELL]: Device ready => SIM pin
AT+CSQ

+CSQ: 99,99

OK
[INFO][CELL]: RSSI unknown
[INFO][CELL]: Setup SIM (timeout 1 s)
[DBG ][CELL]: AT flush
AT+CPIN?

+CPIN: READY

OK
[INFO][CELL]: SIM is ready
[DBG ][CELL]: callback: 4097, err: 0, data: 0
[DBG ][CELL]: CellularContext: event 4097, err 0, data 0
1970-01-01 00:00:09 [3] Connection status: 1, cellular status: 2 (submodules/mbed-system/source/connection/network_connection.cpp:173)
AT+CGREG=2

OK
AT+CREG=2

OK
AT+CGACT?

+CGACT: 1,1

O[DBG ][CELL]: Found active context
K
[DBG ][CELL]: ACTIVE CONTEXT FOUND, skip registering.
AT+CGATT?

+CGATT: 1

OK
[DBG ][CELL]: DEVICE IS ALREADY ATTACHED TO NETWORK, skip registering and attach.
[INFO][CELL]: Setting the packet domain stuffsies...
[INFO][CELL]: AT_CellularNetwork::set_packet_domain_event_reporting
AT+CGDCONT?

+CGDCONT: 1,"IP","public4.m2minternet.com","XXXXXX",0,0,0,0

OK
AT+CGEREP=1

OK
[DBG ][CELL]: SIM pin => Registering network
AT+CSQ

+CSQ: 99,99

OK
[INFO][CELL]: RSSI unknown
[INFO][CELL]: Network registration (timeout 180 s)
AT+CGREG?

+CGREG: 2,4

O[DBG ][CELL]: +CGREG: RegistrationUnknown, LAC -1, cell -1, RAT unknown
K
AT+CREG?

+CREG: 2,5,"FFFE","19A4201",7

[DBG ][CELL]: +CREG: RegisteredRoaming, LAC 65534, cell 26886657, E_UTRAN
OK
[INFO][CELL]: Roaming network.
[DBG ][CELL]: callback: 4098, err: 0, data: 11
[DBG ][CELL]: CellularContext: event 4098, err 0, data 11
1970-01-01 00:00:10 [3] Connection status: 1, cellular status: 18 (submodules/mbed-system/source/connection/network_connection.cpp:173)
[DBG ][CELL]: Registering network => Attaching network
AT+CSQ

+CSQ: 99,99

OK
[INFO][CELL]: RSSI unknown
[INFO][CELL]: Attaching network (timeout 60 s)
AT+CGATT?

+CGATT: 1

OK
[DBG ][CELL]: callback: 4102, err: 0, data: 1
[DBG ][CELL]: CellularContext: event 4102, err 0, data 1
1970-01-01 00:00:10 [3] Connection status: 1, cellular status: 38 (submodules/mbed-system/source/connection/network_connection.cpp:173)
1970-01-01 00:00:10 [3] Connection status: 3, cellular status: 38 (submodules/mbed-system/source/connection/network_connection.cpp:173)
AT+CGDCONT?

+CGDCONT: 1,"IP","public4.m2minternet.com,"XXXXXXXXXXX",0,0,0,0

OK
[INFO][CELL]: Found PDP context 1
[INFO][CELL]: CellularContext PPP connect
AT+CGDATA="PPP",1

ERROR
[DBG ][CELL]: AT error -3012
[ERR ][CELL]: Failed to CONNECT
[ERR ][CELL]: Failed to open data channel!
[DBG ][CELL]: callback: 0, ptr: 2
[DBG ][CELL]: CellularContext: event 0, ptr 2
1970-01-01 00:00:11 [3] Connection status: 1, cellular status: 38 (submodules/mbed-system/source/connection/network_connection.cpp:173)
1970-01-01 00:00:11 [3] Connection status: 1, cellular status: 38 (submodules/mbed-system/source/connection/network_connection.cpp:173)
[INFO][CELL]: CellularContext disconnected
```

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

vidavidorra, 2019-06-27:
I've got this part figured out myself now but I don't have a good source for why yet.

Apparently the R4 module can't use `AT+CGDATA="PPP",` to go to data mode (see AT commands manual [p 135](https://www.u-blox.com/sites/default/files/SARA-R4-SARA-N4_ATCommands_(UBX-17003787).pdf#page=135)). It needs to use `ATD*99***#1` to go to data mode. The only 'real' hint for that is the fact that the R4's AT commands manual doesn't explicitly document the `+CGDATA` command.

The `DTR` pin needs to be pulled low in order to go to data mode. I haven't been able to find a good source (yet) on why that is needed. Only source I have is [this](https://portal.u-blox.com/s/question/0D52p00008HKDbBCAX/ublox-sarar4-disable-to-enter-ppp-state) and [this](https://portal.u-blox.com/s/question/0D52p00008HKDmxCAH/can-i-use-ppp-as-communication-link-with-a-sarar410m02b-module-so-to-access-the-network) forum post.

<br>
WIth this information I don't know why the UBLOX_AT cellular target specifies `AT_CGDATA=1` (see [here](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.12.4/features/cellular/framework/targets/UBLOX/AT/UBLOX_AT.cpp#L31)) for the `TARGET_UBLOX_C030_R41XM` target. Based on the above I don't think that would work.

vidavidorra, 2019-07-01:
I think I've got most of this figured out myself now. Only thing I'm still struggling with is how I can select the protocol to use (e.g. LTE Cat M1) for the SARA-R412M from the application. Can anyone help with that?

partychief, 2019-07-01:
> how I can select the protocol to use (e.g. LTE Cat M1) for the SARA-R412M from the application
network capabilities can be set using AT+URAT
e.g. 
AT+URAT=7 (LTE Cat M1 only)
AT+URAT=8 (NBIoT only)
AT+URAT=7,9 (LTE Cat M1 with EDGE fallback)
Depends what MNO /MVNO SIM you use but in our situation the modem attaches to EDGE instead of LTE Cat M1, because registration took longer on the LTE roaming partner resulting in a timeout and fallback to 2G EDGE.

vidavidorra, 2019-07-01:
I think I need to elaborate a bit. I know how to manually set the the protocol to use using the AT commands, that is indeed what @partychief posted using the `AT+URAT=...` commands.
What I want to know is how I can use the OS drivers (e.g. `CellularNetwork`) to set the protocol, or rather the RAT (radio access technology). I've found the `CellularNetwork:: set_access_technology` function but I don't see anywhere internally how that could possibly be using the `AT+URAT=...` command to set the RAT.

closed 2019-10-25 by ciarmcom