ARMmbed/mbed-os #10903: arm_pack_manager of tools for M263

cyliangtw, 2019-06-26:
### Description

In `/tools/arm_pack_manager`, it's not up to date to cover Nuvoton M263.
Is there any one regular auto-sync mechanism ?
Will sync arm_pack_manager recently ?

### Issue request type

    [ v] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-06-27:
It should be possible to rerun the script - it should update json file and this would be commited. Isn't that the case here?

jeromecoutant, 2019-06-27:
> It should be possible to rerun the script

Any documentation ?

Thx

mark-edgeworth, 2019-07-01:
@0xc0170 can you point us at any documentation for this please?

0xc0170, 2019-07-01:
The only one I am aware of is https://os.mbed.com/docs/mbed-os/v5.13/reference/adding-and-configuring-targets.html

mark-edgeworth, 2019-07-01:
Thanks for the reference!

cyliangtw, 2019-07-18:
Is there any python script to sync  index.json and PDSC (Pack Description) retrieved from each CMSIS Pack ?

mark-edgeworth, 2019-07-18:
Hi @cyliangtw, it looks like the code in tools/arm_pack_manager could possibly do this, but I've found no documentation on this.

cyliangtw, 2019-07-18:
Hi @mark-edgeworth , 
To run` __init__.py`  with hardcode file path in python console but got below error, 
`
{'load': <function load at 0x03C51530>, 'LocalPackIndex': 'C:\\ARM_mbed\\NVT\\mbed-os\\tools\\arm_pack_manager\\index.json', '_CPM_PRESENT': True, 'join': <function join at 0x017B6770>, 'dump': <function dump at 0x03C514B0>, 'warnings': <module 'warnings' from 'C:\Python27\lib\warnings.pyc'>, '__builtins__': {'bytearray': <type 'bytearray'>, 'IndexError': <type 'exceptions.IndexError'>,
`

mark-edgeworth, 2019-07-18:
Ok, to be clear, I don't know how or whether this code works. Sadly it looks like it does not.

cyliangtw, 2019-08-05:
@mark-edgeworth , @c1728p9 , so far I could refer to *.pdsc to manually come out M263 part for index.json, however it seems not a good way.
Is there any doc to guide "Rebuild index.json from *.pdsc" by script or auto-batch ?

cyliangtw, 2019-08-05:
@mark-edgeworth , @c1728p9 ,
I got some error as while  update cmsis pack index.json:
```
[cygwin~mbed-os-example-blinky]$ mbed export --update-packs -i uvision
Downloading Descriptions 2 / 7 [##########>                          ] 28.57 %  Aug 05 20:44:20.854 ERRO download of "https://www.keil.com/pack/Keil.STM32F7xx_DFP.pdsc" failed: unexpected end of file
Downloading Descriptions 3 / 7 [###############>                     ] 42.86 %  Aug 05 20:44:20.897 ERRO download of "https://www.keil.com/pack/Keil.STM32G0xx_DFP.pdsc" failed: unexpected end of file
Downloading Descriptions 4 / 7 [#####################>               ] 57.14 %  Aug 05 20:44:20.899 ERRO download of "https://www.keil.com/pack/Keil.STM32H7xx_DFP.pdsc" failed: unexpected end of file
Downloading Packs 414 / 591 [############################>           ] 70.05 %

```

linlingao, 2019-08-05:
I had to manually add an entry for TI's CC3220SF.
As we're adding support for more targets, .pdsc will always be lacking, I'm not sure relying on pdsc is a good way to go.

cyliangtw, 2019-08-06:
Thanks, I will extract M261 series device from .pdsc and insert to index.json by manually.

closed 2019-08-06 by cyliangtw