ARMmbed/mbed-os #10909: mbed 5.12.4 integration

maniel92, 2019-06-27:
### Description

Hello,
I'm trying to use the SDBlockDevice with mbed-os 5.12.4. I'm working with the Eclipse based tool STM32CubeIDE. Trying to solve my problem I got to [this issue](https://github.com/ARMmbed/mbed-os/issues/8494) but I cannot reproduce the solution.

I cannot find anywhere the folder COMPONENT_SD, not in the mbed-os github code, nor downloading the latest library from the [mbed documentation](https://os.mbed.com/docs/mbed-os/v5.12/apis/sdblockdevice.html), nor elsewhere. 

I have already added this code to my board in targets.json:
`
    	"components": ["SD"],
    	"components_add": ["SD"],
    	"features_add": ["STORAGE"],
`

I ask for help or a bit more of explanation on this issue solving method.

### Issue request type

    [X ] Question
    [ ] Enhancement
    [ ] Bug

jeromecoutant, 2019-06-27:
Hi

COMPONENT_SD is there: https://github.com/ARMmbed/mbed-os/tree/master/components/storage/blockdevice/COMPONENT_SD

Note this is SD over SPI

SD over SDIO (SD connector on ST DISCO board) is still under review: https://github.com/ARMmbed/sdio-driver/pull/1

@MarceloSalazar

maniel92, 2019-06-27:
Hello @jeromecoutant 

Thanks very much! I was looking into features/storage, not into the components folder.
I mark it as closed now.

Best regards,
Manuel

closed 2019-06-27 by maniel92