ARMmbed/mbed-os #10912: How to reset Cellular network interface?

Reda-RM, 2019-06-27:
### Description

I am trying to implement error recovery if the Cellular Network interface reported and error. Currently I attached a status callback and do a system reset (to restart the Cellular state machine and to start connecting from beginning) if I got an error in the status callback. Is there a method to reset the cellular network interface to restart the cellular state machine and start connection from beginning with system reset?

I am using:
- New target based on RiotMicro cellular chip
- Toolchain: GCC_ARM
- MBed-cli
- MBed-OS 5.13 (commit #92a58dff9960788a3582b0f6dced1d5280b7f2f8)

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-07-01:
cc @ARMmbed/mbed-os-wan

AnttiKauppila, 2019-07-03:
@Reda-RM Have you tried: https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/API/CellularDevice.h#L140
?

Reda-RM, 2019-07-03:
soft_power_on() will reset the modem itself. But this will not reset the Cellular framework and its state machine. Then if I used soft_power_on() the modem will be reset, but it will not be triggered to go into the states (Init, SIM initialization, network registration and attaching ).

I think the cellular framework reset sequence (to get the modem up and working again after any failure in modem side) is missing!

AnttiKauppila, 2019-07-03:
Ok, Have you tried CellularDevice::stop()? That will reset the state_machine

Reda-RM, 2019-07-03:
Yes I tried it but it is not enough as it just stop the cellular state machine (may be I am missing something). I tried to delete the cellular context and create a new one but this is not a clean way and did not work as well. 

I tried the following sequence:
`#define SYSTEM_RECOVERY() \
    do { \
        CellularContext * context = (CellularContext *) interface; \
        CellularDevice * device = context->get_device(); \
        device->soft_power_on(); \
        device->stop(); \
        device->shutdown(); /* This will call _state_machine.reset() */ \
    } while(0)`

and then tried to re-connect again but it stopped in CellularDevice::attach_to_network() and did not start the first state (INIT)
some thing is not cleaned yet !

pilotak, 2019-07-04:
I have tried `CellularDevice::stop()` which really stops state machine but does not start again. There is `start` missing and `start_state_machine` is sadly private

AnttiKauppila, 2019-07-05:
@pilotak There is a reason for state machine being private. If you call connect() it should do start_state_machine for you.

@Reda-RM It sounds we might have some inconsistent state handling present. Can you change the status of this to Bug?

Reda-RM, 2019-07-05:
@AnttiKauppila done. Also I think it will be helpful if we could provide some high level method to do network interface resetting/restarting to abstract the user from these details

AnttiKauppila, 2019-07-05:
@Reda-RM You are right we should create a clear API function for this

AriParkkila, 2019-07-15:
@Reda-RM For full reset, you could do:
```
NetworkInterface::disconnect(); // disconnect from cellular network and notify to close sockets
CellularDevice::shutdown(); // prepare modem for power off
CellularDevice::soft_power_off(); // switch off modem
CellularDevice::hard_power_off(); // power supply off to cold boot modem
CellularDevice::stop(); // reset cellular state machine
NetworkInterface::connect(); // connect to network and notify sockets can be created again
```

pilotak, 2019-07-17:
@Reda-RM i have tried it. If I call `CellularDevice::stop();` and than call `connect()` it doesn't start again (it only calls cb NSAPI_STATUS_CONNECTING but does not continue). If i remove `stop` it work ok.

AriParkkila, 2019-07-18:
@pilotak Please see PR #11066.

pilotak, 2019-07-22:
@AriParkkila yes that fixes it

AriParkkila, 2019-07-23:
@pilotak Thanks for testing. I guess we could implement this as `CellularDevice::reset/restart()` in [Cellular API](https://github.com/ARMmbed/mbed-os/blob/master/features/cellular/framework/API/CellularDevice.h)..?

pilotak, 2019-07-23:
that would be good idea. If this would be too difficult at least do some documentation. As far as i know there is only basic `cellular-example` and it would be nice to have advance too. I have started one [here](https://github.com/pilotak/mbed-cellular-boilerplate)

AriParkkila, 2019-08-02:
PR #11154 adds cellular stop in `CellularDevice::shutdown()`, so to reset cellular network interface:
```
    dev->shutdown(); // prepare modem for power off, also disconnects PDNs
    dev->soft_power_off(); // switch off modem
    dev->hard_power_off(); // power supply off to cold boot modem
    ctx->connect();
```
@Reda-RM This issue can be closed?

Reda-RM, 2019-08-04:
PR #11159 adds a cellular restart API.
I will close the issue.

closed 2019-08-04 by Reda-RM