ARMmbed/mbed-os #10915: cannot move location counter backwards (from 20004cf0 to 20004c00)

fengshengchang, 2019-06-28:
Target: NUCLEO_F103RB
net driver config to use : features\cellular\framework\targets\QUECTEL\M26

When compile the follow example code， I got an link error:
cannot move location counter backwards (from 20004cf0 to 20004c00)

code:
	EthernetInterface eth;
	eth.connect();
	
	// Create a socket
	TCPSocket sock;
	sock.open(&eth);
	
	// Connect
	sock.connect("arm.com", 80);
	
	// Send data
	sock.send("GET / HTTP/1.0\r\n\r\n", 18);
	
	// Receive data
	char buf[100];
	sock.recv(buf, 100);
	
	// Close the socket
	sock.close();

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-06-28:
how much memory does `nucleo f103rb` have ?

cc @ARMmbed/team-st-mcd

jeromecoutant, 2019-06-28:
https://github.com/ARMmbed/mbed-os/blob/201a264051205888434e89077318f7d729703a02/targets/TARGET_STM/TARGET_STM32F1/TARGET_NUCLEO_F103RB/device/TOOLCHAIN_GCC_ARM/STM32F103XB.ld#L13

fengshengchang, 2019-06-29:
Is there any solution to this problem?
@jeromecoutant

jeromecoutant, 2019-07-01:
Seems that your net driver is too large for this small STM32F1 target?

jeromecoutant, 2019-10-11:
I think we should close this issue
Thx

closed 2019-10-13 by fengshengchang