ARMmbed/mbed-os #10918: BG96 Cellular test: Assert in features-cellular-tests-api-cellular_network

jeromecoutant, 2019-06-28:
### Description

I am using a DISCO_L496AG with BG96 cellular modem.

1 test is failing with assert in https://github.com/ARMmbed/mbed-os/blob/a2c915244136ee80f6221d5341ff44f790ec74c1/features/cellular/framework/AT/ATHandler.cpp#L1253

Note I have added some printf cto display the command before assert:

````
$ mbed test -t ARM -m DISCO_L496AG -v -n features-cellular-tests-api-cellular_network

[1561724911.75][CONN][RXD] >>> Running case #4: 'CellularNetwork test other functions'...
[1561724911.81][CONN][INF] found KV pair in stream: {{__testcase_start;CellularNetwork test other functions}}, queued...
[1561724911.89][CONN][RXD] DEBUG: cmd +QCFG cmd_chr ="nwscanseq",010203
[1561724911.89][CONN][RXD]
[1561724911.90][CONN][RXD]
[1561724911.92][CONN][RXD] ++ MbedOS Error Info ++
[1561724911.97][CONN][RXD] Error Status: 0x80FF0144 Code: 324 Module: 255
[1561724912.06][CONN][RXD] Error Message: Assertion failed: (3 + strlen(cmd) + cmd_char_len) < BUFF_SIZE
[1561724912.08][CONN][RXD] Location: 0x80163AF
[1561724912.14][CONN][RXD] File: .\features\cellular\framework\AT\ATHandler.cpp+1255
[1561724912.15][CONN][RXD] Error Value: 0x0
[1561724912.26][CONN][RXD] Current Thread: main  Id: 0x200031E4 Entry: 0x8016839 StackSize: 0x1000 StackMem: 0x200019D8 SP: 0x20002804
[1561724912.36][CONN][RXD] For more info, visit: https://mbed.com/s/error?error=0x80FF0144&tgt=DISCO_L496AG
[1561724912.38][CONN][RXD] -- MbedOS Error Info --
````

From https://www.quectel.com/UploadImage/Downlad/Quectel_BG96_AT_Commands_Manual_V2.1.pdf
I could see that AT+QCFG="nwscanseq",010203 should not cause assert...

@LMESTM @MarceloSalazar 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

closed 2019-07-05 by ciarmcom

jeromecoutant, 2019-07-05:
Soem comments are missing.....

0xc0170, 2019-07-05:
Based on the internal ticket, https://github.com/ARMmbed/mbed-os/pull/10925 was merged and should address this. @AnttiKauppila please review

AnttiKauppila, 2019-07-07:
BUFF_SIZE is doubled to 32 and can hold that data. This is fixed by #10925