ARMmbed/mbed-os #10922: Mbed 5.13 Size Report Defect

ifyall, 2019-06-28:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-07-01:
cc @ARMmbed/mbed-os-tools

mark-edgeworth, 2019-07-01:
Thanks, issue was moved to our backlog this morning.

ifyall, 2020-01-23:
FYI that this defect also causes the map.html file in the BUILD directory to report smaller sizes than expected, so it shows up in multiple places.

ifyall, 2020-07-31:
@andypowers Note that this is not fixed in Mbed OS 6.2. The size report is off by a factor of 10 for the CY8CPROTO_062_4343W, for instance.
Ian

andypowers, 2020-07-31:
@ifyall Thanks for checking. Martin is out next week but I'll see if someone else can investigate.

boraozgen, 2020-09-02:
This issue is present for the `NUCLEO_F429ZI` target, `GCC_ARM` compiler and `mbed-os-6.2.0`.

For baremetal blinky example, Mbed CLI reports:
```
Total Static RAM memory (data + bss): 1456(+1456) bytes
Total Flash memory (text + data): 20504(+20504) bytes
```

whereas `arm-none-eabi-size.exe` reports

```
   text    data     bss     dec     hex filename
  23164     136    1600   24900    6144 BUILD/NUCLEO_F429ZI/GCC_ARM/mbed-os-example-blinky-baremetal.elf
```

where the latter is correct when compared to the actual binary size. The inaccuracy is even higher for larger binaries.

ifyall, 2021-05-14:
This is fixed in the CLI 2 and is not critical for backporting to CLI 1.

closed 2021-05-14 by ifyall

boraozgen, 2021-05-25:
> This is fixed in the CLI 2 and is not critical for backporting to CLI 1.

Is it? I am still getting inaccurate size information with CLI 2.