ARMmbed/mbed-os #10923: Nucleo L432KC AnalogIn produces wrong Values with Optimizer level other than O0

timoxd7, 2019-06-29:
### Description

Target: Nucleo L432KC
Toolchain: ST STM32 @ 5.4.3
Tool: VS Code with PlatformIO
SHA: Sorry, i don't know
Steps to Reproduce: Just read three analog Values and print them in an endless loop on the Monitor (for me, each 0.15s)

Problem: If i read analog values from the analog in pins, it produces wrong values. The first reading (it doesnt't metter which pin, only which is first red in the loop, see code below) has a huge offset of about 0.5 (if reading AnalogIn with read() or operator float). The other two values are also not correct, but closer to the real value. If giving voltage on the second red pin, the first pin stucks at about 0.5.

Workaournd: Set Optimizer from Os (standard) to O0 (no optimizations).

Tested also with older Versions of the ST STM32 Toolchain (-> older Versions of mbed) with the same problem.

For the code down below: If i read the values directly in the printf (instead of x1, i write in1), it does print nothing on the monitor instead of wrong values. If i multiply the values (for me, it was 5.0) inside the printf (instead of x1, i write in1 * 5.0) it works but again with the strange offset.

Code to reproduce:

    #include <mbed.h>

    #define REFRESH_INTERVAL 0.15

    AnalogIn in1(A0);
    AnalogIn in2(A1);
    AnalogIn in3(A2);

    int main() {
        while(true) {
            float x1 = in1;
            float x2 = in2;
            float x3 = in3;
            printf("In 1: %.3f\tIn 2: %.3f\tIn 3: %.3f\n", x1, x2, x3);
            wait(REFRESH_INTERVAL);
        }
    }

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-07-01:
cc @ARMmbed/team-st-mcd

jeromecoutant, 2019-07-19:
Hi

I think issue should be closed after #11072
Maybe someone could check ?

Thx

jeromecoutant, 2019-08-22:
@0xc0170 
Let's close this issue ?
Thx

closed 2019-08-22 by 0xc0170

sansil, 2019-11-20:
Hi, I'm experimenting the same issue using mbed-os-5.13.2 and stm32l433rc nucleo board. 
Any update?
Thank you in advance

0xc0170, 2019-11-21:
@sansil 5.14.0 version should have this fixed.

sansil, 2019-11-21:
Yes It's fixed, thank you @0xc0170.