ARMmbed/mbed-os #10926: New STM32Fx target LPTICKER and USTICKER

vidavidorra, 2019-07-01:
### Description

I've got a couple of questions for creating new STM32Fx targets.
1. What determines whether a STM32Fx target has the `LPTICKER`?
  E.g. the `NUCLEO_F207ZG` target doesn't have the `LPTICKER` (see [here](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.13.0/targets/targets.json#L2354)) and the `NUCLEO_F302R8` target does have the `LPTICKER` (see [here](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.13.0/targets/targets.json#L2362)). How do I determine whether a new target, e.g. a STM32F205RG, needs to have the `LPTICKER` removed or not? Is it something with the core used (e.g. `Cortex-M3` doesn't have `LPTICKER` and `Cortex-M4F` does), can I determine it from the datasheet?

2. What determines whether a STM32Fx target has the `USTICKER`?
3. Additionally for the `USTICKER` option, how to determine which timer to use for this? E.g. the `NUCLEO_F207ZG` target uses TIM5 (see [here](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.13.0/targets/TARGET_STM/TARGET_STM32F2/TARGET_NUCLEO_F207ZG/device/us_ticker_data.h#L27)), which happens to be the last one of the general purpose timers. But than the `NUCLEO_F103RB` target uses TIM3 (see [here](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.13.0/targets/TARGET_STM/TARGET_STM32F1/TARGET_NUCLEO_F103RB/device/us_ticker_data.h#L27)), which is the second to last of the general purpose timers. I couldn't really find a pattern for which timer is used and would love some explanation on this.

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

LMESTM, 2019-07-01:
> 1. What determines whether a STM32Fx target has the `LPTICKER`?

To have LPTICKER, you need the STM32 HW to offer either LPTIM or RTC with sub-seconds precision. F2 family doesn't offer this. You can check this in the Reference Manual

> 2. What determines whether a STM32Fx target has the `USTICKER`?

USTICKER relies on the standard TIMER HW from STM32, and all STM32 have this support.

> 3. Additionally for the `USTICKER` option, how to determine which timer to use for this? E.g. the `NUCLEO_F207ZG` target uses TIM5 (see [...]

This is a design choice when enabling the target. We select one General Purpose timer and try to keep as many pins as possible to be used as PWM, especially the pins on Arduino connector. You can see the list from PeripheralPins.c file.

Hope this helps

vidavidorra, 2019-07-01:
@LMESTM Thanks for your quick and elaborate response! I've got a few follow-up questions.

> To have LPTICKER, you need the STM32 HW to offer either LPTIM or RTC with sub-seconds precision. F2 family doesn't offer this. You can check this in the Reference Manual

So for the NUCLEO_F302R8 this is enabled because it has an RTC with sub-seconds precision, right? I've looked in the reference manual [RM0365 §27.6.10](https://www.st.com/content/ccc/resource/technical/document/reference_manual/16/bf/04/cd/f6/25/44/a7/DM00094349.pdf/files/DM00094349.pdf/jcr:content/translations/en.DM00094349.pdf#page=752) to determine this.

> This is a design choice when enabling the target. We select one General Purpose timer and try to keep as many pins as possible to be used as PWM, especially the pins on Arduino connector. You can see the list from PeripheralPins.c file.

So if I understand correctly I need to 
- Keep this option enabled for all STM32-based devices.
- Select one timer which has the least amount of PWM pins that will be lost.
- Disable PWM pins for the selected timer in `PeripheralPins.c` (e.g. all [TIM5 pins](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.13.0/targets/TARGET_STM/TARGET_STM32F2/TARGET_NUCLEO_F207ZG/PeripheralPins.c#L147) for NUCLEO_F207ZG)

Is there a guideline that this has to be one of the last general purpose timers e.g. because the first timer is always used by the OS or something? Does this always need to be a general purpose timer?

LMESTM, 2019-07-01:
> So for the NUCLEO_F302R8 this is enabled because it has an RTC with sub-seconds precision, right? I've looked in the reference manual [RM0365 [...]

Yes I think so. RTC is the default HW being used. 
When LPTIM is selected, you'll find below lines in targets.json

```
            "lpticker_lptim": {
                "help": "This target supports LPTIM. Set value 1 to use LPTIM for LPTICKER, or 0 to use RTC wakeup timer",
                "value": 1
            }
```

> Is there a guideline that this has to be one of the last general purpose timers e.g. because the first timer is always used by the OS or something? Does this always need to be a general purpose timer?

No specific guideline that I know of. We've developed the driver to work with GP timer and we know it is reliable with it. I am not sure if other timers will have all the needed functionalities that are used by this driver. 

In most cases, you should be able to rely on an existing target from the same family you're looking for. 
What family or target do you want to add ?

vidavidorra, 2019-07-01:
> What family or target do you want to add ?

I'm adding some of the bare MCU's for my use case. I have some custom developed PCB's with e.g. a STM32F205RG MCU on it. As a target for such a device I'm creating the TARGET_STM32F205RG, so I'm basing most of the new target on the TARGET_NUCLEO_F207ZG. I believe that is the closest target available.
What I miss in the OS targets are targets for the bare MCU's, which I can only imagine a lot of people that use custom designed PCB's need. For the devices I use, I'm therefore creating a few targets for the bare MCU.

I've also got some other questions regarding the startup, linker and system_clock files. Can I bother you with those as well? 😄

LMESTM, 2019-07-01:
@vidavidorra  Coudl I suggest that you create a Pull Request with the target or at least the MCU part of the target, so that we review it ?    The startup / linker and clock files are derived from the CUBE and mbed targets. you can started from F207 files and update them with inputs from the CUBE files.

vidavidorra, 2019-07-03:
> Coudl I suggest that you create a Pull Request with the target or at least the MCU part of the target, so that we review it ?

Yes, I'll do that eventually but that will take some time for me. I currently have the target working and, as written in my previous comment, still have some questions about why certain values are chosen. When I've created a PR we can better discuss those points I think. During this proces I've also written some steps in order to create the target for myself so I can post those in the PR as well.

I've got one (small) question before that though and hope you can help me with it. Why is the vector table 8-byte aligned? See [here](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.13.0/targets/TARGET_STM/TARGET_STM32F2/TARGET_NUCLEO_F207ZG/device/TOOLCHAIN_GCC_ARM/STM32F207ZGTx_FLASH.ld#L17) for the F207ZG. I've overlooked this step first and thus used 0x184 and that worked fine as well. I've now changed it to 0x188 but am wondering why it is 8-byte aligned.

LMESTM, 2019-07-03:
@vidavidorra  Concerning steps on how to ad a target, you can find the existing wiki here
https://os.mbed.com/teams/ST/wiki/steps-to-create-a-new-STM32-platform
Please feel free to help complete  :-) 

When it comes to the 8 bytes aligned - this is a requirement from ARMC6 compiler and has been applied widely to all compilers for consistenct by arm MBED team. See below.
```
$ git blame targets/TARGET_STM/TARGET_STM32F2/TARGET_NUCLEO_F207ZG/device/TOOLCHAIN_GCC_ARM/STM32F207ZGTx_FLASH.ld -L 17,17
bf1a2c84853 (Deepika 2018-09-05 17:18:11 -0500 17) /* 8-byte aligned(0x184) = 0x188 */
$ git show bf1a2c84853
commit bf1a2c848534fd625852132a091428c925a5569f
Author: Deepika <deepika.bhavnani@arm.com>
Date:   Wed Sep 5 17:18:11 2018 -0500

    [ST]: Fix alignment of execute region to 8-byte boundary in ARM linker files

    --legacyalign, --no_legacyalign are deprecated from ARMC6 compiler, in order to
    remove deprecated flags all linker files should strictly align to 8-byte boundary
```

jeromecoutant, 2019-10-11:
Hi
I think question can be closed?
Thx

closed 2019-10-25 by 0xc0170