ARMmbed/mbed-os #10927: DataFlash check for frequency makes no sense?

DL6AKU, 2019-07-01:
### Description

Target: master, Dataflash
Toolchain: 
```
$ /usr/bin/arm-none-eabi-gcc --version                                                                                                              
arm-none-eabi-gcc (Arch Repository) 9.1.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Tools: mbed CLI 1.10.0
SHA of relevant MBED OS commit: `279925b6fc (HEAD -> master, origin/master, origin/HEAD) Merge pull request #10869 from mathias-arm/master`

Question:

Is the code as it is in https://github.com/ARMmbed/mbed-os/blob/master/components/storage/blockdevice/COMPONENT_DATAFLASH/DataFlashBlockDevice.cpp#L152 correct?

Shouldn't it be:

```C
    /* check that frequency is within range */
    if (freq > DATAFLASH_HIGHEST_FREQUENCY) {

        /* cap frequency at the highest supported one */
        _spi.frequency(DATAFLASH_HIGHEST_FREQUENCY);

    } else {
        /* freqency is valid, use as-is */
        _spi.frequency(freq);
    }
```

### Issue request type

    [X] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-07-01:
Looks like it should be.

@ARMmbed/mbed-os-storage please review

SeppoTakalo, 2019-07-16:
That code seems to originate from https://github.com/ARMmbed/mbed-os/commit/1dfcc87f389723c32ffe6bc1070463f60c709449#diff-8c74e6813e4458cbe31bf312a0f47a9f

@marcuschangarm  and @geky Do you remember the reasoning behind that? Is there some particular reason to cap the frequency to lowest one, or is it a bug?

geky, 2019-07-18:
I'm not really sure, but it is explicitly using the DATAFLASH_OP_READ_LOW_FREQUENCY opcode, which is limited to the low frequency. So the code is correct even if the comment is confusing

I'm not sure why use the low frequency read as opposed to the high frequency read. Skimming the datasheet it looks like this might be to avoid handling dummy bytes?

marcuschangarm, 2019-07-19:
Sorry, I'm not much help here.

geky, 2019-07-20:
It's probably safest to not change unless someone can test that the DATAFLASH_OP_READ_HIGH_FREQUENCY command works. It is currently unimplemented.

DL6AKU, 2019-07-20:
I can test it this week and report back on Monday evening. I would be OK with having the lowest frequency selected but we would have to change the comment because it is misleading.

I will test this on Monday with "HIGH" frequency. I will write a test pattern and read it out again. If required I can set up a testbed with about 10 to 20 Nucleo boards that write and read for 24 hours with the high frequency.

pilotak, 2019-08-09:
@DL6AKU any update on tests?

closed 2019-10-25 by ciarmcom