ARMmbed/mbed-os #10930: Working example with mbed-so and Nordic Softdevice ?

supertoast, 2019-07-01:
### Description

Target: custom board with nrf52832, [see here](https://www.decawave.com/product/dwm1001-module/)
Toolchain: gcc-arm-none-eabi-6-2017-q2-update
Tools: mbed-cli 1.10.0
mbed-os-5.13.0-rc4 #7482462434d

I tried to test the example [BLE_LED](https://github.com/ARMmbed/mbed-os-example-ble/tree/master/BLE_LED). With Cordio-Stack it worked, then I wanted to test it with the Softdevice from Nordic.
Unfortunately I didn't find any examples for mbed-os that use the Softdevice, so I replaced the following lines in mbed-app.json.
```
- "target.extra_labels_add": ["CORDIO", "CORDIO_LL", "SOFTDEVICE_NONE", "NORDIC_CORDIO"],
- "target.extra_labels_remove": ["SOFTDEVICE_COMMON", "SOFTDEVICE_S132_FULL", "NORDIC_SOFTDEVICE"]
+ "target.extra_labels_remove": ["CORDIO", "CORDIO_LL", "SOFTDEVICE_NONE", "NORDIC_CORDIO"],
+ "target.extra_labels_add": ["SOFTDEVICE_COMMON", "SOFTDEVICE_S132_FULL", "NORDIC_SOFTDEVICE"]
```
The code could compiled and started but either it crashes at the first SVCCALL to the Softdevice or it crashes in a SVCCALL in the RTX kernel.
My question is does anyone know an example for the NRF52832 that uses mbed-os in combination with the softdevice that works ?
My code can be found [here](https://github.com/supertoast/mbed-os-example-ble/tree/master/BLE_LED).

### Issue request type

    [ x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-07-01:
> Unfortunately I didn't find any examples for mbed-os that use the Softdevice, so I replaced the following lines in mbed-app.json.

@ARMmbed/mbed-os-pan Is this supported?

pan-, 2019-07-01:
Starting with Mbed-os 5.13 soft device is **not** supported.

paul-szczepanek-arm, 2019-07-01:
As noted in release notes, soft device is not supported any more.

0xc0170, 2019-08-13:
> Starting with Mbed-os 5.13 soft device is not supported.

That should answer the question. As there has been no other answers, I'll close this issue.

closed 2019-08-13 by 0xc0170