ARMmbed/mbed-os #10935: Serial callbacks with STM327H7 panic

frmdstryr, 2019-07-01:
### Description

Any attempt to use serial callbacks do not work. Using serial without callbacks work fine.

- What target does this relate to?
NUCLEO-H743ZI

- What toolchain (name + version) are you using?
mbed-os-5.4.0-rc1-15146-gb0073bb70d

- What tools (name + version - is it mbed-cli, online compiler or IDE) are you using?
mbed-cli 1.10.0

- What is the SHA of Mbed OS (git log -n1 --oneline)?
`b0073bb70d (HEAD -> master, origin/master, origin/HEAD) Merge pull request #10774 from fkjagodzinski/fix-uart_irq`

- Steps to reproduce. 

Use this code

```cpp
#include "mbed.h"

DigitalOut led1(LED1);
DigitalOut led2(PE_1); // LED2 is PE_1 on v2 of this board
DigitalOut led3(LED3);
Serial serial(USBTX, USBRX, 115200);

void on_data() {
    led1 = !led1;
    while (serial.readable()) {
        serial.putc(serial.getc());
    }
}

int main() {
    serial.attach(&on_data);

    while (1) {
        led2 = !led2;
        led3 = !led3;
        wait(0.5);
    }
}
```
Open the port and an send any character.  It will dump this error

```

++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x200019A4, Not allowed in ISR context
Location: 0x8002587
Error Value: 0x200019A4
Current Thread: rtx_idle  Id: 0x20001264 Entry: 0x8002595 StackSize: 0x300 StackMem: 0x200015A8 SP: 0x2001FEB8 
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=NUCLEO_H743ZI
-- MbedOS Error Info --

= System will be rebooted due to a fatal error =
= Reboot count(=1) reached maximum, system will halt after rebooting 
```

If the while loop in `on_data` is removed it just turns on led1 and freezes in

```
0x08006f9e in uart_irq () at ./mbed-os/targets/TARGET_STM/TARGET_STM32H7/serial_device.c:66
66                  if (__HAL_UART_GET_FLAG(huart, UART_FLAG_ORE) != RESET) {
(gdb) backtrace
#0  0x08006f9e in uart_irq () at ./mbed-os/targets/TARGET_STM/TARGET_STM32H7/serial_device.c:66
#1  <signal handler called>
#2  osKernelStart () at ./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_kernel.c:564
#3  0x08002ac6 in osKernelInitialize () at ./mbed-os/rtos/TARGET_CORTEX/rtx5/RTX/Source/rtx_kernel.c:519
#4  0x61000000 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 

```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-07-01:
Similar to this one https://github.com/ARMmbed/mbed-os/issues/10906, use `RawSerial`

frmdstryr, 2019-07-01:
Thanks, that works!

If I remove the while loop it seems to keep calling the irq in a loop, why is this?

closed 2019-07-01 by frmdstryr