ARMmbed/mbed-os #10943: Problem with Makefile.tmpl

tdjastrzebski, 2019-07-02:
I think there is a problem with /tools/export/makefile/Makefile.tmpl line 138,
probably introduced in commit 6918e6a76b7ab3e0e7b4f89319ea39d3dad0972b as a result of #9596
Build under Visual Studio 2019 fails with error:
`/usr/bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
/usr/bin/sh: -c: line 1: syntax error: unexpected end of file
make[1]: *** [DISCO-F769NI_LCD_demo.elf] Error 1
C:/Test Projects/DISCO-F769NI_LCD_Make_VS/Makefile:530: recipe for target 'DISCO-F769NI_LCD_demo.elf' failed`
and MAKEFILE line 529-532 read:
`529> $(PROJECT).elf: $(OBJECTS) $(SYS_OBJECTS) $(PROJECT).link_script.ld 
530> 	+@echo "$(filter %.o, $^)" > .link_options.txt
531> 	+@echo "link: $(notdir $@)"
532> 	@$(LD) $(LD_FLAGS) -T $(filter-out %.o, $^) $(LIBRARY_PATHS) --output $@ @.link_options.txt $(LIBRARIES) $(LD_SYS_LIBS)`

0xc0170, 2019-07-02:
@tdjastrzebski Can you send Pr with the fix please

tdjastrzebski, 2019-07-02:
@0xc0170 I would love to but my knowledge of sh and make is limited :/

0xc0170, 2019-07-02:
cc @ARMmbed/mbed-os-tools

mark-edgeworth, 2019-07-02:
Looks like a mismatched quote, but the code seems to be ok. What was the command line used to generate the makefile - I'll try and repeat it.

tdjastrzebski, 2019-07-02:
@mark-edgeworth I do not know exactly how the makefile gets generated behind the scenes. I just exported the program using web interface. However, it seems file gets generated exactly the way it is supposed to, while VS 2019 compiler does not like it. Is that for a good reason? I am unable to tell.

mark-edgeworth, 2019-07-02:
Ah, ok, you're exporting using the online IDE. That helps me reproduce this. Thanks for the info.

tdjastrzebski, 2019-07-03:
I think the problem is NMAKE does not recognize `$^` macro while 
> a caret (^) within a quoted string is treated as a literal caret character
Source:
https://docs.microsoft.com/en-us/cpp/build/reference/filename-macros?view=vs-2019
https://docs.microsoft.com/en-us/cpp/build/reference/special-characters-in-a-makefile?view=vs-2019

mark-edgeworth, 2019-07-03:
The $^ macro in gnu make represents all of the target's dependencies. Looks like nmake uses '$**' for the same thing. If the '$^' in the generated makefile is replaced with '$**', does this work for you?

tdjastrzebski, 2019-07-03:
@mark-edgeworth That's intersting. Replacing the first (only) occurence of `$^` with `$**` lets the build continue. Some other errors are outputed but, I believe, these are related to different issues.
The second instance of `$^` (outside of quotes, 2 lines down) seems to be interpreted OK.
Here is the output:
```
link: DISCO-F769NI_LCD_demo.elf
c:/program files (x86)/microsoft visual studio/2019/preview/linux/gcc_arm/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: warning: cannot find entry symbol Reset_Handler; defaulting to 08000000
arm-none-eabi-objcopy -O binary DISCO-F769NI_LCD_demo.elf DISCO-F769NI_LCD_demo.bin
===== bin file ready to flash: BUILD/DISCO-F769NI_LCD_demo.bin =====
arm-none-eabi-objcopy -O ihex DISCO-F769NI_LCD_demo.elf DISCO-F769NI_LCD_demo.hex
arm-none-eabi-objcopy: DISCO-F769NI_LCD_demo.hex 64-bit address 0x4b4fa308000000 out of range for Intel Hex file
arm-none-eabi-objcopy:DISCO-F769NI_LCD_demo.hex: bad value
C:/Test Projects/DISCO-F769NI_LCD_Make_VS/Makefile:540: recipe for target 'DISCO-F769NI_LCD_demo.hex' failed
make[1]: *** [DISCO-F769NI_LCD_demo.hex] Error 1
Makefile:25: recipe for target 'all' failed
make: *** [all] Error 2
```

mark-edgeworth, 2019-07-03:
Ok, that looks like progress, although I'd expect to have to replace all of the $^ references in the makefile for this to work correctly. If you can confirm that this resolves the problem that you originally reported I'll put a patch into the tools package to resolve this.

tdjastrzebski, 2019-07-03:
@mark-edgeworth Exactly, but when I replace the second `$^` with `$**` I get:
```
link: DISCO-F769NI_LCD_demo.elf
c:/program files (x86)/microsoft visual studio/2019/preview/linux/gcc_arm/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file BSP_DISCO_F769NI: No such file or directory
collect2.exe: error: ld returned 1 exit status
C:/Test Projects/DISCO-F769NI_LCD_Make_VS/Makefile:530: recipe for target 'DISCO-F769NI_LCD_demo.elf' failed
make[1]: *** [DISCO-F769NI_LCD_demo.elf] Error 1
Makefile:25: recipe for target 'all' failed
make: *** [all] Error 2
```
It looks like undocumented NMAKE 'feature'.
I consider this issue solved.  Thank you!

tdjastrzebski, 2019-07-04:
@mark-edgeworth Opps, I need to take it back.
The **.link_options.txt** file is created empty while, I understand, it should contain some LD params :/
That's why subsequent errors are outputed.
I think use of NMake should be simply discouraged.

tdjastrzebski, 2019-07-05:
@mark-edgeworth Yet another update: I attempted build using:
[GNU MCU Eclipse Windows Build Tools] (https://gnu-mcu-eclipse.github.io/windows-build-tools/)
[GNU Make - latest ver 4.2.1] (https://github.com/mbuilov/gnumake-windows)
In both cases make fails with error `unexpected EOF while looking for matching `"'`
What is more, make 'dev' build displays the following warnings:
```
Makefile:8: warning: using environment variable 'PATH'
Makefile:218: warning: conditionally-defined variable 'LINKER_SCRIPT'
gnumake_dev: warning: using environment variable 'PATH'
gnumake_dev: warning: using environment variable 'PATH'
C:/Test Projects/DISCO-F769NI_LCD_demo/Makefile:8: warning: using environment variable 'PATH'
C:/Test Projects/DISCO-F769NI_LCD_demo/Makefile:49: warning: undefined variable 'OBJECTS'
C:/Test Projects/DISCO-F769NI_LCD_demo/Makefile:69: warning: undefined variable 'SYS_OBJECTS'
C:/Test Projects/DISCO-F769NI_LCD_demo/Makefile:193: warning: undefined variable 'INCLUDE_PATHS'
C:/Test Projects/DISCO-F769NI_LCD_demo/Makefile:218: warning: conditionally-defined variable 'LINKER_SCRIPT'
C:/Test Projects/DISCO-F769NI_LCD_demo/Makefile:218: warning: conditionally-defined variable 'LINKER_SCRIPT'
C:/Test Projects/DISCO-F769NI_LCD_demo/Makefile:232: warning: undefined variable 'C_FLAGS'
C:/Test Projects/DISCO-F769NI_LCD_demo/Makefile:328: warning: undefined variable 'CXX_FLAGS'
C:/Test Projects/DISCO-F769NI_LCD_demo/Makefile:428: warning: undefined variable 'ASM_FLAGS'
gnumake_dev[1]: *** No rule to make target 'BSP_DISCO_F769NI/Drivers/BSP/Components/adv7533/adv7533.o', needed by 'DISCO-F769NI_LCD_demo.elf'.  Stop.
gnumake_dev: *** [Makefile:26: all] Error 2
```
All this makes me believe the problem is related to the generated makefile, rather than make/nmake tool.

tdjastrzebski, 2019-07-07:
It turns out probably there is nothing wrong with the generated makefile.
What makes a difference is shell being used. Current, by default VS Code uses MINGW64 version distributed with Git. See: https://code.visualstudio.com/docs/editor/integrated-terminal
When I run **make**  under this shell I get error. When I run the same under windows CMD, PowerShell or build-in sh everything is OK. It is puzzling.
[image]
[image]

0xc0170, 2019-07-08:
what make versions do they use?

tdjastrzebski, 2019-07-08:
@0xc0170 In both cases I call the same **make** 4.2.1 from GNU MCU Eclipse Windows Build Tools
https://github.com/gnu-mcu-eclipse/windows-build-tools/releases
I have verified both **mingw32-make** (4.2.1) and GNU make from
https://github.com/mbuilov/gnumake-windows
give exactly the same results. What makes a difference is shell.

tdjastrzebski, 2019-07-09:
@0xc0170 It is still important to notice that makefile template can be improved.
Debug make build detects significant problems and even refuses to complete "all" task.

mark-edgeworth, 2019-07-09:
I've just tried this in a unix-like environment (actually a Mac terminal window). My steps were:
1. import the DISCO-F769NI_LCD_demo example from os.mbed.com into the online IDE
2. Add the DISCO-F769NI platform as a target
3. Export the project to 'Make GCC ARM'
4. Ran 'make' (v3.8.1) from the Mac terminal.
All is fine, build completes as expected (with a couple of warnings from the code itself).

Repeated with the 'export to vscode', same result.
Looks like the generated makefiles work ok with straight make inside a bash shell. 
Is there anything that actually needs to change here?

tdjastrzebski, 2019-07-09:
@mark-edgeworth Please try creating new DISCO-F769NI_LCD project, then export as "Make-GCC-ARM", and run make from VS Code bash shell. I just tried with fresh project - it still fails. Interestingly, "Blink LED example" builds OK. Why? It is boyound my understanding.
As long as I use a different shell my project builds fine so I am a happy camper albeit the trap is probably still lurking there.

[image]

[image]

tdjastrzebski, 2019-07-10:
@mark-edgeworth Perhaps my last thought: maybe this shell has shorter command length limits?

mark-edgeworth, 2019-07-10:
Indeed it may - I would so dearly love to reduce the length of those command lines!

adbridge, 2020-03-23:
The supplied PR did not actually fix the issue - thus re-opening

Pog3k, 2020-04-05:
Hey guys, i run into the same issue.

Pog3k, 2020-04-05:
i assume that windows cant handle the following call, because  the command itself is too long..

C:/Program Files (x86)/Git/bin/sh.exe -c "echo \"TextLCD/TextLCD.o main.o mbed-os/drivers/AnalogIn.o mbed-os/drivers/BusIn.o mbed-os/drivers/BusInOut.o mbed-os/drivers/BusOut.o mbed-os/drivers/CAN.o mbed-os/drivers/Ethernet.o mbed-os/drivers/FlashIAP.o mbed-os/drivers/I2C.o mbed-os/drivers/I2CSlave.o mbed-os/drivers/InterruptIn.o mbed-os/drivers/InterruptManager.o mbed-os/drivers/RawSerial.o mbed-os/drivers/SPI.o mbed-os/drivers/SPISlave.o mbed-os/drivers/Serial.o mbed-os/drivers/SerialBase.o mbed-os/drivers/Ticker.o mbed-os/drivers/Timeout.o mbed-os/drivers/Timer.o mbed-os/drivers/TimerEvent.o mbed-os/drivers/UARTSerial.o mbed-os/events/EventQueue.o mbed-os/events/equeue/equeue.o mbed-os/events/equeue/equeue_mbed.o mbed-os/events/equeue/equeue_posix.o mbed-os/events/mbed_shared_queues.o mbed-os/features/filesystem/Dir.o mbed-os/features/filesystem/File.o mbed-os/features/filesystem/FileSystem.o mbed-os/features/filesystem/bd/ChainingBlockDevice.o mbed-os/features/filesystem/bd/HeapBlockDevice.o mbed-os/features/filesystem/bd/MBRBlockDevice.o mbed-os/features/filesystem/bd/ProfilingBlockDevice.o mbed-os/features/filesystem/bd/SlicingBlockDevice.o mbed-os/features/filesystem/fat/ChaN/ccsbcs.o mbed-os/features/filesystem/fat/ChaN/ff.o mbed-os/features/filesystem/fat/FATFileSystem.o mbed-os/features/frameworks/greentea-client/source/greentea_metrics.o mbed-os/features/frameworks/greentea-client/source/greentea_serial.o mbed-os/features/frameworks/greentea-client/source/greentea_test_env.o mbed-os/features/frameworks/unity/source/unity.o mbed-os/features/frameworks/utest/mbed-utest-shim.o mbed-os/features/frameworks/utest/source/unity_handler.o mbed-os/features/frameworks/utest/source/utest_case.o mbed-os/features/frameworks/utest/source/utest_default_handlers.o mbed-os/features/frameworks/utest/source/utest_greentea_handlers.o mbed-os/features/frameworks/utest/source/utest_harness.o mbed-os/features/frameworks/utest/source/utest_shim.o mbed-os/features/frameworks/utest/source/utest_stack_trace.o mbed-os/features/frameworks/utest/source/utest_types.o mbed-os/features/mbedtls/platform/src/mbed_trng.o mbed-os/features/mbedtls/src/aes.o mbed-os/features/mbedtls/src/aesni.o mbed-os/features/mbedtls/src/arc4.o mbed-os/features/mbedtls/src/asn1parse.o mbed-os/features/mbedtls/src/asn1write.o mbed-os/features/mbedtls/src/base64.o mbed-os/features/mbedtls/src/bignum.o mbed-os/features/mbedtls/src/blowfish.o mbed-os/features/mbedtls/src/camellia.o mbed-os/features/mbedtls/src/ccm.o mbed-os/features/mbedtls/src/certs.o mbed-os/features/mbedtls/src/cipher.o mbed-os/features/mbedtls/src/cipher_wrap.o mbed-os/features/mbedtls/src/cmac.o mbed-os/features/mbedtls/src/ctr_drbg.o mbed-os/features/mbedtls/src/debug.o mbed-os/features/mbedtls/src/des.o mbed-os/features/mbedtls/src/dhm.o mbed-os/features/mbedtls/src/ecdh.o mbed-os/features/mbedtls/src/ecdsa.o mbed-os/features/mbedtls/src/ecjpake.o mbed-os/features/mbedtls/src/ecp.o mbed-os/features/mbedtls/src/ecp_curves.o mbed-os/features/mbedtls/src/entropy.o mbed-os/features/mbedtls/src/entropy_poll.o mbed-os/features/mbedtls/src/error.o mbed-os/features/mbedtls/src/gcm.o mbed-os/features/mbedtls/src/havege.o mbed-os/features/mbedtls/src/hmac_drbg.o mbed-os/features/mbedtls/src/md.o mbed-os/features/mbedtls/src/md2.o mbed-os/features/mbedtls/src/md4.o mbed-os/features/mbedtls/src/md5.o mbed-os/features/mbedtls/src/md_wrap.o mbed-os/features/mbedtls/src/memory_buffer_alloc.o mbed-os/features/mbedtls/src/net_sockets.o mbed-os/features/mbedtls/src/oid.o mbed-os/features/mbedtls/src/padlock.o mbed-os/features/mbedtls/src/pem.o mbed-os/features/mbedtls/src/pk.o mbed-os/features/mbedtls/src/pk_wrap.o mbed-os/features/mbedtls/src/pkcs11.o mbed-os/features/mbedtls/src/pkcs12.o mbed-os/features/mbedtls/src/pkcs5.o mbed-os/features/mbedtls/src/pkparse.o mbed-os/features/mbedtls/src/pkwrite.o mbed-os/features/mbedtls/src/platform.o mbed-os/features/mbedtls/src/ripemd160.o mbed-os/features/mbedtls/src/rsa.o mbed-os/features/mbedtls/src/sha1.o mbed-os/features/mbedtls/src/sha256.o mbed-os/features/mbedtls/src/sha512.o mbed-os/features/mbedtls/src/ssl_cache.o mbed-os/features/mbedtls/src/ssl_ciphersuites.o mbed-os/features/mbedtls/src/ssl_cli.o mbed-os/features/mbedtls/src/ssl_cookie.o mbed-os/features/mbedtls/src/ssl_srv.o mbed-os/features/mbedtls/src/ssl_ticket.o mbed-os/features/mbedtls/src/ssl_tls.o mbed-os/features/mbedtls/src/threading.o mbed-os/features/mbedtls/src/timing.o mbed-os/features/mbedtls/src/version.o mbed-os/features/mbedtls/src/version_features.o mbed-os/features/mbedtls/src/x509.o mbed-os/features/mbedtls/src/x509_create.o mbed-os/features/mbedtls/src/x509_crl.o mbed-os/features/mbedtls/src/x509_crt.o mbed-os/features/mbedtls/src/x509_csr.o mbed-os/features/mbedtls/src/x509write_crt.o mbed-os/features/mbedtls/src/x509write_csr.o mbed-os/features/mbedtls/src/xtea.o mbed-os/features/mbedtls/targets/TARGET_STM/aes_alt.o mbed-os/features/mbedtls/targets/TARGET_STM/md5_alt.o mbed-os/features/mbedtls/targets/TARGET_STM/sha1_alt.o mbed-os/features/mbedtls/targets/TARGET_STM/sha256_alt.o mbed-os/features/netsocket/NetworkInterface.o mbed-os/features/netsocket/NetworkStack.o mbed-os/features/netsocket/Socket.o mbed-os/features/netsocket/SocketAddress.o mbed-os/features/netsocket/TCPServer.o mbed-os/features/netsocket/TCPSocket.o mbed-os/features/netsocket/UDPSocket.o mbed-os [... cut ...] i.o mbed-os/targets/TARGET_STM/lp_ticker.o mbed-os/targets/TARGET_STM/mbed_overrides.o mbed-os/targets/TARGET_STM/pinmap.o mbed-os/targets/TARGET_STM/port_api.o mbed-os/targets/TARGET_STM/pwmout_api.o mbed-os/targets/TARGET_STM/rtc_api.o mbed-os/targets/TARGET_STM/serial_api.o mbed-os/targets/TARGET_STM/sleep.o mbed-os/targets/TARGET_STM/stm_spi_api.o mbed-os/targets/TARGET_STM/trng_api.o mbed-os/targets/TARGET_STM/us_ticker_16b.o mbed-os/targets/TARGET_STM/us_ticker_32b.o\" > .link_options.txt"

tdjastrzebski, 2020-04-05:
Yep, it is not widely known but windows command cannot exceed 8192 chars.
For this and other reasons I stopped using **MAKE** tool.
Here is my tutorial on how to live without it  
https://github.com/tdjastrzebski/DISCO-F769NI_LCD_demo
It is LCD demo but the same approach can be used for any **mbed-os 2** tutorial.
For **mbed-os 5** simply use **mbed cli**.

PR #12646 (2020-03-20): Makefile: fix for overlong command line
### Description

The list of object files was so long, that it got truncated by the bash (git-bash).

Error was

    /usr/bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
    /usr/bin/sh: -c: line 1: syntax error: unexpected end of file
    mingw32-make[1]: *** [Makefile:679: mbed-os-example-blinky-baremetal.elf] Error 1
    mingw32-make: *** [Makefile:26: all] Error 2

This may fix #10943

I fixed this problem using [this answer](https://stackoverflow.com/a/37506805/5534993).

### Pull request type

    [x] Fix
    [ ] Refactor
    [ ]  [... cut]