ARMmbed/mbed-os #10945: Problem with VS Code export templates

tdjastrzebski, 2019-07-02:
I believe VS Code templates ([settings.tmpl](https://github.com/ARMmbed/mbed-os/blob/master/tools/export/vscode/settings.tmpl) & [tasks.tmpl](https://github.com/ARMmbed/mbed-os/blob/master/tools/export/vscode/tasks.tmpl)) no longer produce correct **settings.json** and **tasks.json** files.
Tasks version **0.1.0** is depreciated. Current version is **2.0.0**.
Settings **C_Cpp.addWorkspaceRootToIncludePath** property is no longer supported.
As a result project does not build.

0xc0170, 2019-07-03:
@tdjastrzebski Patches welcome !

cc @ARMmbed/mbed-os-tools

tdjastrzebski, 2019-07-03:
@0xc0170 I will try since makefile (#10943) is not the most prefered build approach anyway

40Grit, 2019-07-03:
@tdjastrzebski see what @janjongboom has recently been working on for vs code export.

mark-edgeworth, 2019-07-03:
Be aware of the following (possibly associated) issue: https://github.com/ARMmbed/mbed-os/issues/10943

40Grit, 2019-07-03:
Specifically this. https://github.com/janjongboom/mbed-vscode-generator

linlingao, 2019-07-10:
Our tools team acknowledges this problem but they don't have bandwidth to work on it now. @tdjastrzebski please submit a PR if you can.

linlingao, 2019-07-29:
Closing this issue, reason stated in the previous comment.

closed 2019-07-29 by ciarmcom