ARMmbed/mbed-os #10949: NRF52 SDK 15.0 usable input Interrupts limited to 1 by NRFX_GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS

anton-veretenenko, 2019-07-03:
### Description

Target
NRF5x

Version
mbed-os rev #7482462434d5 (tags: latest, mbed-os-5.13.0)
NRF SDK 15.0

After updating to mbed 5.13 I have lost all my input interrupts except 1.
Currently only 1 pin can be used for input interrupt with InterruptIn with default config.
nrfx_gpiote.c driver checks for NRFX_GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS in "channel_port_alloc" function.
NRFX_GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS is set to 1 by sd_config.h for target.

Example
```c++
InterruptIn btn1(11);
InterruptIn btn2(12);
```
For btn2 "nrfx_gpiote_in_init" in nrfx_gpiote.c will return NRFX_ERROR_NO_MEM leaving this interrupt not initialized without any error message.

Making custom target with
```
"macros_add": ["NRFX_GPIOTE_CONFIG_NUM_OF_LOW_POWER_EVENTS=8"]
```
resolves this issue with higher limit set to 8.

Should we think about vendor sdk when working with mbed?

Additional resources
https://devzone.nordicsemi.com/f/nordic-q-a/34882/wrong-configuration-for-ble_app_hrs_freertos-example-in-sdk-15-0/134040

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ X] Bug

40Grit, 2019-07-03:
@anton-veretenenko  Able to make a PRvto fix?

@trowbridgec @aglass0fmilk

anton-veretenenko, 2019-07-03:
Not sure how to do it correctly yet. Limit can be set as high as 32 available GPIOs but then it just reserves memory for all pins even when not used.

40Grit, 2019-10-03:
@0xc0170 We are running into this issue ourselves now on a customer project.

@j3hill

40Grit, 2019-10-03:
@trowbridgec @farrenv @AGlass0fMilk

adbridge, 2021-06-07:
As there has been no update on this for 18 months I am going to close this. Please re-open if this is still an issue.

closed 2021-06-07 by adbridge