ARMmbed/mbed-os #10956: DeviceInformationService with NULL values results in invalid paramter error

AGlass0fMilk, 2019-07-03:
### Description

Passing a `NULL` pointer to the constructor of the Mbed-OS `DeviceInformationService` BLE class causes the Cordio stack to return `INVALID_PARAMETER` when attempting to add the service to the GATT Server. The constructor declaration specifies input parameters to default to NULL if not given, which will cause the service not to be added at all.

https://github.com/ARMmbed/mbed-os/blob/master/features/FEATURE_BLE/ble/services/DeviceInformationService.h

This should be updated to still work if a given parameter is NULL (do not add that characteristic).

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-07-04:
cc @ARMmbed/mbed-os-pan

pan-, 2021-03-24:
@AGlass0fMilk Would https://github.com/ARMmbed/mbed-os-experimental-ble-services/pull/44 solve your issue ?

closed 2021-05-20 by pan-