ARMmbed/mbed-os #10957: ATCmdParser get error data when at->recv

fengshengchang, 2019-07-03:
### Description
I am using ATCmdParser to drive Sim800C module.
I find It's very easy to lose data when calling at->recv function.

Normal  situation(When I send "AT+CTTS=0" command, I receive data "+CTTS: 0"): 

**AT> AT+CTTS=0**
AT? OK%n
AT< AT+CTTS=0
AT< 
**AT< +CTTS: 0**
AT= OK

Error situation: 
AT> AT+CTTS?
AT? +CTTS: 0
%n
AT< 
AT< AT+CTTS?
AT< 
**AT< +CTTS0**

I expected receive "+CTTS: 0", but i got "+CTTS0". The letter ':' is lost.

And I had insure Sim800C module was return correct data in Error situation.

It means that Sim800C module was return correct data, but I got error data when called at->recv function.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

AriParkkila, 2019-07-22:
@fengshengchang Have you tried with bigger `drivers.uart-serial-rxbuf-size` and without debug traces.

fengshengchang, 2019-12-06:
@AriParkkila In my code， drivers.uart-serial-rxbuf-size is:
        "uart-serial-txbuf-size": {
            "help": "Default TX buffer size for a UARTSerial instance (unit Bytes))",
            "value": 1536
        },
        "uart-serial-rxbuf-size": {
            "help": "Default RX buffer size for a UARTSerial instance (unit Bytes))",
            "value": 1536
        },

And without debug traces
This issue still exists