ARMmbed/mbed-os #10958: Unable to attach debugger to running MCU with pyOCD

linlingao, 2019-07-03:
### Description

Target: CY8CPROTO_062_4343W
Toolchain: gcc_arm
IDE: both Eclipse and ModusToolbox

I'm running mbed-cloud-client-example from this repo:
https://github.com/maclobdell/mbed-cloud-client-example/tree/cy8cproto-testing

The code fails with the following error:
++ MbedOS Error Info ++
Error Status: 0x80010133 Code: 307 Module: 1
Error Message: Mutex: 0x800E254, Not allowed in ISR context
Location: 0x10062E0F
Error Value: 0x800E254
Current Thread: main  Id: 0x8008E24 Entry: 0x10062D3D StackSize: 0x1400 StackMem: 0x1004F6CB SP: 0x80FF64C
For more info, visit: https://mbed.com/s/error?error=0x80010133&tgt=CY8CPROTO_062_4343W
-- MbedOS Error Info –

I need to debug what causes this error. My attempt to attach the debug (using DAPLink) interface fails with the following exception from pyOCD:
0000319:INFO:board:Target type is cy8c6xxa
0000350:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0000368:INFO:ap:AP#0 IDR = 0x84770001 (AHB-AP var0 rev8)
0000381:INFO:ap:AP#1 IDR = 0x84770001 (AHB-AP var0 rev8)
0000397:INFO:ap:AP#2 IDR = 0x24770011 (AHB-AP var1 rev2)
0000414:INFO:rom_table:AP#0 ROM table #0 @ 0xf1000000 (designer=034 part=102)
0000429:INFO:rom_table:AP#1 ROM table #0 @ 0xf0000000 (designer=034 part=102)
0000439:INFO:rom_table:[0]<e00ff000:ROM class=1 designer=43b part=4c0>
0000439:INFO:rom_table:  AP#1 ROM table #1 @ 0xe00ff000 (designer=43b part=4c0)
0000448:INFO:rom_table:  [0]<e000e000:SCS-M0+ class=14 designer=43b part=008>
0000453:INFO:rom_table:  [1]<e0001000:DWT-M0+ class=14 designer=43b part=00a>
0000458:INFO:rom_table:  [2]<e0002000:BPU class=14 designer=43b part=00b>
0000463:INFO:rom_table:[1]<f0002000:CTI class=9 designer=43b part=9a6 devtype=14 archid=1a14 devid=0:0:1040800>
0000469:INFO:rom_table:[2]<f0003000:MTB-M0+ class=9 designer=43b part=932 devtype=31 archid=0a31 devid=0:0:0>
0000478:INFO:rom_table:AP#2 ROM table #0 @ 0xe00ff000 (designer=034 part=102)
0000485:INFO:rom_table:[0]<e0080000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=0:0:40800>
0000492:INFO:rom_table:[3]<e008e000:TPIU-M3 class=9 designer=43b part=923 devtype=11 archid=0000 devid=0:0:ca1>
0000497:INFO:rom_table:[4]<e007f000:ROM class=1 designer=034 part=102>
0000498:INFO:rom_table:  AP#2 ROM table #1 @ 0xe007f000 (designer=034 part=102)
0000506:INFO:rom_table:  [0]<e000e000:SCS-M4 class=14 designer=43b part=00c>
0000512:INFO:rom_table:  [1]<e0001000:DWT class=14 designer=43b part=002>
0000518:INFO:rom_table:  [2]<e0002000:FPB class=14 designer=43b part=003>
0000523:INFO:rom_table:  [3]<e0000000:ITM class=14 designer=43b part=001>
0000530:INFO:rom_table:  [4]<e0042000:CTI class=9 designer=43b part=906 devtype=14 archid=0000 devid=0:0:40800>
0000536:INFO:rom_table:  [5]<e0041000:ETM-M4 class=9 designer=43b part=925 devtype=13 archid=0000 devid=0:0:0>
0000541:INFO:cortex_m:CPU core #0 is Cortex-M0+ r0p1
0000548:INFO:cortex_m:CPU core #1 is Cortex-M4 r0p1
0000560:INFO:cortex_m:FPU present: FPv4-SP
0000568:INFO:dwt:2 hardware watchpoints
0000570:INFO:fpb:4 hardware breakpoints, 0 literal comparators
0000582:INFO:dwt:4 hardware watchpoints
0000586:INFO:fpb:6 hardware breakpoints, 4 literal comparators
0000611:INFO:server:Semihost server started on port 4444
0000612:INFO:gdbserver:GDB server started on port 3333
Started by GNU MCU Eclipse
0000621:INFO:server:Semihost server started on port 4445
0000622:INFO:gdbserver:GDB server started on port 3334
Started by GNU MCU Eclipse
0000823:INFO:gdbserver:One client connected!
0000949:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0001497:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0001861:INFO:gdbserver:Attempting to load Argon
0001864:INFO:gdbserver:Attempting to load RTX5
0001865:INFO:gdbserver:RTX5 loaded successfully
0002240:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0002793:INFO:dap:DP IDR = 0x6ba02477 (v2 rev6)
0006274:ERROR:gdbserver:Unhandled exception in handle_message: 16
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pyocd/gdbserver/gdbserver.py", line 539, in handle_message
    reply = handler(msg[msgStart:])
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pyocd/gdbserver/gdbserver.py", line 1018, in handle_query
    resp = self.handle_query_xml(b'threads', int(data[0], 16), int(data[1].split(b'#')[0], 16))
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pyocd/gdbserver/gdbserver.py", line 1206, in handle_query_xml
    xml = self.get_threads_xml()
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pyocd/gdbserver/gdbserver.py", line 1312, in get_threads_xml
    t.text = thread.description
  File "/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/pyocd/rtos/rtx5.py", line 283, in description
    return "%s; Priority %d" % (self.STATES[self._state], self.priority)
KeyError: 16

Per pyOCD expert, it looks like a thread is in a state that doesn't have a matching description in the STATES list.

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ x] Bug

linlingao, 2019-07-03:
@kjbracey-arm @maclobdell @ARMmbed/team-cypress

0xc0170, 2019-07-04:
Can you actually print prior the breakage what values are hold in `self._state` and self.priority ? 

Are you having debuggin enabled in pyOCD - there's debug msg to print error while reading thread status. (https://github.com/mbedmicro/pyOCD/blob/master/pyocd/rtos/rtx5.py#L256)

kjbracey-arm, 2019-07-04:
I would say there is a pyOCD bug here - it should not crash itself when trying to interpret RTOS state if the RTOS data structures are corrupted.

We should determine for certain whether the state is actually corrupted.

There is an option somewhere in platform you can turn on which will make mbed OS's own crash handler attempt to dump all thread info. It would be interesting to see what it shows on this crash.

I see that you are on a multicore device - it's possible there's a general problem here specific to that. Do you get valid thread info when stepping through the code or hitting breakpoints before the fault?

maclobdell, 2019-09-03:
@linlingao - can this be closed?  Is there a pyocd issue?

linlingao, 2019-09-03:
@maclobdell Yes, there is still pyocd issue. I have moved corresponding Jira ticket to IOTDEBUG backlog and documented how to reproduce it.

maclobdell, 2019-09-11:
moved issue to PyOCD.

closed 2019-09-11 by maclobdell