ARMmbed/mbed-os #10960: MbedOS Error Info(sys_arch_protect error)

deventank7, 2019-07-04:
#include "mbed.h"
#include "EthernetInterface.h"
#include "PubSubClient.h"

Serial pc(USBTX, USBRX, 115200);

char* serverIpAddr = "192.168.100.19";  /*Sever ip address*/
int port = 1883; /*Sever Port*/
void callback(char* topic, char* payload, unsigned int len); /*Callback function prototype*/
PubSubClient mqtt(serverIpAddr, port, callback);
EthernetInterface  eth;

void callback(char* topic, char* payload, unsigned int len)
{
    printf("Topic: %s\r\n", topic);
    printf("Payload: %s\r\n\r\n", payload);
    //Send incoming payloads back to topic "/mbed".
    mqtt.publish("mbed", payload, len);
}

int main() {

    eth.init(); //Use DHCP
    eth.connect();

    pc.printf("IP Address is %s\n", eth.getIPAddress());
    
    pc.printf("MQTTClient Tester");
    
    
    char clientID[] = "mbed";   /*Client nanme show for MQTT server*/
    char pub_topic[] = "mbedqtt";   /*Publish to topic : "/mbed" */
    char sub_topic[] = "sub";   /*Subscribe to topic : "/mirror" */
    
    if(!mqtt.connect(clientID)){
        pc.printf("\r\nConnect to server failed ..\r\n");
        return -1;
    }
    
    pc.printf("\r\nConnect to server sucessed ..\r\n");
    
    
    mqtt.subscribe(sub_topic);
       
      
   //eth.disconnect();
    
    while(1) {
    mqtt.loop();
    
    mqtt.publish(pub_topic, "Hello here is mbed...");
    
    }
}

***After running this program I am getting the following error***
++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x8010271
Error Value: 0x0
For more info, visit: https://armmbed.github.io/mbedos-error/?error=0x80FF0100
-- MbedOS Error Info --
sys_arch_protect error

0xc0170, 2019-07-04:
> An unknown error has occurred. This can also happen if any implementation is still using the legacy error() interface. Look at the Location to figure out the address of the error() caller or try building a non-release version to capture the location(address of the caller) where this error originates from.

From the error info - have you checked what is at the location ? Based on the error msg, this is the function invoking the error: sys_arch_protect()

```
    if (osMutexAcquire(lwip_sys_mutex, osWaitForever) != osOK)
        MBED_ERROR(MBED_MAKE_ERROR(MBED_MODULE_NETWORK_STACK, MBED_ERROR_CODE_MUTEX_LOCK_FAILED), "sys_arch_protect error\n");
    return (sys_prot_t) 1;
```

Why does osMutexAcquire fails? is it invoked from IRQ?

linlingao, 2019-07-09:
@deventank7 Have you managed to find the root cause?

deventank7, 2019-07-10:
> @deventank7 Have you managed to find the root cause?

Still not...

linlingao, 2019-07-12:
There's not enough information in the description to allow us to recreate the issue. If you believe this is a bug in Mbed OS, please provide target/tools/source code.

deventank7, 2019-07-16:
Target - NUCLEO-F429ZI
Tools- GCC-ARM

linlingao, 2019-07-16:
Mbed OS version?

deventank7, 2019-07-18:
Mbed OS 5

linlingao, 2019-07-18:
@deventank7 Please respond to the following inquiry from @0xc0170.
An unknown error has occurred. This can also happen if any implementation is still using the legacy error() interface. Look at the Location to figure out the address of the error() caller or try building a non-release version to capture the location(address of the caller) where this error originates from.

From the error info - have you checked what is at the location ? Based on the error msg, this is the function invoking the error: sys_arch_protect()

    if (osMutexAcquire(lwip_sys_mutex, osWaitForever) != osOK)
        MBED_ERROR(MBED_MAKE_ERROR(MBED_MODULE_NETWORK_STACK, MBED_ERROR_CODE_MUTEX_LOCK_FAILED), "sys_arch_protect error\n");
    return (sys_prot_t) 1;
Why does osMutexAcquire fails? is it invoked from IRQ?

linlingao, 2019-07-29:
Close due to lack of response. You can reopen this if you still face the issue.

closed 2019-07-29 by ciarmcom