ARMmbed/mbed-os #10962: Online compiler generated error for Renesas GR-PEACH target

toyowata, 2019-07-04:
### Description

Problem:

Online compiler generated link error (probably in post build phase) and binary image is not created.

```
Warning: L6314W: No section matches pattern *(HEAP). in "extras/mbed-os.lib/targets/TARGET_RENESAS/TARGET_RZ_A1XX/TARGET_RZ_A1H/device/TOOLCHAIN_ARM_STD/MBRZA1H.sct", Line: 64, Col: 8
Warning: L6314W: No section matches pattern *(NC_DATA). in "extras/mbed-os.lib/targets/TARGET_RENESAS/TARGET_RZ_A1XX/TARGET_RZ_A1H/device/TOOLCHAIN_ARM_STD/MBRZA1H.sct", Line: 74, Col: 8
Error: Bootloader not supported on this target. ROM start not found in targets.json.
Info: Unable to download. Fix the reported errors...
```

[image]

How to reproduce:

1. Import example code here: https://os.mbed.com/teams/Renesas/code/pelion-example-common/
2. Set target to GR-PEACH or GR-LCYHEE
3. Select Manage Connect Certificates from Pelion Device Management menu and create credentials.
4. Compile the project

AFAIK, the example code can be built by online IDE without any problem about few weeks ago.
The error only occurs by online IDE.  I could not see any build error by Mbed CLI build for both GCC_ARM, ARM and IAR toolchain.

cc @ARMmbed/team-renesas 

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

mark-edgeworth, 2019-07-17:
Confirmed as repeatable. Many thanks for the clear 'steps to reproduce', this really helps.

0xc0170, 2019-08-13:
> Confirmed as repeatable. Many thanks for the clear 'steps to reproduce', this really helps.

@mark-edgeworth is this something we can fix?

mark-edgeworth, 2019-08-13:
No idea. The problem is still repeatable in latest mbed-os (5.13@b81aeff) but now only the 'bootloader not supported' message appears.

toyowata, 2019-10-31:
Closed as this was fixed in latest online IDE (1.10.25.0).  This was probably related to pack file [change](https://github.com/ARMmbed/mbed-os/pull/10966) in the build script.
Now, it works for both Renesas GR-PEACH and GR-LYCHEE target.

closed 2019-10-31 by toyowata