ARMmbed/mbed-os #10969: ADR Unconfirmed Redundancy issue

michaburger, 2019-07-05:
mbed-os 5.13 (online compiler, revision 7482462434, mbed-os-example-lorawan) failing the ADR Nbtrans redundancy test required by European operators for LoRaWAN 1.0.2

Please check that the device is increasing the counter after DL reception as it should (see attachment)

Details: 
Test 
• Send ADR command with Redundancy parameter set to 3 
• Try to post a downlink to check if the device stop the repetition
Check that devices effectively retransmits uplinks 3 times, with same uplink counter and same packet. A new downlink must stop the uplink repetition, frame counter must be increased.

[image]

[image]

0xc0170, 2019-07-05:
cc @ARMmbed/mbed-os-wan

AnttiKauppila, 2019-09-13:
“At this time, due to our prioritisation of work on customer projects, we are unable to devote the time to fix this issue. As an open source project, we welcome fixes to Mbed OS, so if would like to contribute a fix yourself then we would very much welcome that, please see our contribution guidelines here - https://os.mbed.com/docs/mbed-os/v5.14/contributing/workflow.html#contributions”

AnttiKauppila, 2019-09-13:
@ARMmbed/mbed-os-maintainers This can be closed

closed 2019-09-13 by 0xc0170

loverdeg-ep, 2019-09-14:
@0xc0170 I REALLY, REALLY, dislike seeing valid issues closed for a project that is open source just because the maintaining entity does not currently have the resources to look into it.

This not only discourages others outside ARM from contributing but burys problems.

@simonfordarm

0xc0170, 2019-09-16:
@loverdeg-ep Thanks for the feedback, I'll review these recent closures.

0xc0170, 2019-09-17:
@michaburger 👍 

@loverdeg-ep We need to update our scripts to provide a way to distinguish "won't fix" issues as this one for instance. I'll keep this one on the list to be re-reviewed.

loverdeg-ep, 2019-09-30:
@0xc0170  Back from vacation. Thanks! I like the concept of having open but wontfix items!