ARMmbed/mbed-os #10983: mcuboot

loverdeg-ep, 2019-07-05:
### Description

mcuboot is quickly gaining traction. If I'm not mistaken even TFM uses it?
The mbed bootloader from @ARMmbed/team-embeddedplanet's experience is not well documented, confusing and less featured.

Not a trivial task I know, since Pelion client needs to be considered as well but,
Any plans to make the switch?

### Issue request type

    [ x ] Question
    [ x ] Enhancement
    [ ] Bug

0xc0170, 2019-07-08:
cc @evedon @alzix

loverdeg-ep, 2019-07-09:
@maclobdell

loverdeg-ep, 2019-07-09:
https://github.com/AGlass0fMilk/mcuboot/tree/mbed-port

alzix, 2019-07-09:
cc @SeppoTakalo

loverdeg-ep, 2019-07-09:
For added incentive, MCU boot supports SUIT manifest.

loverdeg-ep, 2019-07-17:
@maclobdell @SeppoTakalo

FYI @AGlass0fMilk has a POC working.

https://github.com/AGlass0fMilk/mcuboot/tree/mbed-port

AGlass0fMilk, 2019-07-17:
https://github.com/AGlass0fMilk/mbed-mcuboot-demo/tree/master

AGlass0fMilk, 2019-07-21:
mcuboot Mbed port is working. It's not optimized for size at this point. I can write up a guide on how to use it if there's interest. 

Mainly, the user just needs to provide a `BlockDevice` pointer to the secondary mcuboot partition where the update candidates are stored.

0xc0170, 2019-08-13:
Thanks for sharing the links. might come handy for someone. 

cc @ARMmbed/mbed-os-storage

SeppoTakalo, 2019-08-13:
@AGlass0fMilk Yes, I'm definitely interested if you can write some kind of guide that describes how you did the integration.

In theory, it should be possible to convert our Pelion Update client to use MCUBOOT, but no study, or no proof of concept exist yet. But it is definitely something that we are interested to study further.

SeppoTakalo, 2019-08-13:
Purpose of Mbed bootloader is to be easiest to port to any platform that Mbed OS supports. It has never intended to be the most optimal for any specific platform.

For example Pelion Client porting guides don't directly name Mbed bootloader as the choice.

We are beginning to work on improving documentation, and perhaps publish the internal design documents so that it is evident what interactions is happening between update client and the bootloader. That would allow anyone to switch the bootloader, or write from scratch something that is most optimal for their target board.

loverdeg-ep, 2019-08-13:
> We are beginning to work on improving documentation, and perhaps publish the internal design documents so that it is evident what interactions is happening between update client and the bootloader. [...]

Please do!

We are willing to build upon anything that is well documented and logically designed.

SeppoTakalo, 2019-08-19:
I'm closing this ticket for backlog clean up purposes.
This is more of a useful discussion than an issue.

40Grit, 2019-08-19:
Since it fit with discussions internally at ARM and is resulting in design docs being released, the issue  has accomoplished its original intent.

Im fine with closing.

closed 2019-08-19 by ciarmcom