ARMmbed/mbed-os #10985: STM32F4 25MHz HSE problem

masoudr, 2019-07-06:
### Description

Currently, I have an STM32F407VGT6 baseboard which uses 8MHz external oscillator which works fine with below `mbed_app.json`.
```
{
    "macros": [
        "MBEDTLS_USER_CONFIG_FILE=\"mbedtls_config.h\"",
        "MBED_HEAP_STATS_ENABLED=1",
        "MBED_STACK_STATS_ENABLED=1",
        "MBED_MEM_TRACING_ENABLED=1"
    ],
    "config": {},
    "target_overrides": {
        "*": {
            "platform.crash-capture-enabled": true,
            "platform.fatal-error-auto-reboot-enabled": true,
            "platform.error-reboot-max": 2,
            "target.lse_available": true,
            "rtos.main-thread-stack-size": 3584,
            "rtos.thread-stack-size": 1536,
            "events.shared-stacksize": 3072,
            "drivers.uart-serial-txbuf-size": 1024,
            "drivers.uart-serial-rxbuf-size": 1024,
            "target.clock_source": "USE_PLL_HSE_EXTC|USE_PLL_HSI",
            "target.macros_add": ["MBED_TICKLESS", "HSE_VALUE=8000000"],
            "target.features_add": [
                "STORAGE"
            ],
            "target.components_add": [
                "SD"
            ],
            "platform.stdio-baud-rate": 115200,
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10"
        }
    }
}
```

I'm checking clock source with below function which works fine for 8MHz OSC.

```
void clock_source(void)
{
    switch (__HAL_RCC_GET_SYSCLK_SOURCE())
    {
    case RCC_SYSCLKSOURCE_STATUS_HSI:
        printf("HSI used as system clock!\r\n");
        break;
    case RCC_SYSCLKSOURCE_STATUS_HSE:
        printf("HSE used as system clock!\r\n");
        break;
    case RCC_SYSCLKSOURCE_STATUS_PLLCLK:
        printf("HSE oscillator clock used as system clock!\r\n");
        break;
    }
}
```
Now on another same board I replaced the osc with a 25MHz (with the same specs and footprint) and changed the setup:
```
{
    "macros": [
        "MBEDTLS_USER_CONFIG_FILE=\"mbedtls_config.h\"",
        "MBED_HEAP_STATS_ENABLED=1",
        "MBED_STACK_STATS_ENABLED=1",
        "MBED_MEM_TRACING_ENABLED=1"
    ],
    "config": {},
    "target_overrides": {
        "*": {
            "platform.crash-capture-enabled": true,
            "platform.fatal-error-auto-reboot-enabled": true,
            "platform.error-reboot-max": 2,
            "target.lse_available": true,
            "rtos.main-thread-stack-size": 3584,
            "rtos.thread-stack-size": 1536,
            "events.shared-stacksize": 3072,
            "drivers.uart-serial-txbuf-size": 1024,
            "drivers.uart-serial-rxbuf-size": 1024,
            "target.clock_source": "USE_PLL_HSE_EXTC|USE_PLL_HSI",
            "target.macros_add": ["MBED_TICKLESS", "HSE_VALUE=25000000", "HSE_STARTUP_TIMEOUT=500"],
            "target.features_add": [
                "STORAGE"
            ],
            "target.components_add": [
                "SD"
            ],
            "platform.stdio-baud-rate": 115200,
            "target.stdio_uart_tx": "PA_9",
            "target.stdio_uart_rx": "PA_10"
        }
    }
}
```
But when the board startup it seems that MCU freezes and can't boot at all. I tried to change `HSE_STARTUP_TIMEOUT` to higher value like 5000 but the same problem. I also tought that maybe my OSC frequency is not adjust and chage it to `HSE_VALUE=24576000`. What is wrong in here? Do I need to do other checks?
Thanks.
### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

jeromecoutant, 2019-07-08:
Hi

Prescaler values were defined considering that HSE clock is 8 MHz...

So I think you need to update the clock tree according to your clocks:
https://github.com/ARMmbed/mbed-os/blob/e7c8f3c7cff0787bbb58b48f8ad09bb8d7a11d0c/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F407xG/TARGET_DISCO_F407VG/system_clock.c#L163

You can be helper by CubeMX tool you can find on st website.

masoudr, 2019-07-08:
@jeromecoutant Thank you very much. But I'm thinking that setting HSE Frequency in the config file is not complete so it needs to replace this value too.

closed 2019-07-08 by masoudr