ARMmbed/mbed-os #10986: How to use SEGGER RTT in latest mbed os?

Reda-RM, 2019-07-08:
### Description
I want to use SEGGER RTT logger in the latest mbed-os (5.13) but I did not find it. I added it manually and used it. Why it is not included in mbed-os? 

### Issue request type

    [x] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-07-08:
@Reda-RM Which one do you use (please provide reference) ?

The same question was answered here: https://github.com/ARMmbed/mbed-os/issues/6438 (the link there is not valid anymore).

Reda-RM, 2019-07-08:
@0xc0170 I used some modified version from the official RTT library ([here](https://www.segger.com/products/debug-probes/j-link/technology/about-real-time-transfer/)). Then I re-targeted the mbed-os traces to use RTT instead of printfs as follows

`static void trace_print_function(const char *format)`
`{`
`    SEGGER_RTT_printf(0, format);`
`    SEGGER_RTT_printf(0, "\n");`
`}` 

`    mbed_trace_print_function_set(trace_print_function);
`

woodsking2, 2019-08-13:
https://github.com/woodsking2/jlink_real_time_transfer
Maybe you could use this lib.
Just like this
```
#include "mbed.h"
#include "jlink_rtt.h"
FileHandle *mbed::mbed_override_console(int fd)
{
    static Jlink_rtt instance;
    return &instance;    
}

```

Reda-RM, 2019-08-18:
Thanks @woodsking2 :). I will close this issue

closed 2019-08-18 by Reda-RM

jhgorse, 2023-11-09:
The link appears to have died.