ARMmbed/mbed-os #10988: tests-mbedmicro-rtos-mbed-malloc tests failing(ARM, GCC_ARM) on Toshiba's TMPM46B

ganesh-ramachandran, 2019-07-08:
### Description
tests-mbedmicro-rtos-mbed-malloc is allocating memory using malloc until the complete heap is allocated. Testcase may be designed by assuming [100 times](https://github.com/ARMmbed/mbed-os/blob/master/TESTS/mbedmicro-rtos-mbed/malloc/main.cpp#L118) maximum allocation. Hence reserved space for 100 pointers. But in M46B, heap size is 512KB RAM. So the malloc allocation count reaches more than 100 times.
Due to which, we are accessing unreserved memory space which is resulting as Hard Fault.

For the tests-mbedmicro-rtos-mbed-malloc test case, observed that test case is passed by increasing '[ALLOC_ARRAY_SIZE](https://github.com/ARMmbed/mbed-os/blob/master/TESTS/mbedmicro-rtos-mbed/malloc/main.cpp#L111)' from 100 to 113. **Please confirm**.

To reproduce run below command: 

```
mbed test -t <ARM, GCC_ARM> -m TMPM46B -n tests-mbedmicro-rtos-mbed-malloc -v
```

### Issue request type
    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-07-08:
I don't follow - malloc returns space that is unreserved?

cc @ARMmbed/mbed-os-core

ganesh-ramachandran, 2019-07-08:
> I don't follow - malloc returns space that is unreserved?

@0xc0170, malloc allocates till the heap size is available. However 'array' is declared to hold 100 elements. Since M46B RAM is 512 KB, malloc allocated location ('data') is stored beyond the 'array' size ('count' is 100 and above in the below snippet).

`        data = malloc(size);`
`        if (NULL != data) {`
`           array[count++] = data;`

We are accessing beyond the allocated array size which is resulting in Hard fault.

evedon, 2019-07-10:
This is fixed by https://github.com/ARMmbed/mbed-os/pull/10977

closed 2019-07-10 by ganesh-ramachandran