ARMmbed/mbed-os #10992: Unable to override target.components_add in mbed_library.json

AGlass0fMilk, 2019-07-09:
### Description

I am trying to enable the `FLASHIAP` component and the `STORAGE` feature for my port of mcuboot that must use a FlashIAPBlockDevice.

It appears that adding this to my `mbed_lib.json` has no effect:

```
    "target_overrides": {
	"*": {
	    "target.restrict_size": "0x10000",
	    "target.features_add": ["STORAGE"],
	    "target.components_add": ["FLASHIAP"]
	}
    },
```

I get the following error:

```
[Fatal Error] kv_config.cpp@35,33: FlashIAPBlockDevice.h: No such file or directory
[ERROR] ./mbed-os/features/storage/kvstore/conf/kv_config.cpp:35:33: fatal error: FlashIAPBlockDevice.h: No such file or directory
 #include "FlashIAPBlockDevice.h"
```

It appears the build scripts don't allow you to override `target.components_add` in an `mbed_library.json`. Is this true? If so, is it intentional?

### Issue request type

    [ ] Question
    [ ] Enhancement
    [X] Bug

0xc0170, 2019-07-09:
Reading the docs: `COMPONENT, controlled from the configuration value target.components.` - it should work.

cc @ARMmbed/mbed-os-tools

AGlass0fMilk, 2019-07-09:
The documentation for 5.12 mentions to use something like:

```
“target_overrides": {
   "MTB_ADV_WISE_1570": {
         "target.components_add": ["SPIF"]
    }
}
```

To add the SPIF component to a target in your `mbed_app.json`.

I think this should be consistent between `mbed_app.json` and `mbed_lib.json`, similar to how `features` and `device_has` works, right?

loverdeg-ep, 2019-07-09:
@theotherjimmy due to topic knowledge-ability.

AGlass0fMilk, 2019-07-09:
Another inconsistency -- when I used the recommended `target.components` setting in my `mbed_lib.json` to enable `FLASHIAP` block devices, it conflicts with components added by the `mbed_app.json`. The library doesn't explicitly require any other storage components except `FLASHIAP`. My `mbed_app.json` attempts to use `target.components_add` to _also enable_ the `QSPIF` block device.

This results in the following error:

```
[ERROR] Configuration conflict. The component QSPIF both added and removed.
```

40Grit, 2019-07-10:
@trowbridgec

trowbridgec, 2019-07-10:
I'm not certain, but I think the `STORAGE` feature is really the KVStore: https://github.com/ARMmbed/mbed-os/blob/master/docs/design-documents/features/storage/Configuration/CONFIGURATION.md

What happens if you try without adding the `STORAGE` feature?

0xc0170, 2019-08-13:
> The documentation for 5.12 mentions to use something like:

Correct.  Same as we do for instance in the bootloader: https://github.com/ARMmbed/mbed-bootloader-extended/blob/master/mbed_app.json#L51

> [ERROR] Configuration conflict. The component QSPIF both added and removed.

How to reproduce the error?

0xc0170, 2019-08-14:
Is this the same issue - this duplicate :  https://github.com/ARMmbed/mbed-os/issues/9189?

rvasquez6089, 2019-10-18:
I'm having the about the same exact issue. I am trying to use the I2CEEBlockDevice in components and it clearly isn't being picked up by the linker via the mbed_app.json components_json key.
("target.components_add": ["I2CEEBlockDevice", "storage"])  I got around this by adding the .h and .cpp file for I2CEEBlockDevice. Its a little ugly but its a decent temporary work around for people needing a solution. 
My includes look like code below. Hope you guys can get this patched up soon!

`#include <features/storage/filesystem/littlefs/LittleFileSystem.h>
#include "mbed-os/components/storage/blockdevice/COMPONENT_I2CEE/I2CEEBlockDevice.h"

#include "mbed-os/features/storage/blockdevice/BlockDevice.h"
#include "mbed-os/components/storage/blockdevice/COMPONENT_I2CEE/I2CEEBlockDevice.cpp"`

isaev-d, 2020-11-05:
Same issue. mbed 6.2. CLI and Platform IO effected. Looks like only components_add option does not work.

kalyanram23, 2021-01-23:
same issue. Is not recognized by the linker. 

`{
	"target_overrides":{
		"NUCLEO_L152RE":{
			"target.features_add": ["STORAGE"],
			"target.components_add":["FlashIAP"]
		}
	}
}`

0xc0170, 2021-01-26:
As we are moving to the new tools, I'll close this. 
Components will be added via `target_link_libraries` (will be all explained in the migration guide). Give it a try now, and help us to improve the tooling.

closed 2021-01-26 by 0xc0170