ARMmbed/mbed-os #10993: Export CMake GCC ARM not wrapping functions

vidavidorra, 2019-07-09:
### Description

**Toolchain:**
```
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2018-q2-update) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907]
```

**Mbed OS**
Using version 5.12.1 (from the mbed-os-blinky example).
```
0063e5de32fc575f061244c96ac60c41c07bd2e6
```

**Target**
This affects all targets, I tested for `NUCLEO_F207ZG`.

**Tools**
This is the tools version although it is not relevant to this issue I think.
```
$ mbed --version
1.10.0
```

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

### Steps to reproduce

1. Get the blinky example.

   ```shell
   $ mbed import mbed-os-example-blinky
   $ cd mbed-os-example-blinky
   ```

2. Apply patch to debug memory usage.

   ```diff
   diff --git a/main.cpp b/main.cpp
   index e3448f2..98e8332 100644
   --- a/main.cpp
   +++ b/main.cpp
   @@ -10,23 +10,29 @@ DigitalOut led1(LED1);
    
    #define SLEEP_TIME                  500 // (msec)
    #define PRINT_AFTER_N_LOOPS         20
   -
   +size_t malloc_size = 16 * 1024;
    // main() runs in its own thread in the OS
    int main()
    {
   -    SystemReport sys_state( SLEEP_TIME * PRINT_AFTER_N_LOOPS /* Loop delay time in ms */);
   +    SystemReport sys_state(SLEEP_TIME / PRINT_AFTER_N_LOOPS);
    
   -    int count = 0;
   +    // int count = 0;
        while (true) {
            // Blink LED and wait 0.5 seconds
            led1 = !led1;
            wait_ms(SLEEP_TIME);
    
   -        if ((0 == count) || (PRINT_AFTER_N_LOOPS == count)) {
   -            // Following the main thread wait, report on the current system status
   -            sys_state.report_state();
   -            count = 0;
   +        void* ptr = malloc(malloc_size);
   +        printf("Allocated %d bytes (success %d)\n", malloc_size, ptr != NULL);
[... 682 lines cut ...]
   Current heap: 103616
   Max heap size: 103648
   ================ THREAD STATS ===============
   ID: 0x20004060 
   Name: main 
   State: 2 
   Priority: 24 
   Stack Size: 4096 
   Stack Space: 3160 
   ID: 0x200041b0 
   Name: rtx_idle 
   State: 1 
   Priority: 1 
   Stack Size: 512 
   Stack Space: 432 
   ID: 0x2000416c 
   Name: rtx_timer 
   State: 3 
   Priority: 40 
   Stack Size: 768 
   Stack Space: 664 
   ```
   
   </p>
   </details>

0xc0170, 2019-07-09:
What CMake do you use? This looks like an environment issue (this is working with GCC ARM compilation, should also makefile), is this CMake related? How they handle `-` vs `--` ?

cc @ARMmbed/mbed-os-tools

vidavidorra, 2019-07-09:
This is on a recent version of CMake.
```
cmake version 3.14.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).
```

> How they handle - vs -- ?

I've got no idea how CMake handles this. It creates the following command for linking I believe (_BUILD/CMakeFiles/mbed-os-example-blinky.dir/build.make_). This just passes the `-Wl,--wrap,...` options to `arm-none-eabi-cpp` which I think would be the case for the make exporter as well but I'm not sure.

```
	@$(CMAKE_COMMAND) -E cmake_echo_color --switch=$(COLOR) --green --bold --progress-dir=/home/jdebruijn/projects/mbed-os-example-blinky/BUILD/CMakeFiles --progress-num=$(CMAKE_PROGRESS_827) "Linking CXX executable mbed-os-example-blinky"
	cd /home/jdebruijn/projects/mbed-os-example-blinky && arm-none-eabi-cpp -E -P -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,malloc_r -Wl,--wrap,free_r -Wl,--wrap,realloc_r -Wl,--wrap,memalign_r -Wl,--wrap,calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m3 -mthumb -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x100000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x20000 -DMBED_BOOT_STACK_SIZE=1024 -dumpspecs ./mbed-os/targets/TARGET_STM/TARGET_STM32F2/TARGET_NUCLEO_F207ZG/device/TOOLCHAIN_GCC_ARM/STM32F207ZGTx_FLASH.ld -o /home/jdebruijn/projects/mbed-os-example-blinky/BUILD/mbed-os-example-blinky_pp.link_script.ld
	$(CMAKE_COMMAND) -E cmake_link_script CMakeFiles/mbed-os-example-blinky.dir/link.txt --verbose=$(VERBOSE)
	arm-none-eabi-objcopy -O ihex /home/jdebruijn/projects/mbed-os-example-blinky/BUILD/mbed-os-example-blinky /home/jdebruijn/projects/mbed-os-example-blinky/BUILD/mbed-os-example-blinky.hex
	/home/linuxbrew/.linuxbrew/Cellar/cmake/3.14.4/bin/cmake -E echo --\ built:\ /home/jdebruijn/projects/mbed-os-example-blinky/BUILD/mbed-os-example-blinky.hex
```

vidavidorra, 2019-07-09:
I just tried this with the make exporter (`make_gcc_arm`) and it produced the output for heap stats. Might indeed be something with CMake and/or the CMake exporter.

```shell
$ make all --trace
echo "link: mbed-os-example-blinky.elf"
link: mbed-os-example-blinky.elf
arm-none-eabi-gcc -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_memalign_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m3 -mthumb -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x100000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x20000 -DMBED_BOOT_STACK_SIZE=1024  -T mbed-os-example-blinky.link_script.ld  --output mbed-os-example-blinky.elf @.link_options.txt  -Wl,--start-group -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys  -Wl,--end-group
```

vidavidorra, 2019-07-09:
After some more debugging I've found that the last line n the post before with CMake contains the issue. This line shows e.g. `-Wl,--wrap,__free_r` which has two underscores before `free` where the same line from the make post contains just one. I've verified this is the problem by executing those commands manually and removing one of the underscores.
So now the question is, where does that extra underscore come from.

```
[100%] Linking CXX executable mbed-os-example-blinky
cd /home/jdebruijn/projects/mbed-os-example-blinky && arm-none-eabi-cpp -E -P -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,_malloc_r -Wl,--wrap,_free_r -Wl,--wrap,_realloc_r -Wl,--wrap,_memalign_r -Wl,--wrap,_calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m3 -mthumb -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x100000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x20000 -DMBED_BOOT_STACK_SIZE=1024 ./mbed-os/targets/TARGET_STM/TARGET_STM32F2/TARGET_NUCLEO_F207ZG/device/TOOLCHAIN_GCC_ARM/STM32F207ZGTx_FLASH.ld -o /home/jdebruijn/projects/mbed-os-example-blinky/BUILD/mbed-os-example-blinky_pp.link_script.ld
/home/linuxbrew/.linuxbrew/Cellar/cmake/3.14.4/bin/cmake -E cmake_link_script CMakeFiles/mbed-os-example-blinky.dir/link.txt --verbose=1
/usr/local/bin/arm-none-eabi-g++  -std=gnu++98 -fno-rtti -Wvla -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -O0 -g3 -DMBED_DEBUG -DMBED_TRAP_ERRORS_ENABLED=1 -mcpu=cortex-m3 -mthumb -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x100000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x20000  -include mbed_config.h -O3 -DNDEBUG -Wl,--gc-sections -Wl,--wrap,main -Wl,--wrap,__malloc_r -Wl,--wrap,__free_r -Wl,--wrap,__realloc_r -Wl,--wrap,__memalign_r -Wl,--wrap,__calloc_r -Wl,--wrap,exit -Wl,--wrap,atexit -Wl,-n -mcpu=cortex-m3 -mthumb -DMBED_ROM_START=0x8000000 -DMBED_ROM_SIZE=0x100000 -DMBED_RAM_START=0x20000000 -DMBED_RAM_SIZE=0x20000 -DMBED_BOOT_STACK_SIZE=1024  -Wl,--start-group -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys  -Wl,--end-group -T /home/jdebruijn/projects/mbed-os-example-blinky/BUILD/mbed-os-example-blinky_pp.link_script.ld  @CMakeFiles/mbed-os-example-blinky.dir/objects1.rsp  -o mbed-os-example-blinky -lstdc++ -lsupc++ -lm -lc -lgcc -lnosys
```

SibertEnovates, 2021-09-30:
Disclaimer: I'm not an expert in linking and whatnot so the details and terminology might not be totally on point be I believe that the gist of this post is correct.

This issue is seems to be caused by the CMake exporter. In `tools/export/cmake/__init__.py` on line 96:
`'ld_flags': re.sub("--wrap,_(?!_)", "--wrap,__", " ".join(self.toolchain.ld[1:])),`

This line finds wraps that have a single underscore before the symbol and adds another underscore to ensure each wrap conforms to '--wrap,__symbol' instead of '--wrap,_symbol'.

In case of the memory tracing, `mbed_alloc_wrappers.cpp` uses `__real__<allocsymbol>_r` and `__wrap__<allocsymbol>_r`.

When linking, the wrap function will resolve to '__real_<symbol>' and '__wrap_<symbol>'. With the substition done as described above, it will try to resolve to '__real___<allocsymbol>_r' or '__wrap___<allocsymbol>_r'. This causes errors when linking because the linker can't find these symbols.

Solution to this problem is undoing the substitution in the cmake exporter.

0xc0170, 2021-09-30:
Thanks @SibertEnovates for the detailed description. 

As exporters were deprecated long time ago, they are not maintained anymore, I'll close this issue. Btw, you can use mbed-cli 2 (uses CMake).

closed 2021-09-30 by 0xc0170

SibertEnovates, 2021-09-30:
mbed-cli 2 doesn't seem to generate a CMakeList-file (or I can't find how) so it's not really useful in our workflow

0xc0170, 2021-09-30:
It uses the CMakeLists provided in Mbed OS and examples have own project CMakeLists file (https://github.com/ARMmbed/mbed-os-example-sockets/blob/development/CMakeLists.txt)