ARMmbed/mbed-os #10995: Suddenly getting compiler error 230 - Online compiler

Mads-hg, 2019-07-09:
### Description

Hi,
I'm suddenly getting a compiler error 230 ("The build system did not finish successfully"), when compiling programs, in the online compiler, that some days/weeks ago compiled fine without any errors.

Target: FRDM-K64F
Mbed OS revision: 101:7cff1c4259d7

Thanks,
/Mads

### Issue request type
    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-07-09:
I was able to reproduce it, will ask around.

0xc0170, 2019-07-09:
Trying blinky  Mbed OS, also older nxp gpio examples using older mbed 2 release, compile - not that old as you were testing.

I am able to reproduce this on example from 2013, using K64F - I2C_HelloWorld example (https://os.mbed.com/teams/mbed_example/code/I2C_HelloWorld/, the very first revision).

> Mbed OS revision: 101:7cff1c4259d7

This is very old Mbed 2 revision. Is there an example - how to reproduce?

Mads-hg, 2019-07-09:
Hi,

I made a workaround by updating to rev 106:ba1f97679dad. When using an older revision, the before mentioned error occurs. And if I use one of the newest revisions, I get a buttload of errors.... So I'll just stick to 106:ba1f97679dad for now.

Thanks,
/Mads

0xc0170, 2019-08-13:
Thanks @Mads-hg for the update. We will close the issue as resolved (@mark-edgeworth provided an answer).

closed 2019-08-13 by 0xc0170