ARMmbed/mbed-os #10996: MBED_OS_DIR not being resolved

jacob-larsen-samsung, 2019-07-09:
### Description

I am trying to use the variable MBED_OS_DIR to place my os directory outside the app directory, but I get errors. This is my result on Linux for mbed-cli:
$ mbed compile -v
[mbed] Working path "/mbed" (program)
[mbed] WARNING: Missing Python modules were not auto-installed.
       The Mbed OS tools in this program require the following Python modules: hidapi, pywin32, wmi
       You can install all missing modules by running "pip install -r requirements.txt" in "/mbed-os"
       On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
---
[mbed] Exec "/usr/bin/python -u /mbed-os/tools/make.py -t GCC_ARM -m S3JT100 --source . --build ./BUILD/S3JT100/GCC_ARM -v" in "/mbed"
Building project mbed (S3JT100, GCC_ARM)
Scan: mbed
Traceback (most recent call last):
  File "/mbed-os/tools/make.py", line 77, in wrapped_build_project
    src_dir, build_dir, mcu, *args, **kwargs
  File "/mbed-os/tools/build_api.py", line 583, in build_project
    src_paths, toolchain, inc_dirs=inc_dirs)
  File "/mbed-os/tools/resources/__init__.py", line 581, in scan_with_toolchain
    toolchain.config.load_resources(self)
  File "/mbed-os/tools/config/__init__.py", line 1514, in load_resources
    self.validate_config()
  File "/mbed-os/tools/config/__init__.py", line 1457, in validate_config
    raise error
UndefinedParameter: Attempt to override undefined parameter 'platform.stack-stats-enabled' in 'application[*]'
[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/mbed"
       Command: "/usr/bin/python -u /mbed-os/tools/make.py -t GCC_ARM -m S3JT100 --source . --build ./BUILD/S3JT100/GCC_ARM -v"
       Tip: You could retry the last command with "-v" flag for verbose output

Note that “mbed” is wrapped as a docker container in order to properly resolve the build dependencies of mbed. 

If I place a symlink as app/mbed-os and point to the same location as MBED_OS_DIR points to, then compilation works. This looks like a bug where the build system does not resolve MBED_OS_DIR.

I put MBED_OS_DIR as a local app variable as the global variables do not map well to a docker container

### Issue request type

    [ ] Question
    [ ] Enhancement
    [x] Bug

0xc0170, 2019-07-09:
How to reproduce this?

cc @ARMmbed/mbed-os-tools

jacob-larsen-samsung, 2019-07-09:
git clone https://github.com/ARMmbed/mbed-os.git
git clone https://github.com/ARMmbed/mbed-os-example-blinky.git

cd mbed-os-example-blinky

Contents of .mbed:
ROOT=.
TOOLCHAIN=GCC_ARM
TARGET=K64F
MBED_OS_DIR=../mbed-os

$ mbed compile -v
[mbed] Working path "/mbed" (program)
[mbed] WARNING: Missing Python modules were not auto-installed.
The Mbed OS tools in this program require the following Python modules: hidapi, pywin32, wmi, psutil
You can install all missing modules by running "pip install -r requirements.txt" in "../mbed-os"
On Posix systems (Linux, etc) you might have to switch to superuser account or use "sudo"
[mbed] Exec "/usr/bin/python -u /mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM -v" in "/mbed"
Building project mbed (K64F, GCC_ARM)
Scan: mbed
Traceback (most recent call last):
File "/mbed-os/tools/make.py", line 78, in wrapped_build_project
args, **kwargs
File "/mbed-os/tools/build_api.py", line 584, in build_project
src_paths, toolchain, inc_dirs=inc_dirs)
File "/mbed-os/tools/resources/init.py", line 582, in scan_with_toolchain
toolchain.config.load_resources(self)
File "/mbed-os/tools/config/init.py", line 1514, in load_resources
self.validate_config()
File "/mbed-os/tools/config/init.py", line 1457, in validate_config
raise error
UndefinedParameter: Attempt to override undefined parameter 'platform.stack-stats-enabled' in 'application[]'
[mbed] ERROR: "/usr/bin/python" returned error.
Code: 1
Path: "/mbed"
Command: "/usr/bin/python -u /mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM -v"
Tip: You could retry the last command with "-v" flag for verbose output

jacob-larsen-samsung, 2019-07-09:
This workaround makes compile work:

ln -s ../mbed-os mbed-os