ARMmbed/mbed-os #10998: Please add the STM32L496RG (ST NUCLEO L496RG) to the target list

chandumvm, 2019-07-09:
### Description

### Issue request type

    [ ] Question
    [ ] Enhancement
    [ ] Bug

0xc0170, 2019-07-09:
@chandumvm Adding this target to Mbed OS ? 

cc @ARMmbed/team-st-mcd

chandumvm, 2019-07-10:
> Thanks for the reply @0xc0170 Martin Kojtal 
> cc @ARMmbed/team-st-mcd 

 Yes, I need the Mbed OS of this ( STM32L496RG ) board, which is missing in the following link :
  https://github.com/ARMmbed/mbed-os/tree/master/targets/TARGET_STM/TARGET_STM32L4/TARGET_STM32L496xG

chandumvm, 2019-07-10:
> @ciarmcom thanks for the help!

  this link is not opening 
  https://jira.arm.com/browse/MBOCUSTRIA-1421

LMESTM, 2019-07-10:
@chandumvm  We could not find a NUCLEO with this part number, can you share a link to the board you need ? If you need MCU support for your own board, you may follow the steps below and we can review the Pull Request
https://os.mbed.com/teams/ST/wiki/steps-to-create-a-new-STM32-platform

0xc0170, 2019-08-13:
> @chandumvm We could not find a NUCLEO with this part number, can you share a link to the board you need ? If you need MCU support for your own board, you may follow the steps below and we can review the Pull Request
https://os.mbed.com/teams/ST/wiki/steps-to-create-a-new-STM32-platform

I believe this is resolved. I'll close this issue.

closed 2019-08-13 by 0xc0170