ARMmbed/mbed-os #10999: UARTserial wrong buffer type

CodingGhost, 2019-07-09:
### Description

the UARTserial library uses a circular buffer, but as soon as it gets full, the rx IRQ is stopped, so no more characters are loaded, and no SIGIO event will be triggered, this essentially makes it a simple fifo buffer with overflow protection, and it is no longer a circular buffer.

This leads to following:
the buffer gets full, no SIGIO will be triggered anymore, so no characters a read from the buffer, thus the buffer will stay full, and no character will be read ever again.

why cant it just be a regular circular buffer?
I have made a dirty quick workaround for this issue, by  removing

`     if (_rx_irq_enabled && _rxbuf.full()) {
         disable_rx_irq();
     }`

and adding `|| _rxbuf.full()` to the if statement.
this quick fix makes it a circular buffer again, but it has to be done better ofcourse.

### Issue request type

    [ ] Question
    [x ] Enhancement
    [ ] Bug

0xc0170, 2019-07-09:
cc @kjbracey-arm

kjbracey-arm, 2019-07-09:
What do you mean by "circular"? I guess you mean dropping old data rather than new data? I suppose that's an option - in networking the fashion is to do that on the grounds that newer packets are more likely to contain useful info than old ones. Not sure it carries over to a serial buffer though.  Mainly the current implementation is just that way for symmetry with Tx, really. I find it hard to see how changing overrun behaviour is going to help - overrun means you're in trouble anyway.

The current version does have the minor benefit that if the application is not reading data (and that's possibly quite common!), rx interrupts get automatically shut down as soon as the buffer is full, saving any more power from serial activity. If we dropped at the other end, the system will keep churning the buffer on Rx IRQs to no benefit.

Regardless of that choice though, the application does not need to receive a new signal. The buffer had data available, and still has data available. The state has not changed.

The signal handler is only called when the buffer transitions from empty to non-empty - it's edge-based, not level-based.

    if (was_empty && !_rxbuf.empty()) {
        wake();
    }

The application has to empty the buffer to get a new signal - it must keep reading until there's nothing left.  If it doesn't read until empty, it will have problems whether or not the buffer gets full. I don't see how the overrun choice has anything to do with that.

I'm guessing you've got a faulty application that's let the buffer become full because you don't read all the data after a sigio, and adding another sigio when it gets full gives you a recovery path. But it's a bit late - you've probably lost data.

CodingGhost, 2019-07-22:
ok then, this makes sense to me. But I cannot find the cause of the issue.
when a sigio gets called, I read into a buffer of 6016 bytes (uartSerial buffer has 6000 bytes). so in theory the whole buffer should be cleared, but still, the buffer gets full and wont be emptied.

CodingGhost, 2019-07-22:
@kjbracey-arm I just tried something.
with the current implementation, sigio only gets called, at the transition from a empty buffer to a non-empty buffer, but not on the transition from an half full buffer to a 3/4 full buffer. would it be a possible/good Idea to call it on every change of the file? I think it would add very little to the cpu load, as we can still disable the IRQ if the buffer is full.

kjbracey-arm, 2019-08-12:
There should be no need to generate these events, as long as the application doesn't stop reading before data runs out.

Your application reasoning should not be based on any view of buffer sizes or the like, or it will be fragile. When it gets a sigio, it should keep reading data until it gets a "would block" response to say there's no more data available.

You don't have to do this all in one go - you could conceivably pace yourself in an event type mechanism:

     sigio_handler:
          queue packet_handler

    packet_handler:
          try to read one datagram
          if datagram received
              process it
              queue packet_handler

But you can't rely on getting another sigio for incoming data until a socket read has told you "no more data".  The assumption is that it's already told you there's data, and it hasn't told you there isn't, so it doesn't need to tell you again.

This is fairly standard behaviour for such event systems, and adding extra events for level changes will just tend to partially hide issues. If your program only works because an extra event happens to come in later for a 1/2->3/4 transition as extra data arrives, then you're relying on following data "forcing" your unread data out. What if that extra following data doesn't come?

0xc0170, 2019-08-13:
@CodingGhost Have we answered your question? If yes, please close the issue as resolved.